raspiblitz/raspiblitz #1562: Loop Out's are failing

rene78, 2020-09-18:
I am not able to use Loop Out anymore.
[image]

Is it related to this error message?
[image]

rootzoll, 2020-09-18:
Can you check if the macaroon really exist? Have you updated from an older verson? Have you tried to reset macaroons with: MENU > LNDCRED > RESET ?

rene78, 2020-09-18:
@rootzoll The file doesn't exist
[image]
Even after resetting the macaroons as instructed by you the file is still missing and the same error message appears after entering _loopd_
I cannot tell you whether this version is an updated one because I bought the Raspiblitz (metal case) off eBay one week ago.

frennkie, 2020-09-18:
I would have recommended MENU > LNDCRED > SYNC.

rene78, 2020-09-18:
@frennkie Still the same outcome. Same 3 files 🤔...

frennkie, 2020-09-18:
Running loopd as user admin is currently not correct on Raspiblitz. There is a Systemd service that is running loopd (as bitcoin if I recall correctly).

The failing loops can have many reasons within the LN network.. E.g. Capacity issues.

rene78, 2020-09-18:
@frennkie You seem to be right. Just tried a Loop Out with only 250k Satoshi and it seems to work.
I'd close the issue again. Thanks for the help everyone!

closed 2020-09-18 by rene78