raspiblitz/raspiblitz #1564: Idea : underclocking to make the blitz greener

laurentdebricon, 2020-09-18:
Hey, 

i now mainly use my raspiblitz without the screen (with the var hdmi #767 (it should be on the README.md by the way this is great !) because i'm used to ssh, it makes the pi hotter, and the bright screen attracts too much my kids ... )

My pi is at 64°C with the heatsink, and i thought why not underclock it now that I am synced, making my full node cheaper and environmentally more friendly ... yeah!

I have found a post about good results and 10 degrees less , just with 8 variables . 
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=250167&sid=bf75a4ceb8b61d53a08acb0c6a7c8830&start=25#p1712952

```
core_freq_min=150
gpu_freq_min=150
h264_freq_min=150
isp_freq_min=150
v3d_freq_min=150

over_voltage=-2
over_voltage_min=-4

over_voltage_sdram=-1

```
Who has tried ? in which file to put this for the raspiblitz ? I want to try, i also want to know what you think of it before maybe wasting my time.

laurentdebricon, 2020-09-19:
wow this guy even turn off ethernet port LEDs !
https://community.roonlabs.com/t/underclocked-pi-4-the-perfect-low-budget-77-usb-bridge/93743/8

stealth mode , me gusta !

rootzoll, 2020-09-20:
If anybody can report good results on this (everything keeps running OK and temp goes down) then we can add this to the settings as a "eco mode" people can activate by choice.

holgern, 2020-09-22:
I used the following setup:
```

gpu_freq_min=200
core_freq_min=250

h264_freq_min=100
isp_freq_min=100
v3d_freq_min=100
h264_freq=100
isp_freq=100
v3d_freq=100
# over_voltage=-2

##turn off mainboard LEDs
dtoverlay=act-led

# Disable the ACT LED.

dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Disable the PWR LED.

dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

##turn off ethernet port LEDs
#dtparam=eth_led0=4
#dtparam=eth_led1=4

# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

# Disable Wlan
dtoverlay=disable-wifi
#Disable bt
dtoverlay=disable-bt
```
It is important to set min freq and freq to the same value to prevent frequency stepping. Reducing the voltage did not help much in my case. I increase also core_freq_min and gpu_freq_min, as the old values lead to  a slow device.

The setup works on my raspi and reduced the temperature by around 4 °.

holgern, 2020-09-22:
Installing cpufrequtils
```
sudo apt-get install cpufrequtils
```
and set 
```
sudo nano /etc/default/cpufrequtils
```
to
```
GOVERNOR="ondemand"
MAX_SPEED="1000MHZ"
```
and restart it
```
sudo systemctl restart cpufrequtils
```
Reduces the temperatur by another 1 °C

Marcoblue, 2020-09-22:
@holgern But it reduces also the top-speed from 1500Mhz to 1000MHz.
Set the GOVERNOR=conservative spares also 1°C at my Pi 4. It was alredy mentioned at https://github.com/rootzoll/raspiblitz/issues/1532 

Very useful is also sudo cpufreq-info -s -m, when cpufrequtis is installed.

Marcoblue, 2020-09-22:
Changing core_freq in config.txt is not supported on the Pi 4, any change from the default will almost certainly cause a failure to boot.

It is recommended when overclocking to use the individual frequency settings (isp_freq, v3d_freq etc) rather than gpu_freq, as since it attempts to set core_freq (which cannot be changed on the Pi 4), it is not likely to have the desired effect.

Marcoblue, 2020-09-22:
In summary, this could be the configuration:

```
core_freq_min=250

h264_freq_min=100
isp_freq_min=100
v3d_freq_min=100
h264_freq=100
isp_freq=100
v3d_freq=100
# over_voltage=-1
# over_voltage_min=-2

# Disable the ACT LED.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Disable the PWR LED.
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

##turn off ethernet port LEDs
dtparam=eth_led0=4
dtparam=eth_led1=4

# The information below is intended as a template and should be adapted individually.

# Enable DRM VC4 V3D driver on top of the dispmanx display stack
# dtoverlay=vc4-fkms-v3d
# max_framebuffers=2

# Disable WIFI
# dtoverlay=disable-wifi

# Disable BT
# dtoverlay=disable-bt

# Disable HDMI
# hdmi_blanking=2

# Disable audio:
# dtparam=audio=on #comment out, it is already included
```

I personally added an 'initial_turbo=30'. This means that the turbo is on for the first 30 seconds. This ensures a faster startup.
I will do a test with these settings.

rootzoll, 2020-09-23:
Thanks for all this testing & research .. I will mark this feature idea for the v1.7 release.

laurentdebricon, 2020-10-09:
Hey, 
i am running @Marcoblue settings for 2 weeks now, temp 53°C
i put duck tape on the light of the hdd, complete stealth mode. i like it. thanks.

rootzoll, 2021-04-05:
Because UX (WebUI) is now on high prio ... I am moving this to the v1.8.1

woeisme, 2023-06-30:
was there any update for this? v.1.9.0 is running HOT on my node now.