raspiblitz/raspiblitz #1569: Refresh Option at info screen

Marcoblue, 2020-09-20:
It is possible to add a refresh button at terminal info screen? This prevents that you have to press Enter to exit and to call a new one with I.

frennkie, 2020-09-20:
Are you logged in with SSH? I use this if I want to have it auto refresh (as admin):

```
while true; do ./00infoBlitz.sh; sleep 2; done
```

Marcoblue, 2020-09-20:
I mean this window about SSH.
[image]

frennkie, 2020-09-20:
Can you use X and exit to the regular CLI and run the command above? Is that what you are looking for?

Marcoblue, 2020-09-20:
Yeah, that works well. But I suggest a setting or a checkbox for this behavior when pressing I in the main menu at SSH. I do not use an LCD or a screen.
But for me this is OK, it was just an idea to make it simpler. Ticket can be closed if not wanted.

frennkie, 2020-09-20:
@rootzoll this is up to you. We could change the current "INFO" option to have instead of "ENTER" 1-3 Options 1) return, 2) re-run 3) loop.

Marcoblue, 2020-09-20:
Yeah, that's what I had in mind. It is only an enhancement for people that dont have a LCD.

senf42, 2020-09-20:
I'd love to see the 1-3 Options!

rootzoll, 2020-09-20:
Still breaking the loop should not need a CTRL+C - I will look into a easy to end loop by default with "to end loop keep x pressed" kind of featrure.

rootzoll, 2020-09-21:
OK I merged an implementation to show the info screen in a loop. @Marcoblue & @senf42 to test this make a patch update: MENU > PATCH > PATCH MENU > PATCH ... let me know if this fits your need.

Marcoblue, 2020-09-21:
Yes for me it is okay. It is not too complicated like the 1,2,3 solution.

senf42, 2020-09-21:
Looks great to me! Thanks!

rootzoll, 2020-10-08:
Final version of this is part of the `1348lnbits` branch ... waiting for last LNbits stuff to be fixed before merge.
Tested so far - works and will be part of v1.6.1.

rootzoll, 2020-10-14:
1348lnbits merged. This looks good now - will be part of v1.6.1 - closing issue.

closed 2020-10-14 by rootzoll