raspiblitz/raspiblitz #1570: RTL backup

frennkie, 2020-09-20:
I'm getting an error on RTL backup.

RTL version 0.8.4-beta

Reading File Failed!
Error Code: -2
Error Message: ENOENT
API URL: ./api/conf/file?channel=ALL

[image]

rootzoll, 2020-09-21:
I can reproduce this bug. Looking at the RTL logs it says:
```
Sep 21 15:00:05 raspberrypi node[17746]: [1B blob data]
Sep 21 15:00:05 raspberrypi node[17746]: ERROR: Conf(207) => Reading File Failed!
```

This error gets thrown by RTL here:
https://github.com/Ride-The-Lightning/RTL/blob/1b86a8bc84fae5708ec364711fbe26b22507906b/controllers/RTLConf.js#L207

@saubyk how can we make sure reading the file correct?

kravens, 2020-09-28:
I had the same error, still want to be able to download a backup though... will try a scp command instead.

saubyk, 2020-09-28:
Pinging @shahanafarooqui

ShahanaFarooqui, 2020-09-28:
Above error only explains that the file doesn’t exist at the location. Just take the backup first and it will download the file.

You need to click on the ‘Backup All’ button to take the backup first. Once that button is clicked, the backup file will be saved in the backup folder and the file will be available for download after that. 

We have updated the error message to make it clearer. @saubyk, this is the same UX issue which we discussed two days back. We can backup channels if the file doesn’t pre-exist then download. But, you were not in favor to include these two actions in a single click.

rootzoll, 2020-09-28:
@ShahanaFarooqui I just checked ... when I first click on "Backup All" I get a "all channels backuped sucessfully" but when I then click the "Download Backup" I still get the error described above. 

I checked ... the files gets created and is readable for all users:
[image]

Marcoblue, 2020-09-28:
@ShahanaFarooqui I also clicked first to "Backup all". Still the same error. Maybe, because it runs never before here?
With Thunderhub the backup is still working.

NilByte, 2020-11-29:
Restore looks for backup file in:

Restore folder location: /home/rtl/RTL-SCB-backup-<ALIAS>/restore

Files are located in: /home/rtl/RTL-SCB-backup-<ALIAS>

ShahanaFarooqui, 2020-11-29:
This bug fix will be released with next RTL version (https://github.com/Ride-The-Lightning/RTL/issues/536). It was due to case sensitive Unix file system. It is searching for file as ‘channel-ALL’ and unable to find the file. 

@NilByte The restore feature requires the user to place file in the ‘restore’ folder but this bug is about downloading the backup file on the local machine.

rootzoll, 2021-03-25:
OK tested the backup again on new RTL 0.10.1 release included in v1.7RC1 --> not getting a error, but my backupfile is just 3 bytes @ShahanaFarooqui is there anything further to be changed/fixed?

rootzoll, 2021-09-20:
In my latest tests in preparation of the v1.7.1 of RaspiBlitz I see the backup now also on the local directory with 0 bytes:
[image]

ShahanaFarooqui, 2021-09-20:
@rootzoll The issue has been fixed with release 10.0 and I am unable to reproduce the issue anymore. Can you please check if you are getting any error in RTL console related to channel backup?

ShahanaFarooqui, 2021-09-20:
@openoms Thanks a lot for detailed testing and instructions. I was finally able to reproduce and fix the issue with your help :). 

The bug has been fixed and will be released with RTL v0.12.0.

closed 2022-05-09 by rootzoll