raspiblitz/raspiblitz #1581: [Feature Request] Add Circuit Breaker - it is to Lightning what firewalls are to the internet

openoms, 2020-09-22:
A project by Joost Jager announced in this thread: https://twitter.com/joostjgr/status/1308414374789427200
https://github.com/lightningequipment/circuitbreaker

Needs at least lnd v0.11.0

rootzoll, 2020-09-22:
Looks like a good fit for v1.7 when >=lnd 0.11.0 will be standard ... lets keep it on that milestone list.

openoms, 2020-09-28:
Starting off with some notes:
```
# create dedicate user (or just use admin to test first)
# check Go and PATH
# add symlink for .lnd and permissions for macaroons
git clone https://github.com/lightningequipment/circuitbreaker.git
cd circuitbreaker
go install
# create config file
# ~/.circuitbreaker/circuitbreaker.yaml
# include example:
# maxPendingHtlcs: 5

# create systemd service or run:
circuitbreaker --network ${chain}net
```

rootzoll, 2020-10-12:
No that lnd 0.11.0 will be a foreced update for v1.6.1 we can include Circuit Breaker already as config-script.
Will be part of v1.6.1 - use to activate: `/home/admin/config.scripts/bonus.circuitbreaker.sh on`

More improvents planned for v1.7 release: see #1654

closed 2020-10-12 by rootzoll