raspiblitz/raspiblitz #1583: Lots of issues with processes going defunct

daldwinc, 2020-09-23:
Hi Team,

I am trying to repurpose my Casa2 node, since they they no longer updating the software.

The first time I flashed the SD, I got all the way to having a synced chain. I wanted to use Tor, so I did that and then the system went down for a reboot.. or it tried to. It said it was rebooting, and then just sat at the terminal. I was unable to do `sudo shutdown -r now` or `reboot now`, or the fancy systemctl one (I forget now). The only thing that would shut it down was `sudo reboot -f now`. 

When it came back up, things were bad. bitcoind wouldn't start and was in a defunct state. If I tried to use systemctl, any command would time out. I used the collect debug logs function, but couldnt extract them with that `nc` command. Again, the process timed out.

So now I am re-flashing the SD. I SSH'd in as soon as I could, and I watched the setup log. Eventually tail stopped updating here

```
Creating config file /etc/ufw/after6.rules with new version
Created symlink /etc/systemd/system/multi-user.target.wants/ufw.service → /lib/systemd/system/ufw.service.
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for rsyslog (8.1901.0-1) ...
```

I left it for 30 minutes and there was still no update, so I logged in with another session and found:

```
admin@raspberrypi:~ $ ps auwx | grep defunct
root       394  0.0  0.0      0     0 ?        Zs   12:41   0:00 [rsyslogd] <defunct>
admin     2482  0.0  0.0      0     0 ?        Z    12:41   0:00 [sshd] <defunct>
admin     8493  0.0  0.0   7348   512 pts/0    S+   12:59   0:00 grep --color=auto defunct
```

`sudo shutdown -r now` does not appear to be working so I am going to pull the power in a moment. Since the setup process failed mid way, I think I'll just flash the SD again quickly.

Any suggestions?

Thanks!

rootzoll, 2020-09-23:
So from your logs it stops on the recovery on installing the `ufw` firewall ... which is done in `90finishSetup.sh` that is called during recovery from `_bootstrap.provision.sh` before services like Tor are installed.

So I dont think your problem is related to "activating Tor" ... its an ususal freeze where I would suspect some hardware issues. But having a Casa2 base that should from the parts be a solid setup. I am lacking a bit on ideas what this is based on. Can you reproduce this recovery error on a second try ... or does ist give you different (error) results?

daldwinc, 2020-09-23:
I am going through the bootstrapping with my fresh flash of the SD. Instead of logging in immediately, I waited about 30 minutes. I got the setup menu this time, where I set my Password A, and verified my Password C for the LND auto-unlock. It verified fine and said it needed to go for a reboot.

There is a shutting down now message on the CLI, but it never rebooted. I'm still at the prompt. There don't appear to be any defunct processes this time. Is there any info you want me to grab before I try a `sudo reboot now` then `sudo reboot -f now`, and finally pulling the power if none of those work?

rootzoll, 2020-09-23:
At least looks fine beside the not rebooting thing. Check if you can find any process that may interrupt the reboot before pulling the plug.

openoms, 2020-09-23:
This issue is known to happen when using the shimfan and LCD together.
You can use the node without an LCD, but getting a heatsink case instead of the shimfan would work even better.
Here is which type of screws you need to fit it on the X825 board: https://twitter.com/DIYnodes/status/1248587471983607808?s=19

daldwinc, 2020-09-23:
@openoms 

> This issue is known to happen when using the shimfan and LCD together.

I don't have an LCD though. The Casa nodes are headless. 

I wanted to keep poking, so I did the following...

```
admin@raspberrypi:~ $ sudo killall -u pi -HUP

admin@raspberrypi:~ $ who
admin    pts/1        2020-09-23 16:21 (172.16.40.83)

admin@raspberrypi:~ $ sudo reboot now

admin@raspberrypi:~ $ ps auwx | grep defun
root      1288  0.0  0.0      0     0 ?        Zs   14:58   0:00 [login] <defunct>
admin    16178  0.0  0.0   7348   512 pts/1    S+   17:52   0:00 grep --color=auto defun
admin    24587  0.0  0.0      0     0 ?        Z    15:14   0:00 [sshd] <defunct>
admin    24590  0.0  0.0      0     0 ?        Zs   15:14   0:00 [bash] <defunct>
root     26215  0.0  0.0      0     0 ?        Z    16:20   0:00 [su] <defunct>
root     29759  0.0  0.0      0     0 pts/1    Z    16:24   0:00 [lnd.check.sh] <defunct>

admin@raspberrypi:~ $ shutdown -f -r now
User pi is logged in on tty1.
Please retry operation after closing inhibitors and logging out other users.
Alternatively, ignore inhibitors and users with 'systemctl reboot -i'.

admin@raspberrypi:~ $ who
admin    pts/1        2020-09-23 16:21 (172.16.40.83)

admin@raspberrypi:~ $ ps auwx | grep tty1
admin    17153  0.0  0.0   7348   516 pts/1    S+   17:55   0:00 grep --color=auto tty1

admin@raspberrypi:~ $ systemctl reboot -i

admin@raspberrypi:~ $ systemctl reboot -i
Error getting authority: Error initializing authority: Error calling StartServiceByName for org.freedesktop.PolicyKit1: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.PolicyKit1': timed out (service_start_timeout=25000ms) (g-dbus-error-quark, 20)
Failed to set wall message, ignoring: Connection timed out
Failed to reboot system via logind: Connection timed 
Failed to start reboot.target: Connection timed out
See system logs and 'systemctl status reboot.target' for details.
It is possible to perform action directly, see discussion of --force --force in man:systemctl(1).
admin@raspberrypi:~ $ 
```

/power removed

daldwinc, 2020-09-23:
Interesting development....

Instead of pulling the power when the OS goes nuts, I held the ON/OFF button on the x825 board. I then used it to power back on.

[image]

```
admin@raspberrypi:~ $ bitcoin-cli -getinfo
{
  "version": 200000,
  "blocks": 649554,
  "headers": 649554,
  "verificationprogress": 0.9996127260078472,
  "timeoffset": 0,
  "connections": 0,
  "proxy": "127.0.0.1:9050",
  "difficulty": 19314656404097,
  "chain": "main",
  "relayfee": 0.00001000,
  "warnings": ""
}
```

Maybe it really doesn't like having the power pulled... we will see.

I have heatsinks coming in the mail too, just to test that theory.

openoms, 2020-09-23:
Can you just try without the shimfan? 
Don't worry about going up to 80 degrees, been there plenty and did not see fried Pi yet.

daldwinc, 2020-09-24:
@openoms Yea, no problem. I'll install the heatsinks tomorrow and get the rest of the answers about the hardware then.

It was working very reliably as the Casa. 

One additional note - the last time I rebooted it, I used the ON/OFF button on the X825 board. Held it down to power off, then pressed again to power on. Since then... almost perfectly stable.

rootzoll, 2020-09-24:
Ah OK ... the Casa2 cam with the Shimfan ... didnt knew about that. Yes the Shimfan is a known problem.

daldwinc, 2020-09-28:
Its been two days running without the Shimfan and I haven't had any issues at all. 

Thanks for your help, I'll close this.

closed 2020-09-28 by daldwinc