raspiblitz/raspiblitz #1587: Raspiblitz not booting

radar24, 2020-09-24:
my raspiblitz has been running 24/7 and i just found out that it has been frozen few days.. 
I rebooted the system and it gives me this and raspiblitz seem not working... 
does this mean i'm getting dead sd card? and i have to re setup all over again?

[image]

openoms, 2020-09-24:
Flashing a new SDcard does not mean to set up all over again, it will restore your wallets , channels and settings form the disk.

You might as well try to reboot by pulling the power and/or attach a keyboard and press CTRL+D.

The cause of this is yet unknown to me, but flashing a new SD solves it for good. Use a new card if possible.

ChuckNorrison, 2020-09-25:
Hey, if you have any issues on this, check out my way on this root password state:
https://github.com/rootzoll/raspiblitz/issues/1551

Still dont know how this happens exactly

tl;dr
lnd-rescue file works great, check out the backup docs on the raspiblitz

amiroff157, 2020-10-04:
same issue #1618

openoms, 2020-10-08:
This happens to me often after a cold restart (pulling the power) but can't say that always (also reluctant to test just for the sake of it).
Normal restarts are usually ok, but happened with that as well.
What helps:
- attach a USB keyboard and press CTRL +D as the screen says
- reflashing the SDcard solved it (takes a long time with recovering many services)

ChuckNorrison, 2020-10-12:
Happens to me 4 times. Last time right now after Backup LND rescue file via menu and shutdown.

I observed a corrupted css file on my RTL webinterface and decided to backup the lnd. This was running on v.1.6.0 default without any modifications except OpenVPN Client setup and some ufw rules to get it work.

Anything i can check to get this issue solved?

martinschultzkristensen, 2020-11-08:
> Flashing a new **SDcard does not mean to set up all over again**, it will restore your wallets , channels and settings form the disk.
>
> You might as well try to reboot by pulling the power and/or [...]

If I download a fresh & flash an image to a new SD card, will my wallet then also restore, or does it have to be a backup img-file?
Thanks for the answer.

[image]

[image]

openoms, 2020-11-09:
>If I download a fresh & flash an image to a new SD card, will my wallet then also restore, or does it have to be a backup img-file

as wrote above "it will restore your wallets, channels and settings from the disk.".

martinschultzkristensen, 2020-11-09:
> > If I download a fresh & flash an image to a new SD card, will my wallet then also restore, or does it have to be a backup img-file
> 
> as wrote above "it will restore your wallets, channels and settings from the disk.".

Thanks. I just wanted to make sure that it also goes for a new image.

muzhik, 2020-11-15:
I've had this issue as well when setting up Raspiblitz, it started when raspberry pi did final reboot after LND setup. I've tried different SD Card, went and bought USB keyboard but for some odd reason keyboard wouldn't get detected. What seems to solve it is plugging HDD into regular USB port, non 3.0. Not sure if it's related or not, but after that problem went away and Raspiblitz started node syncing.. fingers crossed this solved it.

ChuckNorrison, 2020-11-15:
long time test showed up several tweaks did the trick:
1. /etc/fstab
2. USB2.0 for external HDD
3. Disable BT and WiFI via /boot/config.txt with: 

```
dtoverlay=disable-wifi
dtoverlay=disable-bt
```

All that stuff works around the power consumption root issue on the Pi 4

muzhik, 2020-11-15:
> long time test showed up several tweaks did the trick:
> 
> 1. /etc/fstab with additional options for /mnt/hdd
>    `UUID=02fa8fa9-85ff-43b3-a9c2-ae7ee29c064c /mnt/hdd ext4 noexec,defaults,_netdev,nofail 0 2`
> 2. USB2.0 for external HDD
> 3. Disable BT and WiFI via /boot/config.txt with:
>    `dtoverlay=disable-wifi`
>    `dtoverlay=disable-bt`
> 
> All that stuff works around the power consumption root issue on the Pi 4
> 
> // RaspiBlitz v1.6.1 still seems to be affected to that. Did the tweaks after reflashed sd card and recovery installation.

I'm a newb.. can you provide step by step from SSH terminal? thx

ChuckNorrison, 2020-11-16:
as newb, you should wait for the next blitz release or get some google skills up. Just search for the filename together with "raspi pi4". Mostly `sudo nano` is your best friend :)

muzhik, 2020-11-16:
gotta start learning somewhere.. ok, i'll see if i can get this figured out.

Moonkmeander, 2020-12-28:
I have the same issue, minus the ctrl+d option. It just boots up and ends at i2c /dev entries where it ends and stay. I followed the instructions to flash the card using balena and used the image provided in the link -> put the SD card in raspberry and booted. 

However, the raspberry has been used with the default software before this. 

Any clues to what might be wrong here?

radar24, 2020-12-28:
I'm the OP and after i opened this issue, i couldn't figure this out and i was quite busy with other things so i kinda abandoned my raspiblitz. 3 months later now i finally got time to fix it so i powered on my raspiblitz and it just works now.. i wonder why but it just worked.. syncing three months data took quite a time. my raspiblitz may be needed some resting? i don't know..

nyxnor, 2021-05-17:
@radar24 Please close this issue.
Also, no issues about this after v1.7 came, seems to be fixed.

closed 2021-05-17 by rootzoll