raspiblitz/raspiblitz #1589: don't know what kind of error this is....

DrBlackross, 2020-09-24:
but when i try to link it to Zeus or Zap it gives this error before posting the qr code for the node. the wallets give me a error about not connecting each time.

fbi: no process found
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 15, in <module>
    from blitzpy import RaspiBlitzConfig, BlitzError
ImportError: cannot import name 'BlitzError' from 'blitzpy' (/usr/local/lib/python3.7/dist-packages/blitzpy/__init__.py)
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 15, in <module>
    from blitzpy import RaspiBlitzConfig, BlitzError
ImportError: cannot import name 'BlitzError' from 'blitzpy' (/usr/local/lib/python3.7/dist-packages/blitzpy/__init__.py)
# lndconnect is already installed
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 15, in <module>
    from blitzpy import RaspiBlitzConfig, BlitzError
ImportError: cannot import name 'BlitzError' from 'blitzpy' (/usr/local/lib/python3.7/dist-packages/blitzpy/__init__.py)
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 15, in <module>
    from blitzpy import RaspiBlitzConfig, BlitzError
ImportError: cannot import name 'BlitzError' from 'blitzpy' (/usr/local/lib/python3.7/dist-packages/blitzpy/__init__.py)
Wrote QR Code to file "lndconnect-qr.png"
# OK - file exists: /home/admin/lndconnect-qr.png

zap tells me Hostname verification failed, it does not match hostname provided in the cert. (?)

zeus says its there and active, but clicking on it says "error: stream was reset: internal error", i did have raspiblitz set it up to be used only on the local lan.... do i have to enable the local lan somewhere in the configs in raspiblitz.

BTW, its sync'd does receive btc to fund Lnd, but havent seen anything really take off yet.

DrBlackross, 2020-09-24:
also when scanning qr or pasting lndconnect urls with both lan and internet ips i get the same errors

rootzoll, 2020-09-28:
@DrBlackross did you trued with the lastest scripts: MENU > UPDATE > PATCH > PATCHMENU > PATCH

DrBlackross, 2020-09-29:
no.... i did it when it was a beta (would have to check my post here), i still can't run update and pull it from github, right. i have to reflash the sdcard to do that update?

rootzoll, 2020-10-05:
If you have a final main release like "v1.6" thane you can use the "MENU > UPDATE > PATCH > PATCHMENU > PATCH" to update even to a minor relase like "v1.6.1" or any smaller upadtes. If you had a Release Candidate or you want to make a jump to a niew main verison ... you would need go thru the sd card update.

grnqrtr, 2021-01-29:
I've been running on @openoms [64 bit image](https://github.com/rootzoll/raspiblitz/issues/1199#issuecomment-737305510) synced to the 1.6 branch for a week now and I'm getting similar errors related to the subscriptions. My first problem was that the subscriptions were pointing to an old IP, so I thought I would just cancel them and recreate them. But when trying to cancel them I run into this error too.

```
admin@raspberrypi:~ $ python /home/admin/config.scripts/blitz.subscriptions.ip2tor.py subscription-cancel fdff55b4-fbb5-48c6-afb5-11d0a9b9fd8f
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 15, in <module>
    from blitzpy import RaspiBlitzConfig, BlitzError
ImportError: cannot import name 'BlitzError' from 'blitzpy' (/usr/local/lib/python3.7/dist-packages/blitzpy/__init__.py)

admin@raspberrypi:~ $ python /home/admin/config.scripts/blitz.subscriptions.letsencrypt.py subscription-cancel grnqrtr.duckdns.org
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.letsencrypt.py", line 14, in <module>
    from blitzpy import RaspiBlitzConfig,BlitzError
ImportError: cannot import name 'BlitzError' from 'blitzpy' (/usr/local/lib/python3.7/dist-packages/blitzpy/__init__.py)
```

rootzoll, 2021-01-29:
@grnqrtr can you go to terminal and run: `/home/admin/XXsyncScripts.sh -justinstall` .. and check if you see any error nera the line `# BlitzPy changed --> UPDATING to Version 0.3.0`

grnqrtr, 2021-01-29:
I don't see any errors, but it looks like it's updating to `Version 0.2.0` not `Version 0.3.0`
```
admin@raspberrypi:~ $ /home/admin/XXsyncScripts.sh -justinstall
# *** SYNCING RASPIBLITZ CODE WITH GITHUB ***
# This is for developing on your RaspiBlitz.
# THIS IS NOT THE REGULAR UPDATE MECHANISM
# and can lead to dirty state of your scripts.
# REPO ----> origin     https://github.com/rootzoll/raspiblitz.git (push)
# BRANCH --> v1.6
# ******************************************
Already up to date.
# Cleaning assets .. 
# COPYING from GIT-Directory to /home/admin/
# ******************************************
# Checking if the content of BlitzPy changed ..
# checkSumBlitzPyBefore = af63c7d6fe324493cf759dd1711eef06  -
# checkSumBlitzPyAfter  = af63c7d6fe324493cf759dd1711eef06  -
# BlitzPy changed --> UPDATING to Version 0.2.0
# ******************************************
# OK - shell scripts and assets are synced
# Reboot recommended
```

NilByte, 2021-01-29:
I had this experience when pairing Zap mobile wallet on 1.6.2.

https://github.com/rootzoll/raspiblitz/issues/766#issuecomment-750619167

rootzoll, 2021-03-19:
This should be fixed for the v1.7 release ... test this with the mobile wallets again.

closed 2021-04-15 by rootzoll