raspiblitz/raspiblitz #1590: SSH access via Tor hidden service

woeisme, 2020-09-25:
Would be nice to allow SSH access via Tor hidden service to be able to administer the Raspiblitz without having to worry about NAT. 
Should just be another hidden service entry right?

woeisme, 2020-09-25:
actually got this working in about 10 minutes with the following process:
(run all modifications as root via sudo -s)

edit /etc/tor/torrc 

> HiddenServiceDir /mnt/hdd/tor/sshd_2222/
HiddenServiceVersion 3
HiddenServicePort 2222 127.0.0.1:22

> systemctl restart tor

>cat /mnt/hdd/tor/sshd_2222/hostname

From then on I used Torbot and Connectbot on an Android phone to connect to the admin menu.
I had previously imported a public key from Connectbot to the Raspiblitz.

woeisme, 2020-09-27:
Could someone nice add a feature / enhancement label to this Issue ?

openoms, 2020-09-28:
this might make a good FAQ entry but I don't think that offering ssh access as a Hidden Service is a good idea from the security point of view.
Also it needs Tor on connecting machine so adds further complexity to the setup.

This can be adapted to be more specifix to the RaspiBlitz: https://github.com/openoms/joininbox/blob/master/FAQ.md#ssh-through-tor-from-linux

rootzoll, 2020-09-28:
Just adding a reminder to check if this setting survives a reset/recovery for v1.7 version.

openoms, 2020-09-28:
I don't think it would since the Hidden Service is defined on the SDcard in /etc/tor/torrc and not restored by any script.

It would also be a security issue when during the first boot ssh can be accessed remotely with the default password (`raspiblitz`).

I don't think that persisting this setting would be a good idea.

rootzoll, 2020-09-28:
OK makes sense - closing issue.

closed 2020-09-28 by rootzoll