raspiblitz/raspiblitz #1591: Idea: Activate possibility for RBF (Replace by Fee)

Marcoblue, 2020-09-25:
In order to be able to use RBF (Replace by Fee), this function must be activated before the (first) transaction is done. 

For this purpose, walletrbf=1 must be added to the bitcoin.conf.

Does it make sense to do this as default?

libaohua-qb, 2020-09-28:
in my view, RBF is default available is you do nothing

rootzoll, 2020-09-28:
@openoms is the RBF active has any side effects on e.g. Joinmarket?

openoms, 2020-09-28:
what is the usecase here?
@Marcoblue are you planning to contruct transactions with bitcoin-cli?
Because this setting would not have effect on anything else.
Every wallet (including joinmarket) is setting their own RBF flag.

Marcoblue, 2020-09-28:
@libaohua-qb At default RBF is inactive. I cannot see where it is switched on.
It can be switched on during start with startoption -walletrbf or thru bitcoin.conf.

@openoms I had a problem with a channel openeing thru RTL. So you mean RBF is possible there over the console?
Maybe this:
bitcoin-cli -conf=/mnt/blockchains/bitcoin/bitcoin.conf bumpfee txid

Can't test it now, because my transaction worked already after 3 days.

openoms, 2020-09-28:
See my answer here: https://github.com/rootzoll/raspiblitz/issues/640#issuecomment-699898165

I suggest to close this since the request is based on a misunderstanding.

closed 2020-09-28 by Marcoblue