raspiblitz/raspiblitz #1596: Idea: Add minchansize to lnd.conf

Marcoblue, 2020-09-27:
Can we add the follwing to the standard lnd.conf? So it's much easyer to config theses option later.

; The smallest channel size (in satoshis) that we should accept. Incoming
; channels smaller than this will be rejected, default value 20000.
; minchansize=250000

Maybe there are more options that would make sense here?

rootzoll, 2020-09-28:
I would not use "out-commented" config-settings because this is harder to handle from the scripts to switch on/off/change then. But it can make sense to set the "default values" so that the user on manual edit already can see possible edits.

I will add this to the v1.7 milestone so we can collect more default-values that make sense to put in here.

Marcoblue, 2020-09-28:
@rootzoll Good point.

So far I have the following:
```
# The smallest channel size (in satoshis) that we should accept. Incoming
# channels smaller than this will be rejected, default value 20000.
minchansize=1000000
```

```
# The smallest channel that the autopilot agent should create (default: 20000)
autopilot.minchansize=1000000
```

```
# The largest channel that the autopilot agent should create (default: 16777215)
autopilot.maxchansize=5000000
```

```
# The maximum number of incoming pending channels permitted per peer, default value 5.
maxpendingchannels=1
```

openoms, 2020-10-01:
The https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf has been extended largely recently, so there are many option to look at.

Generally the use of the autopilot is discouraged for now so would rather avoid including default config options about it.

openoms, 2020-10-01:
It could be more like a FAQ entry rather then put explanations in the lnd.conf itself. Youd might be interested in this sample of (https://github.com/openoms/lightning-node-management/blob/master/lnd.conf.example mine) with some extra options I use in practice like: 
```
[Application Options]
minchansize=2000000
protocol.wumbo-channels=1 
bitcoin.feerate=2200

[Bitcoind]
bitcoind.estimatemode=CONSERVATIVE
```

Marcoblue, 2020-10-22:
Attention, the entry
maxpendingchannels=5
no longer works since v1.6.1.

I don't know why.

openoms, 2020-10-23:
> Attention, the entry
> maxpendingchannels=5
> no longer works since v1.6.1.

What do you mean by no longer works?
Did you see more than 5 pending channels while the setting was active?

Marcoblue, 2020-10-23:
> What do you mean by no longer works?
> Did you see more than 5 pending channels from one peer while the setting was active?
> https://github.com/lightningnetwork/lnd/blob/426db548ddac7031fdea5156bd14d0a4218e1bf1/sample-lnd.conf#L213

No, lnd says at start, that this is an unknown command. The minchansize=xxxx is still in and work.

openoms, 2020-10-24:
So this could be a change in lnd itself with v0.11.1, not made by raspiblitz.
Can you catch the log entry?

Did not see it with:
`sudo cat  /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | grep "maxpendingchannels"`

Marcoblue, 2020-10-25:
Sorry, I don't have a test-node, so I don#t want to test it again at the moment. Maybe there was a fault with my lnd.conf, so if nobody else have this problem, just ignore it.

openoms, 2021-03-18:
I'd suggest not to add a `minchansize` by default until at least we can have a separate config for private channels (which would make sense to be smaller).

See the feature request in LND: https://github.com/lightningnetwork/lnd/issues/5107

openoms, 2021-06-23:
@maestro32 see https://openoms.gitbook.io/lightning-node-management/advanced-tools/lnd.conf and the links.

`bitcoin.basefee` is in msats cannot use decimals. 1000 msat = 1 sat

openoms, 2021-11-08:
`michansize` should not be dicatated by the raspiblitz devs. Closing