raspiblitz/raspiblitz #1604: onion socket 9735 is closed

holgern, 2020-09-30:
I had the problem that the tor socket 9735 was not reachable. I checked it by installing
```
sudo apt-get install proxychains4 nmap
```
on a server and checked the port with:
```
proxychains4 nmap -Pn -p 9735 -sT -v my_onion_addr.onion
```
The command returned a 
```
PORT     STATE SERVICE
9735/tcp closed unknown
```

Only a 
```
./config.scripts/internet.tor.sh off
```
reboot
```
./config.scripts/internet.tor.sh on
```
reboot
did repair the port.
Now I see:
```
PORT     STATE SERVICE
9735/tcp open  unknown
```

It is also possible to do the proxychains4 command on the raspi itself.

holgern, 2020-09-30:
After testing, I found out that the port 9735 is closed every time tor is restarted due to installing something on the raspiblitz.

holgern, 2020-09-30:
I found a fix:
lnd must be stopped before tor is restarted

```
sudo systemctl stop lnd
sudo systemctl restart tor@default
sudo systemctl start lnd
```

openoms, 2020-09-30:
you can get the LND HS address with:
`echo "$(lncli getinfo)" | grep "uris" -A 1 | tr -d '\n' | cut -d '"' -f4 | cut -d @ -f2 | cut -d : -f1`
so testing with:
`proxychains4 nmap -Pn -p 9735 -sT -v $(echo "$(lncli getinfo)" | grep "uris" -A 1 | tr -d '\n' | cut -d '"' -f4 | cut -d @ -f2 | cut -d : -f1)`

It always gives me:
```
PORT     STATE  SERVICE
9735/tcp closed unknown
```

I often find that if some of my peers are disconnecting restarting tor (and often lnd too) helps.
It is sufficient to do:
```
sudo systemctl restart tor
sudo systemctl restart lnd
```

holgern, 2020-09-30:
I'm getting a denied when I only restart tor.
```
:9735 <--denied
```
This is the problem, when installing something on the raspiblitz. I don't know if this happens only to my installation, or if this is a  general problem.
I'm on the current patched github for branch 1.6.1, so I have included the PR #1571 (I did all the necessary changes on my system)

openoms, 2020-09-30:
Thank you for documenting this, it is an interesting find, don't know what to make of it yet, but certainly shows what could have been  the problem with a problematic Tor node of mine which has been hardly restarted ever.
While testing now I could only get 
```
PORT     STATE SERVICE
9735/tcp open  unknown
```
after restarting the whole node.
@frennkie @rootzoll there could be a new way test for the occasional IP2Tor problems as well. 
The internal port of the Hidden Service might be closed in some occasions when restarting Tor?

rootzoll, 2020-09-30:
Hmm interesting .. let me know if there is anything to add to the v1.6.1 on short notice. Otherwise I mark this for v1.7 to take a deeper look.

rootzoll, 2021-03-26:
@openoms is this still relevant?

openoms, 2021-03-26:
don't think so, a lot has been changed. Could reopenif comes up again.

rootzoll, 2021-03-26:
So closing for now - please reopen is something shows up.

openoms, 2021-04-03:
I can reproduce this still, suggest to introduce a manual check for it in the CONNECT menu https://github.com/rootzoll/raspiblitz/issues/2090

This causes long channel downtimes and problem using Pool so will take up with LND.

openoms, 2021-04-06:
Added monitoring for this problem in #2137.

As for now it seems there is no other way to mitigate this only to restart LND with Tor as well. 
This would add unnecessary downtime and more handling of the passwordC.

One partila solution could be to avoid manula restarts at least is to set up a separate Tor instance for LND, so it is not restarted by the `internet.hiddenservice.sh` when new .onion services are added or removed.

How to create muliple Tor instances: https://coderwall.com/p/01_gpg/multiple-instance-of-tor
LND Tor docs: https://github.com/lightningnetwork/lnd/blob/master/docs/configuring_tor.md

openoms, 2021-04-06:
Can test the dedicated tor instance (in #2148) with these:
Leaned from:  https://www.torservers.net/wiki/setup/server#multiple_tor_processes

```
    # lnd-tor instance
    NODENAME="lnd"
    SOCKSPORT=9070
    CONTROLPORT=$((SOCKSPORT+1))
    echo "# Creating a dedicated Tor instance for $NODENAME"
    sudo tor-instance-create $NODENAME

    echo "# Make sure the user bitcoin is in the _tor-$NODENAME group"
    sudo usermod -a -G _tor-$NODENAME bitcoin

  # create tor data directory if it not exist
  if [ ! -d "/mnt/hdd/tor-$NODENAME" ]; then
    echo "# - creating tor data directory"
    sudo mkdir -p /mnt/hdd/tor-$NODENAME
    sudo mkdir -p /mnt/hdd/tor-$NODENAME/sys
  else
    echo "# - /mnt/hdd/tor-$NODENAME data directory exists"
  fi
  # make sure its the correct owner
  sudo chmod -R 700 /mnt/hdd/tor-$NODENAME
  sudo chown -R _tor-$NODENAME:_tor-$NODENAME /mnt/hdd/tor-$NODENAME

  echo "
DataDirectory /mnt/hdd/tor-$NODENAME/sys
PidFile /mnt/hdd/tor-$NODENAME/sys/tor.pid
SocksPort $SOCKSPORT
ControlPort $CONTROLPORT
CookieAuthentication 1
CookieAuthFileGroupReadable 1
SafeLogging 0
Log notice stdout
Log notice file /mnt/hdd/tor-$NODENAME/notice.log
Log info file /mnt/hdd/tor-$NODENAME/info.log
" | sudo tee /etc/tor/instances/$NODENAME/torrc
    sudo chmod 644 /etc/tor/instances/$NODENAME/torrc

  sudo mkdir -p /etc/systemd/system/tor@$NODENAME.service.d
  sudo tee /etc/systemd/system/tor@$NODENAME.service.d/raspiblitz.conf >/dev/null <<EOF
    # DO NOT EDIT! This file is generated by raspiblitz and will be overwritten
[Service]
ReadWriteDirectories=-/mnt/hdd/tor-$NODENAME
[Unit]
After=network.target nss-lookup.target mnt-hdd.mount
EOF

  echo "Setup logrotate"
  # add logrotate config for modified Tor dir on ext. disk
  sudo tee /etc/logrotate.d/raspiblitz-tor-$NODENAME >/dev/null <<EOF
/mnt/hdd/tor-$NODENAME/*log {
        daily
        rotate 5
        compress
        delaycompress
        missingok
        notifempty
        create 0640 debian-tor debian-tor
        sharedscripts
        postrotate
                if invoke-rc.d tor status > /dev/null; then
                        invoke-rc.d tor reload > /dev/null
                fi
        endscript
}
EOF
    sudo systemctl daemon-reload
    sudo systemctl enable tor@$NODENAME
    sudo systemctl start tor@$NODENAME
    
    # config for lnd self-test
    echo "
strict_chain
proxy_dns 
tcp_read_time_out 15000
tcp_connect_time_out 8000
[ProxyList]
socks5 	127.0.0.1 $SOCKSPORT
" | sudo tee /etc/proxychains.$NODENAME.conf

    # modify LND service
    echo "# Make sure LND is disabled"
    sudo systemctl disable lnd 2>/dev/null

    echo "# Editing /etc/systemd/system/lnd.service"
    sudo sed -i "s/^ExecStart=\/usr\/local\/bin\/lnd.*\
/ExecStart=\/usr\/local\/bin\/lnd --tor\.active --tor\.streamisolation --tor\.v3 --tor\.socks=$SOCKSPORT --tor\.control=$CONTROLPORT --listen=127\.0\.0\.1\:9735 \${lndExtraParameter}/g" \
    /etc/systemd/system/lnd.service

    echo "# Enable LND again"
    sudo systemctl enable lnd
    echo "# OK"
    echo 
```

rootzoll, 2021-04-08:
OK PR #2148 merged - will be part of v1.7RC2 ... up for final testing.

nyxnor, 2021-07-09:
Ok, now there is 2 instances.
The problem was happening because Tor was restarting, but guess what, it doesnt need to restart when editing the torrc.
https://github.com/rootzoll/raspiblitz/issues/2409#issuecomment-876340098
Adding another instance gives some complication handling bridges that needs a workaround, deleting and restoring lines.
Can this be tested on a v1.6.3 Raspiblitz with reloading tor instead of restarting? Just to see if this issue could be fixed without another instance.
I think adding one reload line keep it more simple than setting a new tor instance although different tor instances could have it place in the future.

@holgern Are you still on v.1.6.3?

openoms, 2021-07-12:
`_tor-lnd` is set up automaticelly by tor when the new ibstnce ia created same as `debian-tor` is the default.

The main question is: 
can we activate a Hidden Service without needing to restart and unlock LND?

Can test on v1.7.0 as well. Try reloading (not sure what you refer yet) instead of restarting and test if the lnd Hidden Service on 9735 is still connectable.

nyxnor, 2021-07-13:
This is the state now, I tried restarting tor@lnd, tor@default, lnd and nothing changed. Also, it is referring to port 9050, it should goes for the 9070 for tor@lnd.
```
proxychains4 nmap -p9735 $(lncli getinfo|jq -c .uris|cut -d@ -f2|cut -d: -f1)
```
>[proxychains] config file found: /etc/proxychains4.conf
[proxychains] preloading /usr/lib/aarch64-linux-gnu/libproxychains.so.4
[proxychains] DLL init: proxychains-ng 4.13
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  []:80 <--denied
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  []:9735 <--denied

>PORT     STATE  SERVICE
9735/tcp closed unknown

openoms, 2021-07-13:
OK so reloading with `sudo pkill -sighup tor` did not break the LND port as `sudo systemctl restart tor@lnd` did so this is awesome: 

admin@bob:~ $ proxychains4 nmap -p9735 $(lncli getinfo|jq -c .uris|cut -d@ -f2|cut -d: -f1)
[proxychains] config file found: /etc/proxychains4.conf
[proxychains] preloading /usr/lib/aarch64-linux-gnu/libproxychains.so.4
[proxychains] DLL init: proxychains-ng 4.13
Starting Nmap 7.70 ( https://nmap.org ) at 2021-07-13 14:10 BST
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion:80 <--denied
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion:9735  ...  OK
Nmap scan report for a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion (224.0.0.1)
Host is up (2.5s latency).
rDNS record for 224.0.0.1: all-systems.mcast.net

PORT     STATE SERVICE
9735/tcp open  unknown

Nmap done: 1 IP address (1 host up) scanned in 3.64 seconds

admin@bob:~ $ sudo pkill -sighup tor

admin@bob:~ $ proxychains4 nmap -p9735 $(lncli getinfo|jq -c .uris|cut -d@ -f2|cut -d: -f1)
[proxychains] config file found: /etc/proxychains4.conf
[proxychains] preloading /usr/lib/aarch64-linux-gnu/libproxychains.so.4
[proxychains] DLL init: proxychains-ng 4.13
Starting Nmap 7.70 ( https://nmap.org ) at 2021-07-13 14:11 BST
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion:80 <--denied
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion:9735  ...  OK
Nmap scan report for a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion (224.0.0.1)
Host is up (1.2s latency).
rDNS record for 224.0.0.1: all-systems.mcast.net

PORT     STATE SERVICE
9735/tcp open  unknown

Nmap done: 1 IP address (1 host up) scanned in 1.79 seconds

admin@bob:~ $ sudo systemctl restart tor@lnd

admin@bob:~ $ proxychains4 nmap -p9735 $(lncli getinfo|jq -c .uris|cut -d@ -f2|cut -d: -f1)
[proxychains] config file found: /etc/proxychains4.conf
[proxychains] preloading /usr/lib/aarch64-linux-gnu/libproxychains.so.4
[proxychains] DLL init: proxychains-ng 4.13
Starting Nmap 7.70 ( https://nmap.org ) at 2021-07-13 14:11 BST
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion:80 <--denied
[proxychains] Strict chain  ...  127.0.0.1:9050  ...  a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion:9735 <--denied
Nmap scan report for a4w37h23fiium7anpy27ziflcnwb22l6okwtenjde2xmmf4icsawnuqd.onion (224.0.0.1)
Host is up (3.8s latency).
rDNS record for 224.0.0.1: all-systems.mcast.net

PORT     STATE  SERVICE
9735/tcp closed unknown

Nmap done: 1 IP address (1 host up) scanned in 5.49 seconds
```

@nyxnor your node should start with: 
```
PORT     STATE SERVICE
9735/tcp open  unknown
```
try restarting lnd last or a full reboot.
When the port is closed the node is not connectable leading to offline channels, force closes and general unreliability.

Next we can test if using `sudo pkill -sighup tor` only creates the new HiddenServices from the torrc and if yes it shouldreplace the `sudo systemctl restart tor@default` in `internet.hiddenservice.sh` (and other places) and we can try get rid of the `tor@lnd` instance.

nyxnor, 2021-07-13:
>your node should start with:
Will need a full reboot, already restarted lnd and tor@lnd a few times now.

>Next we can test if using sudo pkill -sighup tor only creates the new HiddenServices from the torrc and if yes it shouldreplace the sudo systemctl restart tor@default in internet.hiddenservice.sh (and other places) and we can try get rid of the tor@lnd instance.

Yes it can and I am using that on my day to day use. Any config lines for the torrc that is added or replaced will be used after reloading tor, adding or removing hidden services/client authorization.

xanoni, 2021-08-23:
By the way, I think the "official" way to do this is `systemctl reload tor` (instead of restart)

nyxnor, 2021-08-23:
@xanoni I tested different methods to reload and documented the differences here https://github.com/rootzoll/raspiblitz/issues/2409#issuecomment-876340098

xanoni, 2021-08-23:
OK. I just checked and the effect should be exactly the same though ... see `/etc/init.d/tor`:

```bash
211     if start-stop-daemon --stop --signal 1 --quiet --pidfile $TORPID --exec $DAEMON
212     then
213         log_action_end_msg 0
214     elif kill -0 $pid 2>/dev/null; then
215         log_action_end_msg 1 "Is $pid not $NAME?  Is $DAEMON a different binary now?"
216         exit 1
217     else
218         log_action_end_msg 1 "$DAEMON died: process $pid not running; or permission denied"
219         exit 1
```

I understand line 211 just sends a `HUP` (signal = 1).

openoms, 2021-08-23:
ok @xanoni thanks I have tested this with c-lightning and `sudo systemctl reload tor` does not close the port. The question is if it creates the new hidden service?

See the sequence:
```
admin@raspberrypi:~ $ proxychains4 nmap -p19736 anibj3bogldf26ftlj2ap7brjo6drzoecgrc57fw3inmxdy3h7pa4nad.onion

PORT      STATE SERVICE
19736/tcp open  unknown

admin@raspberrypi:~ $ sudo systemctl reload tor
admin@raspberrypi:~ $ proxychains4 nmap -p19736 anibj3bogldf26ftlj2ap7brjo6drzoecgrc57fw3inmxdy3h7pa4nad.onion

PORT      STATE SERVICE
19736/tcp open  unknown

admin@raspberrypi:~ $ sudo systemctl restart tor
admin@raspberrypi:~ $ proxychains4 nmap -p19736 anibj3bogldf26ftlj2ap7brjo6drzoecgrc57fw3inmxdy3h7pa4nad.onion

PORT      STATE  SERVICE
19736/tcp closed unknown

admin@raspberrypi:~ $ sudo systemctl restart tlightningd
admin@raspberrypi:~ $ proxychains4 nmap -p19736 anibj3bogldf26ftlj2ap7brjo6drzoecgrc57fw3inmxdy3h7pa4nad.onion

PORT      STATE SERVICE
19736/tcp open  unknown
```

openoms, 2021-08-23:
Great, the .onion is created too. See the test adding to torrc:
```
admin@raspberrypi:~ $ sudo nano /etc/tor/torrc 
admin@raspberrypi:~ $ sudo cat /mnt/hdd/tor/test/hostname
cat: /mnt/hdd/tor/test/hostname: No such file or directory
admin@raspberrypi:~ $ sudo systemctl reload tor
admin@raspberrypi:~ $ sudo cat /mnt/hdd/tor/test/hostname
fv56gfrfrus6tuzl6upo6degrexs6ip5zm2eqhon7ctgojcihdo2taad.onion
```

@nyxnor seems that  
`sudo systemctl reload tor`
has the same effect as
`sudo pkill -sighup tor`.

I think we should use the cleaner method:
`sudo systemctl reload tor`
What do you think, should it be changed in: #2419 ?

xanoni, 2021-08-23:
Yes it really shouldn't make a difference. It's just a syntax difference, unless Tor changes how it triggers config reloads (unlikely, given HUP is very standard). Using `systemctl` is a bit more universal, though, because should something change, then the Debian team would update the `reload` command for us :-)

nyxnor, 2021-08-24:
@openoms 
> As they seem to be equivalent in function I see no reason to change this in #2419 but good to be aware.

It is here commented https://github.com/nyxnor/raspiblitz/blob/tor-patch/home.admin/config.scripts/tor.functions.lib#L186, will update.

rootzoll, 2021-09-20:
@openoms can you summarize the state of this issue? is there something for the v1.7.1 needed?

nyxnor, 2021-09-20:
@rootzoll he already did https://github.com/rootzoll/raspiblitz/commit/4874d7dd15c936e3bbd3c5cc8a0ceeba1c5abf69#diff-eb88b0a6c4c9512eeda4cdba40f682e1cb07504afd895dc465191da40ddc09a4

openoms, 2021-09-20:
@rootzoll yes basically the separate `lnd@tor` instance now removed because changing 
`sudo systemctl restart tor@default`
to
`sudo systemctl reload tor@default`
solves the issue as tested.

So changes are:
* reloading Tor instead of restarting on `torrc` changes
* using only one tor instance now: `tor@default`

The lnd tor streamisolation remains active which opens new circuits for all the lnd connections 
There is no similar (streamisolation) option for C-lightning so introducing a dedicated Tor instance could be considered.

openoms, 2021-09-20:
Since we are not restarting Tor any more this issue can be closed.

closed 2021-09-20 by rootzoll

xanoni, 2021-09-21:
Sorry to be contrarian here, but I actually set up a separate Tor instance for my `lnd` after seeing it in this repo and I'm still using it :) I seems it helped me fix some channel stability issues.

However, I think this is because I have a "shit ton" of other things using Tor on the same Pi. Probably not an issue for normal RaspiBlitz users. Just FYI.

openoms, 2021-09-21:
@xanoni see this https://github.com/rootzoll/raspiblitz/issues/2560 for further discussion about the possible benefits of reintroducing separate Tor instances for the Lightning nodes.