raspiblitz/raspiblitz #1610: Joinmarket QTGUI Windows Instructions

woeisme, 2020-10-02:
I believe the instructions under the 'QTGUI'->'WINDOWS need to be updated to reflect the proper location of the python joinmarket-qt.py executable. Currently the instructions read:

> In the terminal type:

python joinmarket-qt.py

But the location of the joinmarkey-qt.py file is actually located in joinmarket-clientserver/scripts/joinmarket-qt.py
The instructions should be updated to reflect this. Or joinmarket-qt.py should be aliases or added to PATH.

openoms, 2020-10-02:
Thanks, this can be fixed in https://github.com/openoms/joininbox

openoms, 2020-10-02:
for now went for the safest option to include the path.
Will be part of the next release once the joininbox version is bumped.
Related: https://github.com/rootzoll/raspiblitz/issues/1488

closed 2020-10-06 by rootzoll