raspiblitz/raspiblitz #1618: Stuck at "i2c /dev entries driver"

amiroff157, 2020-10-04:
My raspiblitz is refusing to boot up after working perfectly for two months. I tried to reinstall a new copy on a NEW SD card and still having the same issue. It was able to only connect to the network via ethernet at theinitial install process but then when it rebooted it just hangs and does nothing. No network. No SSH. Nothing.

This is a log from the 1st install before it shuts down and never boot up again"

```
OK - a restart is needed: sudo shutdown -r now
Provisioning UPS - not active
Provisioning LNbits - keep default
Provisioning JoinMarket - keep default
Provisioning Specter - keep default
Provisioning Faraday - keep default
Provisioning Balance of Satoshis - keep default
Provisioning ThunderHub - keep default
Provisioning letsencrypt - keep default
Provisioning kindle-display - keep default
Provisioning Stacking Sats Kraken - keep default
Running the custom install script ..
'/mnt/hdd/app-data/custom-installs.sh' -> '/home/admin/custom-install.sh'
There are no custom user installs so far.
Done

*** Replay backup of LND conf/tls
*** Repair Bitcioin Conf (if needed)
Aligning lnd.conf & bitcoin.conf
Not setting local network hostname
Prepare fstab for permanent data drive mounting ..
datadisk --> sda
DONE - Give raspi some cool off time after hard building .... 5 secs sleep
END Provisioning
rebooting
shutdown in 1min

Broadcast message from root@raspberrypi (Sun 2020-10-04 01:06:05 BST):

The system is going down for reboot at Sun 2020-10-04 01:07:05 BST!

```

Now it just powers up and shows "i2c /dev entries driver" as a last message on the HDMI screen.

rootzoll, 2020-10-04:
There some LCD setups that hang that way ... but usually its just the LCD output and in the background system is still running. If you run into "not able to ssh in" check if you using the Shimfan and remove it. Also some HDDs make problems - connect them to the USB2 ports.

amiroff157, 2020-10-04:
> There some LCD setups that hang that way ... but usually its just the LCD output and in the background system is still running. If you run into "not able to ssh in" check if you using the Shimfan and remove it. Also some HDDs make problems - connect them to the USB2 ports.

thanks for the quick assist! were there any breaking changes in the last update? because I was always connecting my USB 3.0 HDD to the right USB 3.0 port.

amiroff157, 2020-10-04:
tried connecting the HDD to the USB2 port, still behaving the same. It's not connecting to ethernet, only on the install before shutdown it did, now it's just the RPI running and doing nothing

rootzoll, 2020-10-04:
Ah OK .. when it worked before good then you wil probaly not have that bad HDD casing that just works on USB2 and also no Shimfan.

Try to make a fresh sd card and start it without LCD - by not connecting the LCD screen and turn it to HDMI output like this with placing a file on the boot section before first start: https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#can-i-run-the-raspiblitz-without-a-displaylcd

amiroff157, 2020-10-04:
> Ah OK .. when it worked before good then you wil probaly not have that bad HDD casing that just works on USB2 and also no Shimfan.
>
> Try to make a fresh sd card and start it without LCD - by not [...]

never had the LCD on just a cooling fan enclosure. Did the HDMI file method anyway and still same result. Also conencted the HDD to USB2...

openoms, 2020-10-05:
This might be a known issue when the RaspiBlitz is not booting with a HDMI cable connected.
Try booting without HDMI and connect via ssh.
You can connect the HDMI screen later and use the menu shortcut command:
`hdmi`  
to run:  
`sudo /home/admin/config.scripts/blitz.lcd.sh hdmi on`

amiroff157, 2020-10-05:
> This might be a known issue when the RaspiBlitz is not booting with a HDMI cable connected.
> Try booting without HDMI and connect via ssh.
> You can connect the HDMI screen later and use the menu shortcut command:
> `hdmi`
> to run:
> `sudo /home/admin/config.scripts/blitz.lcd.sh hdmi on`

no HDMI cable is connected just RPi, Ethernet cable and HDD.

P.S. It' hangs at a point where  no SSH and no ethernet are working

amiroff157, 2020-10-05:
Update:

Okay upon installing raspiblitz another time I get this error at the end of the install:

**$ sudo tail -n1000 -f raspiblitz.log**:

```
/home/admin/config.scripts/blitz.touchscreen.sh: line 129: /mnt/hdd/raspiblitz.conf: Input/output error
sed: can't read /mnt/hdd/raspiblitz.conf: Input/output error
OK - a restart is needed: sudo shutdown -r now
Provisioning UPS - not active
Provisioning LNbits - keep default
Provisioning JoinMarket - keep default
...
```

rootzoll, 2020-10-06:
> /home/admin/config.scripts/blitz.touchscreen.sh: line 129: /mnt/hdd/raspiblitz.conf: Input/output error
sed: can't read /mnt/hdd/raspiblitz.conf: Input/output error

This does not look good. You seem to have I/O errors on your HDD. I would call (if possible) `repair` and run `BACKUP-LND` to download the rescue-file to your laptop - so that you have your LND funds+channels safe. After that your RaspiBlitz will turn off. Make sure that rescue-file is a TAR file you can unzip on your laptop and also contains the `channel.backup` file.

Then you can give the HDD another chance by fresh formatting it on another computer and then start with a fresh sd card - setting up a complete fresh RaspiBlitz (with resyncing the blochchain) and the choose "OLD" LND wallet and reupload that the rescuefile.

amiroff157, 2020-10-07:
> This does not look good. You seem to have I/O errors on your HDD. I would call (if possible) `repair` and run `BACKUP-LND` to download the rescue-file to your laptop - so that you have your LND [...]

my HDD was untouched and the raspiblitz wasn't working for around a month before this happened. My instinct was to just redo everything from scratch but I was more bugged to know how this happened and how a fresh install of raspiblitz on a new SD card didn't fix the issue...

P.S don't need to backup anything as I didn't have any relevant data there so it's fine to just delete all and skip recovering, I was just worried about downloading the blockchain again...

rootzoll, 2020-10-07:
I/O erros are hard to pin down. A common source for I/O errors is the power supply. If on writing the power is weak it can corrupt the HDD data.

Aractor, 2020-10-12:
I'm experiencing this same issue, trying to to get a new Raspiblitz install running.

I've tried booting from the pre-built image, and building an image on top of a Rasbian install, both times I end up stuck with the boot hanging on "i2c /dev entries driver". This happens with or without my LCD hat connected, and with & without HDMI connected.

amiroff157, 2020-10-12:
> I/O erros are hard to pin down. A common source for I/O errors is the power supply. If on writing the power is weak it can corrupt the HDD data.

what guarantees that my future raspiblitz install doesn't corrupt my blockchain and have to re-download it all again let alone recovering any lost wallet info if there were any?

my HDD was off and untouched for quite sometime. now it's corrupted because of the new version of raspiblitz. do you suggest I format my HDD and redo all again?

I'm looking for an alternative...

amiroff157, 2020-10-12:
P.S assuming the HDD is broken why raspiblitz can't format and ask to re-downloads again like on a fresh install?

amiroff157, 2020-10-12:
> I/O erros are hard to pin down. A common source for I/O errors is the power supply. If on writing the power is weak it can corrupt the HDD data.

got 2 original power supply adapters

ChuckNorrison, 2020-10-12:
i observed some sedxxx files in the home directory, maybe we got a bug with a script which is abusing sed?

rootzoll, 2020-10-13:
> i observed some sedxxx files in the home directory, maybe we got a bug with a script which is abusing sed? This could break the system to the point we currently are in

Thats interesting ... can you give more details on this for debug?

ChuckNorrison, 2020-10-13:
Hey, i observed an empty sed file in /mnt/hdd and also in /home/admin. I could try to get the timestamps for those sed files and compare to some logs what was happening at that point.

Maybe someone could recheck the filesystem for some sed* files

openoms, 2020-10-13:
> Maybe someone could recheck the filesystem for some sed* files

Found nothing like you describe:
```
$ sudo find / -name sed* 
find: ‘/proc/4597’: No such file or directory
find: ‘/proc/10709/task/10709/net’: Invalid argument
find: ‘/proc/10709/net’: Invalid argument
/home/admin/.cache/yarn/v6/npm-shelljs-0.8.3-a7f3319520ebf09ee81275b2368adb286659b097-integrity/node_modules/shelljs/src/sed.js
/home/thunderhub/thunderhub/node_modules/shelljs/src/sed.js
find: ‘/run/user/1000/gvfs’: Permission denied
/var/lib/dpkg/info/sed.md5sums
/var/lib/dpkg/info/sed.list
/usr/include/linux/sed-opal.h
/usr/share/vim/vim81/syntax/sed.vim
/usr/share/doc/sed
/usr/share/doc/sed/sedfaq.txt.gz
/usr/share/locale/et/LC_MESSAGES/sed.mo
/usr/share/locale/de/LC_MESSAGES/sed.mo
/usr/share/locale/fr/LC_MESSAGES/sed.mo
/usr/share/locale/pt/LC_MESSAGES/sed.mo
/usr/share/locale/it/LC_MESSAGES/sed.mo
/usr/share/locale/fi/LC_MESSAGES/sed.mo
/usr/share/locale/zh_TW/LC_MESSAGES/sed.mo
/usr/share/locale/id/LC_MESSAGES/sed.mo
/usr/share/locale/he/LC_MESSAGES/sed.mo
/usr/share/locale/eo/LC_MESSAGES/sed.mo
/usr/share/locale/uk/LC_MESSAGES/sed.mo
/usr/share/locale/eu/LC_MESSAGES/sed.mo
/usr/share/locale/pl/LC_MESSAGES/sed.mo
/usr/share/locale/tr/LC_MESSAGES/sed.mo
/usr/share/locale/da/LC_MESSAGES/sed.mo
/usr/share/locale/sv/LC_MESSAGES/sed.mo
/usr/share/locale/af/LC_MESSAGES/sed.mo
/usr/share/locale/ro/LC_MESSAGES/sed.mo
/usr/share/locale/nl/LC_MESSAGES/sed.mo
/usr/share/locale/sl/LC_MESSAGES/sed.mo
/usr/share/locale/gl/LC_MESSAGES/sed.mo
/usr/share/locale/ca/LC_MESSAGES/sed.mo
/usr/share/locale/el/LC_MESSAGES/sed.mo
/usr/share/locale/ko/LC_MESSAGES/sed.mo
/usr/share/locale/hr/LC_MESSAGES/sed.mo
/usr/share/locale/es/LC_MESSAGES/sed.mo
/usr/share/locale/cs/LC_MESSAGES/sed.mo
/usr/share/locale/sr/LC_MESSAGES/sed.mo
/usr/share/locale/ru/LC_MESSAGES/sed.mo
/usr/share/locale/hu/LC_MESSAGES/sed.mo
/usr/share/locale/nb/LC_MESSAGES/sed.mo
/usr/share/locale/sk/LC_MESSAGES/sed.mo
/usr/share/locale/vi/LC_MESSAGES/sed.mo
/usr/share/locale/pt_BR/LC_MESSAGES/sed.mo
/usr/share/locale/ja/LC_MESSAGES/sed.mo
/usr/share/locale/bg/LC_MESSAGES/sed.mo
/usr/share/locale/zh_CN/LC_MESSAGES/sed.mo
/usr/share/locale/ga/LC_MESSAGES/sed.mo
/usr/share/locale/ast/LC_MESSAGES/sed.mo
/usr/share/info/sed.info.gz
/usr/share/man/man1/sed.1.gz
/bin/sed
```

You could have run buggy `sed` command manually. Happens to me often.

rootzoll, 2020-10-13:
@ChuckNorrison can you describe when those start to appear - is that before the RaspiBlitz get Stuck or is it after and you were able to mount the drive again?

ChuckNorrison, 2020-10-13:
Hey, sorry for the sed hint, seems not related to that. I found a great answer on raspis so called "maintenance mode". This seems to gets caused by external USB storage. Maybe this is not available in time while boot. 

This can be solved by edit fstab and remove the dependency to external storage... maybe mount after boot?

amiroff157, 2020-10-14:
Update:

connected the HDD to Windows/Linux OS and it looks dead and needs a manual reformat since raspiblitz can't do on it's own while installing.. 

Looking into how #552 might be related

amiroff157, 2020-10-14:
@rootzoll 

The final conclusion is that HDDs are risky and shouldn't be used on a raspiblitz setup, even like the one I have which is USB 3.0 spec... I will be looking for a SSD since they are more durable and require less power than HDDs and powering the raspberry pi with a proper 5v 3A power adapter should be no issue for powering other peripherals connected to it.

ChuckNorrison, 2020-10-14:
This should not be closed, this is essential to build this blitz up properly. So maybe the docs needs to get changed if SSD is not affected or we can get deeper into the USB3.0 HDD Limitations of a PI 4

ChuckNorrison, 2020-10-14:
And here we go:

> The adapters work fine under USB2.0 on the Raspberry Pi 4 and on x86 under USB3.0, **without quirks and UAS enabled.**
> This issue probably pertains to VL805 firmware or the Raspberry kernel and both of those are the vendor's responsibility to fix.

https://github.com/raspberrypi/linux/issues/3070

I dont know what the quirks or UAS is and how to configure it

openoms, 2020-10-15:
@ChuckNorrison the UAS / UASP issue with RPi4-s have been discussed extensively.
Check the issues labeled `hardware` : https://github.com/rootzoll/raspiblitz/issues?q=is%3Aissue+label%3Ahardware
: https://github.com/rootzoll/raspiblitz/issues/708#issuecomment-518850854
https://github.com/rootzoll/raspiblitz/issues/691#issuecomment-520526004

ChuckNorrison, 2020-10-15:
Hey, thanks for the thread, i will get into this. Anyway you recommended some hardware one comment above, i bought exactly this usb 3.0 case from the first amazon link. 1TB SSD is used. So SSD is not a solution.

After all that i observed several things to force that behaviour.
- wlan and bluetooth both not disabled in /boot/config.txt
- USB 3.0 was used
- UAS is enabled

So this results in some filesystem check errors while boot and ends up in the so called maintenance mode with press Control-D. 
if check filesystem with `e2fsck -n -v /dev/sda1` tresults in errors, we need to add `forcefsck` to /boot/cmdline.txt. This is risky?! Target should be to avoid this with some preparation and a well configured config.txt and cmdline.txt

openoms, 2020-10-15:
great research, just please keep it clear if you are writing about this `Stuck at "i2c /dev entries driver"` issue which is isolated, not reproducible (I haven't seen it at least) and probably related to a faulty disk or a UASP incompatible adapter.

or the `press Control-D to continue` https://github.com/rootzoll/raspiblitz/issues/1587 problem which is happening to more people and the cause is even less clear.

Might they be related?

ChuckNorrison, 2020-10-15:
I did it 👍 

you need to edit your /etc/fstab and add nofail to your usb hdd

openoms, 2020-10-15:
> you need to edit your /etc/fstab and add nofail to your usb hdd

does this solve this issue or #1587?

ChuckNorrison, 2020-10-15:
this will resolve the control-D, yes. Iam in blitz upgrade screen... maybe the usb hdd is not available and it is not a real solution. but it proofes the usb hdd as the culprit as expected

argentoblitz, 2020-10-16:
First time poster but am having the same "Stuck at 'i2c /dev entries driver'" issue. 

Can SSH in to box. Can run repair. But anything lightning is telling me LND is not working/running. 

Not sure how to add my voice here in an organized fashion, any help appreciated. 

Using a Kingston 16GB Canvas Select Plus microsd, 5v 3.1A power supply, and microHDMI to HDMI cable to an hdtv.

ChuckNorrison, 2020-10-16:
where does your blitz ends up after Control-D? What is the output of `sudo ls -laht /home/lnbits/lnbits`

I solved lnbits issues with `chown -R lnbits /home/lnbits/lnbits` cause permissions are lost

michaelWuensch, 2020-10-17:
Hi, 
I have the same issue now.
After my raspiblitz worked for weeks without issues, I wanted to update lnd today. For some reason the raspiblitz was in sync mode at 99% although it was in sync yesterday... I restarted it, needed to use Ctrl + D to continue on boot, but then it was stuck in an LND restart loop with the same error message as issue #720.
I made an LND backup, and then reset the HDD but kept blockchain data. I flashed my SD card with 1.6.1 RC1.
And when starting I am now completely stuck at 
**i2c /dev entries driver**
There is not even the possibility to press Ctrl +D anymore.

My hardware setup is exactly like before, no cables moved.
@amiroff157  I have an SSD, so I don't think using a SSD instead of an HDD will fix anything.

openoms, 2020-10-17:
@michaelWuensch can you try flashing the SD card again?
Can use the same v1.6.1RC1 image.

michaelWuensch, 2020-10-17:
@openoms just reflashed it, same.

openoms, 2020-10-17:
how it is if you boot without the disk attached?
Which disk and adapter you have?

michaelWuensch, 2020-10-17:
Stuck at the same point without the disk attached.
I will now try to flash 1.6 again which was the version that worked before and see if that changes anything.

michaelWuensch, 2020-10-17:
Hi, it boots now with 1.6.1.
Sorry for the trouble guys, I forgot to copy the hdmi file into the boot section of the sd card.
I run a raspberry pi 4 without lcd.

argentoblitz, 2020-10-17:
Where does one acquire this hdmi file?

michaelWuensch, 2020-10-17:
@argentoblitz just create a new empty file and name it hdmi.
For more info have a look over here:
https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#can-i-run-the-raspiblitz-without-a-displaylcd

argentoblitz, 2020-10-17:
OK, I re-loaded the system with Raspberry Pi Imager, and then used my mac
terminal to do
touch hdmi

and
touch ssl

which both wrote files as expected.

Now, I get rainbow screen but no terminal text on the HDTV -- where
previously I would get scrolling terminal text until the *i2c /dev entries
driver*
issue.

Thanks for any help.

Matt

ChuckNorrison, 2020-10-24:
Iam at this point again after 2 days and a reboot :( At Startup i observed extfs filesystem errors. This blitz project seems not to work at a raspi anymore?!

@rootzoll is your blitz running?

rootzoll, 2020-10-26:
@ChuckNorrison My Blitzes work fine ... but I can see the "control-d" things gets pretty anoying for some people. I will try to put higher priority on booting from SSD feature 1235 - moving away from booting from sd card.

argentoblitz, 2020-10-26:
I ended up going with raspibolt done by hand and it’s fine. Takes time but
raspiblitz in its current form even with all recommended equipment was too
opaque for me to make progress in, although we are surely all supremely
grateful for all of it.

Matt

ChuckNorrison, 2020-10-27:
@argentoblitz Can you please keep this updated if your bolt is running over time? Maybe the RaspiBolt will show you this issue too. Would be very helpful to know. Good Luck.

argentoblitz, 2020-10-27:
So far zero such issues. Zero issues at all, in fact. Outside of the usual
blockchain download wait, it took plenty of time to type and/or paste
commands, but it’s been up and running solidly.

But yes, definitely will stay in thread with any related such thing.

ChuckNorrison, 2020-11-03:
No issues after all the tweaks above and  updating to 1.6.1. Several reboots and install/uninstall RTL, lnbits successful. Even lnd 0.11 is running fine.

Maybe we could add some advice on using USB 2.0 and free up all available USB ports. Further disable unused connections in config.txt could free up some power consumption.

amiroff157, 2020-11-04:
> No issues after all the tweaks above and updating to 1.6.1. Several reboots and install/uninstall RTL, lnbits successful. Even lnd 0.11 is running fine.
>
> Maybe we could add some advice on using USB [...]

updated mine to 1.6.1 and wiped the entire drive for a fresh install. Blockchain progress is at 87% will update once it's done and stable!

rootzoll, 2020-11-11:
For the ext4 HDD is switched to the fstab options (including nofail) recommended by RaspiBolt guide for a lucky chance that this will ease some pain with the Control-D ... but to be tested. Will merge PR #1727 

TODO: Retest fresh install with HDD and then push this issue to v1.7 for next iteration.

amiroff157, 2020-11-11:
> For the ext4 HDD is switched to the fstab options (including nofail) recommended by RaspiBolt guide for a lucky chance that this will ease some pain with the Control-D ... but to be tested. Will merge PR #1761
> 
> TODO: Retest fresh install with HDD and then push this issue to v1.7 for next iteration.

Already got 1.6.1 running with a fresh install. Looking forward to 1.7 👍 

Thanks @rootzoll !

rootzoll, 2020-12-02:
OK looks good. Closing for v1.6.2 release.

closed 2020-12-02 by rootzoll