raspiblitz/raspiblitz #1627: [Feature Request] Add xud

kilrau, 2020-10-06:
[We](https://github.com/ExchangeUnion/) would provide a PR, this issue serves for discussing & agreeing on technical details with the RaspiBlitz community on the way. We target to have a presentable MVP for the upcoming Hacksprint on October 17th.

### Why?
[xud](https://github.com/ExchangeUnion/xud) as optional service enables RaspiBlitz users to use their bitcoin in existing lightning channels to offer orders on the [OpenDEX Network](https://opendex.network/) and to earn via arbitrage profits from trades. [Here](https://blog.exchangeunion.com/launching-opendex-the-decentralized-exchange-standard-b907d2a99b5a) an explanation what OpenDEX is, how it uses lightning and how the arbitrage works. [Here](https://decrypt.co/40669/opendex-wants-bring-defi-style-returns-bitcoin) an independent article covering the concept.

### What?
Adding [xud](https://github.com/ExchangeUnion/xud), packaged with minimal dependencies [arby](https://github.com/ExchangeUnion/market-maker-tools) and [connext](https://github.com/connext/rest-api-client) as service menu item within RaspiBlitz.

### How? TBD.
Installation Type:
1. Preferred: [xud-docker](https://github.com/ExchangeUnion/xud-docker), with only docker (compose not needed) as dependency. Pro: extensible, well-tested, streamlined setup flow, avoids dealing with nodejs incompatibility, copying tls certs and additionally keeps the OS clean.
2. Alternative: Native installation of xud, connext and arby from source with nodejs as dependency. Pro: fits Raspiblitz style, could use [this existing Raspiblitz Nodejs installation script](https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/bonus.nodejs.sh). Con: complicated to set up, each feature (e.g. a web ui) will need separate installation, battling with nodejs upgrades and incompatibility. 

More:
3. xud needs lnd with hold-invoice support. The default lightning labs binaries don't include this. Option a) add hold-invoice support via [this](https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/lnd.update.sh) existing script or b) use a separate lnd withing xud-docker. Both options can be made available to the user.
4. switch to a 64-bit base OS. [Considered inevitable](https://github.com/rootzoll/raspiblitz/issues/1199#issuecomment-640215632) (needed for LND to support a db >2GB) and greatly improves performance for docker.
5. installation of docker. Easiest probably via a script:
```
curl -fsSL https://get.docker.com -o get-docker.sh && 
sudo sh get-docker.sh
sudo usermod -aG docker ${USER}
```

kilrau, 2020-10-29:
Small update here:
- went the xud-docker way and install docker as dependency in the installation flow.
- decided to make the docker installation a generic docker install script available to other services
- developing & testing straight on the [arm64 image](https://github.com/rootzoll/raspiblitz/issues/1199)

rootzoll, 2021-03-27:
Good news everyone ... the RaspiBlitz v1.7RC1 image based on the RaspiOS 64-bit is now available:
https://github.com/rootzoll/raspiblitz/tree/dev#downloading-the-software

Also there is a basic docker framework that can be activaed now:
https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/config.scripts/blitz.docker.sh

So happy to see someone hacking and adding OpenDEX to the RaspiBlitz - remeber to work/test with the `dev` branch :)

cstenglein, 2023-06-07:
Closing this since OpenDEX doesn't seems to be [actively maintained anymore](https://opendex.network/)

closed 2023-06-07 by cstenglein