raspiblitz/raspiblitz #1629: v1.6 RTL Bad Gateway

LilChange, 2020-10-06:
Hallo! Long time fan first time caller.

I am having similar issues to #1285 with RTL. I have two Blitzes running v1.6 and the first device runs perfectly.

With this new one I am getting Bad Gateway on both the onion and the local IP after installing RTL. I restarted RTL a few times, rebooted and even reinstalled NPM after I had issues with it not installing correctly. BTCPay server runs fine over Tor.

`sudo nginx -t` gives us:
```
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
```

And `systemctl status RTL` sadly gives us this:  

```
● RTL.service - RTL daemon
   Loaded: loaded (/etc/systemd/system/RTL.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
```

So my next step is to flash the memory card and start again but before I did that I thought I would let you know I am having a similar issue to the one reported and if you wanted me to send you any logs let me know and I will gladly do that. I am not super technical but I have used command line before.

rootzoll, 2020-10-06:
Hi :) If you call `debug` command you should get also a section with the RTL logs ... can you see any error log there?

LilChange, 2020-10-06:
The only part with RTL is this section:

```
*** LAST 20 RTL LOGS ***
sudo journalctl -u RTL -b --no-pager -n20
-- Logs begin at Tue 2020-10-06 16:11:05 BST, end at Tue 2020-10-06 19:37:22 BST. --
-- No entries --
```

rootzoll, 2020-10-06:
So it really seems to be the install falied .. on the command line can you first run:

`/home/admin/config.scripts/bonus.rtl.sh off`

then run and checking for any error output the install again with: 

`/home/admin/config.scripts/bonus.rtl.sh on`

LilChange, 2020-10-06:
I still get 502. This is what is reported last time I installed it. Here's the full log.

```
***********************************
* RaspiBlitz Commandline
* Here be dragons .. have fun :)
***********************************
LND command line options: lncli -h
Back to main menu use command: raspiblitz

admin@raspberrypi:~ $ cd /home/admin/config.scripts/
admin@raspberrypi:~/config.scripts $ ./bonus.rtl.sh on
making sure services are not running
*** INSTALL RTL ***
nodeJS is already installed
Installed nodeJS v12.18.4
Adding user `rtl' ...
Adding new group `rtl' (1003) ...
Adding new user `rtl' (1003) with group `rtl' ...
Creating home directory `/home/rtl' ...
Copying files from `/etc/skel' ...
*** make sure rtl is member of lndadmin ***
*** make sure symlink to central app-data directory exists ***
*** Get the RTL Source Code ***
Cloning into '/home/rtl/RTL'...
remote: Enumerating objects: 327, done.
remote: Counting objects: 100% (327/327), done.
remote: Compressing objects: 100% (264/264), done.
remote: Total 13857 (delta 201), reused 120 (delta 63), pack-reused 13530
Receiving objects: 100% (13857/13857), 50.45 MiB | 2.70 MiB/s, done.
Resolving deltas: 100% (11177/11177), done.
HEAD is now at 12fd25e CLT update channel fee bug fix
OK - RTL code copy looks good

*** Run: npm install ***
npm WARN tarball tarball data for material-design-icons@3.0.1 (sha1-mnHEh0chjrylHlGmbaaCA4zct78=) seems to be corrupted. Trying one more time.

> rtl@0.9.0-beta postinstall /home/rtl/RTL
> ngcc

Compiling @angular/core : fesm2015 as esm2015
Compiling @angular/cdk/keycodes : fesm2015 as esm2015
Compiling @angular/animations : fesm2015 as esm2015
Compiling @angular/compiler/testing : fesm2015 as esm2015
Compiling @angular/animations : fesm5 as esm5
Compiling @angular/animations : esm2015 as esm2015
Compiling @angular/animations : esm5 as esm5
Compiling @angular/animations : main as umd
Compiling @angular/cdk/keycodes : fesm5 as esm5
Compiling @angular/cdk/keycodes : esm2015 as esm2015
Compiling @angular/cdk/keycodes : esm5 as esm5
Compiling @angular/cdk/keycodes : main as umd
Compiling @angular/compiler/testing : fesm5 as esm5
Compiling @angular/compiler/testing : esm2015 as esm2015
Compiling @angular/compiler/testing : esm5 as esm5
Compiling @angular/common : fesm2015 as esm2015
Compiling @angular/compiler/testing : main as umd
Compiling @angular/cdk/observers : fesm2015 as esm2015
Compiling @angular/animations/browser : fesm2015 as esm2015
Compiling @angular/cdk/collections : fesm2015 as esm2015
Compiling @angular/cdk/platform : fesm2015 as esm2015
Compiling @angular/platform-browser : fesm2015 as esm2015
Compiling @angular/cdk/bidi : fesm2015 as esm2015
Compiling @angular/cdk/a11y : fesm2015 as esm2015
Compiling @angular/platform-browser/animations : fesm2015 as esm2015
Compiling @angular/forms : fesm2015 as esm2015
Compiling @angular/cdk/scrolling : fesm2015 as esm2015
Compiling @angular/cdk/portal : fesm2015 as esm2015
Compiling @angular/material/core : fesm2015 as esm2015
Compiling @angular/cdk/layout : fesm2015 as esm2015
[... 393 lines cut ...]
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
*** Updating Firewall ***
Rule updated
Rule updated (v6)
Rule updated
Rule updated (v6)

*** Install RTL systemd for bitcoin on main ***
Created symlink /etc/systemd/system/multi-user.target.wants/RTL.service → /etc/systemd/system/RTL.service.
OK - the RTL service is now enabled

# Hidden Service for RTL
HiddenServiceDir /mnt/hdd/tor/RTL
HiddenServiceVersion 3
HiddenServicePort 80 127.0.0.1:3002
HiddenServicePort 443 127.0.0.1:3003

Restarting Tor to activate the Hidden Service...

The Tor Hidden Service address for RTL is:
3b62vfsqcgnms7dd7mxqoggfkq6quvnpqau6265dz7t3zbdjskzsurqd.onion
use with the port: 80
or the port: 443
```

rootzoll, 2020-10-06:
OK so NPM install looks good .... what do you get when you run:
`sudo -u rtl /usr/bin/node /home/rtl/RTL/rtl --lndir /home/rtl/.lnd/data/chain/bitcoin/mainnet`

LilChange, 2020-10-06:
Aha now it's started working. Not sure what caused that though. I got:

```Please note that, RTL has encrypted the plaintext password into its corresponding hash.
Server is up and running, please open the UI at http://localhost:3000```

But I can now reach it on localhost and onion. Thanks very much. I will post back if I have any further glitches.

closed 2020-10-06 by LilChange