raspiblitz/raspiblitz #1632: [Connection refused] Can't pair WooCommerce with BTCPay; IP2Tor Bridge + letsencrypt active

DoktorWhat, 2020-10-07:
Not sure if this is the right place to ask for help because I can't pinpoint what exactly is causing this issue. It would be nice if someone could help me troubleshoot this issue tho because a few people already reported the same problem in the raspiblitz telegram channel.

### Problem

The IP2Tor bridge and letsencrypt(DuckDNS) subscriptions are active and working meaning BTCPay is reachable over clearnet.

Still BTCPay won't with woocommerce.

(I already reinstalled BTCPay, tried different IP2Tor bridge subscriptions and other web browsers. Doesn't fix it.)

It seems like the BTCPay woocommerce plugin doesn't pick up the API keys

I am getting the following errors:

### Web browser error message:

"Failed to connect to (url): connection refused"

### BTCPay woocommerce plugin error logs:

BTCPay Woocommerce payment plugin object constructor called. Plugin is v3.0.10 and server is PHP v7.3
[Info] $this->api_key            = 
[Info] $this->api_pub            = 
[Info] $this->api_sin            = 
[Info] $this->api_token          = 
[Info] $this->api_token_label    = 
[Info] $this->api_url        = 
[Error] The API Key was NOT an instance of PrivateKey!  Instead, it appears to be a NULL value.
[Error] The Public Key was NOT an instance of PublicKey!  Instead, it appears to be a NULL value.
[Error] The API Token was NOT an instance of Token!  Instead, it appears to be a NULL value.
[Info] Transaction speed is now set to: default
[Info] The plugin is NOT valid for use!
[Info] Entered generate_api_token_html()...
[Info] Leaving generate_api_token_html()...
[Info] Entered generate_order_states_html()...
[Info] Leaving generate_order_states_html()...

rollomcfloogle, 2020-11-15:
Any luck on this one?  I've having the same problem.

openoms, 2020-11-17:
[image]
https://coincharge.io/en/raspiblitz-btcpay-server/

rollomcfloogle, 2020-11-18:
I had been using that walkthrough as help (they did an excellent job).  I've tried each of the three tor bridges and used both DuckDNS and DYNU (only for one of the bridges; I only used DuckDNS for the other two).  Still no luck.  I'm also unable to connect using my own subdomain.

I've used BTCPay Server on my site previously and the API tokens worked then.

I'll keep plugging away and see if I can figure anything out.

openoms, 2020-11-18:
There has been a fix for dynu.com in #1772. Might worth a shot.

DoktorWhat, 2020-12-13:
I have the same problem with dynu.com

rootzoll, 2020-12-14:
@dennisreimann do you know if there are any additional ports needed in the pairing process of WooComerce & BTCPay that we need to take take of in a IP2TOR setup.

Another gues might be that the WooComerce Plugin cannot handle BTCPay running on another port then http 80 or https 443.

dennisreimann, 2020-12-14:
I don't think it is a port issue, but maybe related to SSL: Is there a valid certificate for the IP/domain you are using?

Another idea: The key values are set up during the pairing process. Can you try to redo the pairing and see if you get any error during that? Also check the debug logs for more detailed information.