raspiblitz/raspiblitz #1647: URGENT - Partial LND Vulnerability Disclosure, Upgrade to 0.11.x

AugustoResende, 2020-10-09:
```-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Hi all,

We are writing to let the Lightning community know about the existence of
vulnerabilities that affect lnd versions 0.10.x and below. The full details of
these vulnerabilities will be disclosed on October 20, 2020. The circumstances
surrounding the discovery resulted in a compressed disclosure timeline compared
to our usual timeframes. We will be publishing more details about this in the
coming weeks along with a comprehensive bug bounty program.

While we have no reason to believe these vulnerabilities have been exploited in
the wild, we strongly urge the community to upgrade to lnd 0.11.0 or above ASAP.
Please ping us on the #lnd IRC channel, the LND Slack, or at
support at lightning.engineering if you need any assistance in doing so. Upgrade
instructions can be found in our installation docs:
https://github.com/lightningnetwork/lnd/blob/master/docs/INSTALL.md#installing-lnd.

Regards,
Conner Fromknecht
-----BEGIN PGP SIGNATURE-----

iQIzBAEBCAAdFiEEnI1hhop8SSADsnRO59c3tn+lkscFAl9/ozwACgkQ59c3tn+l
kscVvBAAk21z6tlHPkOSwfj1lBE0pqc65A6Qa927WEjN5hdUpjjof4Xo2j+GzbnN
Uoj4HGZu+koakzoVpJ4mzN+vg086zAnv+K668hhl7bbPHsQu6FqA1ALiAyy0nH6H
1yukXxpRflq53RTIVPjrEnFVdt6FCLhkCm9LuOk0a/SUf8D4b/N6OaB1Bxupeceu
QFSCIkb9kvW/Eplwkv7PEnx/IZNGIQP9F11DaKLTAjWY5RnIxmCw/oamvlP8Mxt8
/AqlzWVtPVqvwgJLhbMziraXNVV05naHrIXvbXrOI2Q7FZjdaxF+S4EKT4feuq1w
iW7NYSS/u5N2FP3yK8YIdoX0I/nwYQQcpsfbAv2dS4Ql2Td/dyREId4NcchmaKSV
N3w1jByMPWrgUtinl5WEDDOJdUKS2PHkQ95t3s/1uYDFsPz1kXJR2x37a/1AVz/K
6zQ45wFvHEopFR49hu/CV6MUvsvn4XKzPa46Ii7puaBaNqygx0RwuwlxbxCNxPNQ
v45CaCUEq2Tj3stu7YoYGntFvrXVkxXJocn51eK6D+g0bIEXxaGlPJeTuvifKMTO
3T3ZEEbCe9UhDUT8Ja2boP2IIi8wAyExGS59k0tndQGzMSjkzWZ0fzgYyyf+y4nt
r3nTCGi5WWe4y1i2KpiYZTRrQkbrNkRf+fnVdlnTS4lcgEWFFiY=
=8t9Q
-----END PGP SIGNATURE-----```

If you want, you can use the upgrade scripts from this page: https://openoms.gitbook.io/lightning-node-management/lnd.updates
(But don't trust, verify the sh before running on your RaspiBlitz.)

daldwinc, 2020-10-09:
I was going to post an issue about the same topic, but phrased differently.

- Is there any reason to believe that the LND interim update function in rapiblitz is not a good choice to resolve this vuln? 
- Would it be possible to get a raspiblitz patch that include an updated LND that isn't just a pull from master?

woeisme, 2020-10-10:
I'm testing the ssh menu LND reckless update now, seems to be the same update as the openoms suggested script. 
Rebooted and am running 0.11-beta currently. Let's keep updates and questions going in this thread if possible as I'm sure others will benefit.

`menu -> UPDATE -> LND -> RECKLESS Experimental LND update to v0.11.1-beta `

openoms, 2020-10-10:
The difference between this script: https://github.com/openoms/lightning-node-management/blob/master/lnd.updates/lnd.update.v0.11.1-beta.sh offered here: https://twitter.com/openoms/status/1314439082509037569 and the built in reckless option in the UPDATE menu is my script does PGP verify the binary and works on the previous raspiblitz versions too.

The menu option does always go for the latest release so can't be tested or predict who signs it therefore given the reckless tag.

rootzoll, 2020-10-13:
The soon to be released v1.6.1 will have LND 0.11.1 by default.

nyxnor, 2021-04-18:
Please close this issue. mentioned about lnd updates here. https://github.com/rootzoll/raspiblitz/issues/1661#issuecomment-821979634

AugustoResende, 2021-04-18:
closed @nyxnor

closed 2021-04-18 by AugustoResende