raspiblitz/raspiblitz #1650: Help resloving this Error: sudo: unable to resolve host xxxx Name or service not known

CoderBoogs, 2020-10-10:
When I ssh into my blitz after entering the password the program runs as normal and then I get the above error. When searching the old issues I noticed someone was able to change the hostname but when inside powershell the above error runs twice and powershell is unresponsive. Need help? should i re-flash the SD card?

openoms, 2020-10-11:
This is likely a network realted problem. You can try connecting from an other device like using Termux on your phone.

Also should help if you set a fixed IP for your raspiblitz in your router's settings.

CoderBoogs, 2020-10-11:
Thanks. My router is a bit dated thinking of upgrading soon maybe go a mesh network that is wifi-6 compatible. When I make the switch anything I should do specifically with my blitz?

openoms, 2020-10-11:
just connect the blitz in to the new network and it will display the new LAN IP where you can connect. Generally a wired connection is more reliable and private.

CoderBoogs, 2020-10-12:
Okay thanks. I just tried ssh to blitz using termius and got the same error. What should I do next?

openoms, 2020-10-12:
you can try reseting the ssh certificates before reflashing the SDcard: https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-can-i-repair-my-ssh-login

CoderBoogs, 2020-10-17:
Tried this and now the raspiblitz is not showing up on my network (using angry IP scanner)

openoms, 2020-10-17:
If you don't have a GPIO display to see where it is stuck reflashing the SDcard is your best bet.

CoderBoogs, 2020-10-17:
I just flashed a new SD image and updated to v 1.6 and the IP address showed up. 

 I shh in It seems the raspi is booting up now.

Hoping doing this I don't lose my channels

CoderBoogs, 2020-10-17:
Everything is up and running thanks for the help @openoms

closed 2020-10-17 by CoderBoogs