raspiblitz/raspiblitz #1651: Stuck on boot

othervice, 2020-10-10:
Hello,

i followed the readme: Downloading image, write it to the SD-Card aso. Did this several times before with other versions.

After booting the Raspiblitz v1.6 (latest download) it freezes after around 4 seconds on the same line every time. I tried different SD-Cards and SSD drives. All with same result. I had v1.5 running on the same setup without problems. An image of MyNode worked without problems.

Im on Raspberry Pi 4 4GB with 1TB Crucial SSD - I use a 825x board for the SSD, may this cause a problem in the new version? Seems like everything is working fine until its stuck. 

[image]

Regards
Other

chiefuschrist, 2020-10-11:
Did you add the "hdmi" file to the boot drive? It looks like you're using an actual LCD screen. I had the same problem until I added the hdmi file. I think it's expecting to boot everything onto the screen you can attach to the raspi, unless I'm mistaken!

othervice, 2020-10-11:
Hi! No i do not use the LCD screen. Im using my TV screen on first boot just to see whats happening. I installed a few raspiblitz before and did it always that way without any problems. 

Where to add the hdmi file? And where to get it?

openoms, 2020-10-11:
just boot without the HDMi connected and connect via ssh as described in the readme. Similar as here: https://github.com/rootzoll/raspiblitz/issues/1618#issuecomment-703639582

othervice, 2020-10-11:
That did the trick! Strange, but thanks a lot!!

closed 2020-10-11 by othervice