raspiblitz/raspiblitz #1653: Unable to create server: no active channels exist

gogy1985, 2020-10-11:
Hi. All was good till now, but today my node stopped with message: "Unable to create server: no active channels exist". Looks like the reason is: " bitcoind.service: Failed to parse PID from file /home/bitcoin/.bitcoin/bitcoind.pid: Invalid argument" What can i do with this? Full debug log:

```
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.0
chainnetwork: bitcoin / main
 18:48:09 up 44 min,  2 users,  load average: 0.08, 0.22, 0.41

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-10-11 18:06:10 BST; 41min ago
  Process: 5917 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 5923 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 6021 (bitcoind)
    Tasks: 18 (limit: 4497)
   Memory: 1.1G
   CGroup: /system.slice/bitcoind.service
           └─6021 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitco…

**Oct 11 18:06:10 raspberrypi systemd[1]: bitcoind.service: Failed to parse PID from file /home/bitcoin/.bitcoin/bitcoi…argument**
Oct 11 18:06:10 raspberrypi systemd[1]: Started Bitcoin daemon.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2020-10-11 18:02:26 BST, end at Sun 2020-10-11 18:48:09 BST. --
Oct 11 18:06:10 raspberrypi systemd[1]: Starting Bitcoin daemon...
**Oct 11 18:06:10 raspberrypi systemd[1]: bitcoind.service: Failed to parse PID from file /home/bitcoin/.bitcoin/bitcoind.pid: Invalid argument**
Oct 11 18:06:10 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-10-11T17:08:41Z New outbound peer connected: version: 70015, blocks=652269, peer=2 (full-relay)
2020-10-11T17:08:46Z New outbound peer connected: version: 70015, blocks=652269, peer=3 (full-relay)
2020-10-11T17:08:48Z New outbound peer connected: version: 70015, blocks=652270, peer=4 (full-relay)
2020-10-11T17:09:41Z UpdateTip: new best=000000000000000000067c2a608b39b47b39fc5fa697958c19cd68770fd4b310 height=652270 version=0x37ffe000 log2_work=92.36051 tx=576739151 date='2020-10-11T17:08:17Z' progress=1.000000 cache=1.4MiB(12444txo) warning='70 of last 100 blocks have unexpected version'
**2020-10-11T17:09:41Z socket send error Broken pipe (32)**
2020-10-11T17:09:54Z New outbound peer connected: version: 70015, blocks=652270, peer=6 (full-relay)
2020-10-11T17:10:00Z New outbound peer connected: version: 70015, blocks=652270, peer=7 (full-relay)
2020-10-11T17:10:28Z New outbound peer connected: version: 70015, blocks=652270, peer=8 (full-relay)
2020-10-11T17:10:28Z New outbound peer connected: version: 70015, blocks=652270, peer=9 (full-relay)
2020-10-11T17:10:35Z New outbound peer connected: version: 70015, blocks=652270, peer=10 (full-relay)
2020-10-11T17:10:39Z New outbound peer connected: version: 70015, blocks=652270, peer=11 (block-relay)
2020-10-11T17:10:41Z Imported mempool transactions from disk: 1129 succeeded, 660 failed, 0 expired, 4 already there
2020-10-11T17:10:44Z New outbound peer connected: version: 70015, blocks=652270, peer=12 (block-relay)
2020-10-11T17:11:19Z New outbound peer connected: version: 70015, blocks=652270, peer=13 (full-relay)
2020-10-11T17:12:52Z New outbound peer connected: version: 70015, blocks=652270, peer=15 (full-relay)
2020-10-11T17:15:31Z Pre-allocating up to position 0x500000 in rev02276.dat
2020-10-11T17:15:31Z UpdateTip: new best=000000000000000000060319da48c26cfe4fa5c2e26d9d295e23d57176c58361 height=652271 version=0x3fffe000 log2_work=92.360529 tx=576741139 date='2020-10-11T17:15:22Z' progress=1.000000 cache=2.4MiB(21827txo) warning='71 of last 100 blocks have unexpected version'
2020-10-11T17:16:57Z New outbound peer connected: version: 70015, blocks=652271, peer=18 (full-relay)
2020-10-11T17:18:08Z UpdateTip: new best=000000000000000000082a7a5c6bd40876bec0b35d50a97882bd2eab968da255 height=652272 version=0x20000000 log2_work=92.360548 tx=576741828 date='2020-10-11T17:17:55Z' progress=1.000000 cache=2.7MiB(24290txo) warning='70 of last 100 blocks have unexpected version'
2020-10-11T17:28:00Z New outbound peer connected: version: 70015, blocks=652272, peer=27 (full-relay)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-10-11 18:31:21 BST; 16min ago
  Process: 3319 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 3321 (lnd)
    Tasks: 13 (limit: 4497)
   Memory: 12.8M
   CGroup: /system.slice/lnd.service
           └─3321 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735

Oct 11 18:31:21 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Oct 11 18:31:21 raspberrypi systemd[1]: Started LND Lightning Daemon.

[... 103 lines cut ...]
tempCelsius='39.4'
uptime=2665
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=652272
initialSync=0
syncProgress=100.00
startcountLightning=3
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

```

openoms, 2020-10-11:
nothing here apart from
`2020-10-11 18:31:21.925 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
`

run;
`lncli unlock`

or use shortcut:
`blitz`  
for the menu

gogy1985, 2020-10-12:
> nothing here apart from
> `2020-10-11 18:31:21.925 [INF] LTND: Waiting for wallet encryption password. Use `lncli create`to create a wallet,`lncli unlock`to unlock an existing wallet, or`lncli [...]

Thanks for your answer. I tried to unlock before and now again, this dont helps, after making lnd unlocked i get "lnd succesfull unlocked", then if i go to `menu`>INFO i getting my nodes **appropriate** info with red line down "LND   Not Started | Not Ready Yet" then ENTER again INFO and receive "Unable to create server: no active channels exist". If i run `blitz` directly after unlock it from the start said  "Unable to create server: no active channels exist". So after unlocking it becomes locked again.

openoms, 2020-10-12:
you need to update to at least lnd 0.11.x. See: https://github.com/rootzoll/raspiblitz/issues/1647
Can use the UPDATE menu.

gogy1985, 2020-10-12:
> 
> 
> you need to update to at least lnd 0.11.x. See: #1647
> Can use the UPDATE menu.

I made lnd update via gui `menu>UPDATE>LND>RECKLESS..`. After update node was auto restarted. Then was looped LND unlocking via gui (endless asking for unlocking again after 30 sec- 1 min), then i exited to terminal, chacked lncli version, it was 0.11.1-beta , trying again unlock lnd via terminal, and all was like previously said, "The LND service is running with error", "unable to create server: no active channels exist"

Current debug log have some changes:

`
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.0
chainnetwork: bitcoin / main
 08:34:30 up 40 min,  2 users,  load average: 0.58, 0.82, 0.75

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-10-12 07:56:11 BST; 38min ago
  Process: 7681 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 7687 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 7756 (bitcoind)
    Tasks: 18 (limit: 4497)
   Memory: 1.5G
   CGroup: /system.slice/bitcoind.service
           └─7756 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitco…

Oct 12 07:56:11 raspberrypi systemd[1]: Starting Bitcoin daemon...
Oct 12 07:56:11 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2020-10-12 07:53:28 BST, end at Mon 2020-10-12 08:34:31 BST. --
Oct 12 07:56:11 raspberrypi systemd[1]: Starting Bitcoin daemon...
Oct 12 07:56:11 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-10-12T06:58:49Z New outbound peer connected: version: 70015, blocks=652351, peer=6 (full-relay)
2020-10-12T06:59:17Z New outbound peer connected: version: 70015, blocks=652351, peer=8 (full-relay)
2020-10-12T06:59:30Z New outbound peer connected: version: 70015, blocks=652351, peer=9 (full-relay)
2020-10-12T06:59:38Z New outbound peer connected: version: 70015, blocks=652351, peer=11 (full-relay)
2020-10-12T06:59:43Z New outbound peer connected: version: 70015, blocks=652351, peer=12 (block-relay)
2020-10-12T06:59:49Z New outbound peer connected: version: 70015, blocks=652351, peer=13 (block-relay)
2020-10-12T07:00:58Z New outbound peer connected: version: 70015, blocks=652351, peer=16 (full-relay)
2020-10-12T07:01:04Z UpdateTip: new best=00000000000000000007ca32c90ee0e963b19a242e9805d265053bd3200e73e8 height=652352 version=0x20000000 log2_work=92.362052 tx=576887931 date='2020-10-12T07:00:57Z' progress=1.000000 cache=1.2MiB(10900txo) warning='74 of last 100 blocks have unexpected version'
2020-10-12T07:03:27Z New outbound peer connected: version: 70015, blocks=652352, peer=17 (full-relay)
2020-10-12T07:08:58Z New outbound peer connected: version: 70015, blocks=652352, peer=21 (full-relay)
2020-10-12T07:19:59Z New outbound peer connected: version: 70015, blocks=652352, peer=31 (full-relay)
2020-10-12T07:22:35Z New outbound peer connected: version: 70015, blocks=652352, peer=34 (full-relay)
2020-10-12T07:22:57Z socket send error Broken pipe (32)
2020-10-12T07:23:21Z UpdateTip: new best=000000000000000000042730d62a3b39ba0589c68b8b5edc0602f642272802e3 height=652353 version=0x20400000 log2_work=92.36207 tx=576890464 date='2020-10-12T07:23:09Z' progress=1.000000 cache=3.2MiB(27774txo) warning='75 of last 100 blocks have unexpected version'
2020-10-12T07:27:48Z UpdateTip: new best=000000000000000000037baeca08665405661fea105b207876e193e006be6a17 height=652354 version=0x3fff0000 log2_work=92.362089 tx=576892655 date='2020-10-12T07:27:36Z' progress=1.000000 cache=4.0MiB(35313txo) warning='75 of last 100 blocks have unexpected version'
2020-10-12T07:27:57Z New outbound peer connected: version: 70015, blocks=652354, peer=39 (full-relay)
2020-10-12T07:30:28Z socket send error Broken pipe (32)
2020-10-12T07:33:15Z New outbound peer connected: version: 70015, blocks=652354, peer=44 (full-relay)
2020-10-12T07:34:06Z Pre-allocating up to position 0x1000000 in rev02276.dat
2020-10-12T07:34:07Z UpdateTip: new best=000000000000000000082a9e325470211c8081493ab6cb418f89a27d7eb51542 height=652355 version=0x27ffe000 log2_work=92.362108 tx=576894626 date='2020-10-12T07:33:45Z' progress=1.000000 cache=6.3MiB(54913txo) warning='76 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-10-12 08:07:23 BST; 27min ago
  Process: 28704 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 28706 (lnd)
    Tasks: 9 (limit: 4497)
   Memory: 19.1M
   CGroup: /system.slice/lnd.service
[... 102 lines cut ...]
tempFAIL=0
tempWARN=0
tempMAX='4800 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.100'
tempCelsius='39.4'
uptime=2452
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=652355
initialSync=0
syncProgress=100.00
startcountLightning=5
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1
`

openoms, 2020-10-12:
looks like this:
https://github.com/lightningnetwork/lnd/issues/2396

Did you have channels open? If not use the REPAIR menu to:
backup lnd
reset LND
and recreate the wallet with the previous seed (will restore the onchain balance only)

gogy1985, 2020-10-12:
I have about 300usd locked in opened channels. I read linked post before.. But dont saw the solution.. There is no way to restore whole lnd node with channels?

gogy1985, 2020-10-12:
> 
> 
> looks like this:
> [lightningnetwork/lnd#2396](https://github.com/lightningnetwork/lnd/issues/2396)
> 
> Did you have channels open? If not use the REPAIR menu to:
> backup lnd
> reset LND
> and recreate the wallet with the previous seed (will restore the onchain balance only)

I have about 300usd locked in opened channels. I read linked post before.. But dont saw the solution.. There is no way to restore whole lnd node with channels?

gogy1985, 2020-10-13:
Acttions: via gui reset lnd and recreate with seed and static channel restoring(file for beckup channels:  /home/admin/beckups/scb/channel.beckup)
Result: Static channel beckup restores your costs, not channels, all channels closed, most costs now on-chain but 2 channels still pending for closing and costs for them are not available for now, hope they will after closing

closed 2020-10-13 by gogy1985