raspiblitz/raspiblitz #1654: Add Circuitbreaker to SSH-GUI

rootzoll, 2020-10-12:
See #1581 & #1649 .. evolve this for SSH-GUI.

openoms, 2021-02-22:
Circuitbreaker is already part of the recovery process: https://github.com/rootzoll/raspiblitz/blob/9ae1a2d9bfdd4d644ed103ef69442dd6b31899f7/home.admin/_bootstrap.provision.sh#L544

I think this service should be on by default with the option to switch it off from the `SETTINGS` menu.
https://twitter.com/openoms/status/1363781464832286720
https://twitter.com/openoms/status/1363792824068677634

rootzoll, 2021-03-24:
I would not change it to "on by default" (yet) .. first now make it easy to activate it under SETTINGS - see if we get any error or side effects reports.

openoms, 2021-03-24:
Agreed, although I am using it om multiple successful routing nodes.
There is a high amount of probes reported from time-to-time, but force closes are very rare now.
It gives a peace of mind that those node are at least less vulnerable to a flooding attack.

rootzoll, 2021-03-24:
OK now ready for v1.7 release and final testing.

rootzoll, 2021-04-13:
Tests look good. Will be part of v1.7. Closing issue.

closed 2021-04-13 by rootzoll