raspiblitz/raspiblitz #1655: Release v1.6.1 Testing

rootzoll, 2020-10-12:
Also because of: #1635

- [ ] update version to v.1.6.1
- [ ] test with fresh setup (blockchain already available)
- [ ] test recover/update from default v1.6.0
- [ ] test recovery with all services on

DrBlackross, 2020-10-12:
im always game, just need to back everything up (i think)

bitsam, 2020-10-13:
Is there a link to the 1.6.1 beta?

rootzoll, 2020-10-13:
@DrBlackross & @bitsam will prepare a v1.6.1RC1 to test the next days - will post link here.

rootzoll, 2020-10-15:
@DrBlackross @bitsam @openoms @frennkie
OK here is the first RC1 to test for v1.6.1 release: https://raspiblitz.org/raspiblitz-v1.6.1RC1-2020-10-15.img.gz
SHA-256: c6254dd39a3ddd4ccd83725728ee6f54f476c3925a60ef83e464d0aee82ce847

rootzoll, 2020-10-18:
OK here is the first RC2 to test for v1.6.1 release:
https://raspiblitz.org/raspiblitz-v1.6.1RC2-2020-10-18.img.gz
SHA-256: 750cab91caaee067ea67bd2da7b1ca2273b647d298b7546e0d52e8978fd475f3

If there are no further error reports this file is planed to be come the final release.

rootzoll, 2020-10-18:
Ah found a small thing .. will make new image within the next hours.

rootzoll, 2020-10-19:
OKnext try RC3 to test for v1.6.1 release:
https://raspiblitz.org/raspiblitz-v1.6.1RC3-2020-10-19.img.gz
SHA256: 635b9fcacf7140c84117850a158ad19695ffbd8d90013a6c309f137cd1b0d32

rootzoll, 2020-10-20:
OK added latest PRs  and prepared the final release file - doing last testing on file:
https://raspiblitz.org/raspiblitz-v1.6.1-2020-10-20.img.gz
SHA256: 6b9925f274f68ca109771b047c0272df31c01dc0a5d2b9e05b582accfa6029ec

closed 2020-10-21 by rootzoll