raspiblitz/raspiblitz #1656: Lot of DNS requests v4.ipv6-test.com

Marcoblue, 2020-10-12:
Raspiblitz requests "v4.ipv6-test.com" 3x per minute via ipv4 and ipv6. Does it have to be that often?
[image]

rootzoll, 2020-10-12:
Did you tested with RaspiBlitz v1.6.0 image or with latest scripts preparing for v1.6.1?

Marcoblue, 2020-10-14:
It is with v1.6.0

rootzoll, 2020-10-14:
If your node is for testing you can update 1.6.0 to the latest scripts with: MENU > UPDATE > PATCH > PATCH-MENU > PATCH and do a reboot. Please let me know if you still see that many requests - that would be great.

rootzoll, 2020-10-16:
Will move this to v1.7 milestone .. with the default Tor getting the public IP gets less important anyway. Seems like a good time to also then make changes on this.

Marcoblue, 2020-10-25:
At 1.6.1 it is still the same.

rootzoll, 2020-10-29:
Hi @Marcoblue the merged PR should now fix the too much calling .. can you verify?
To get the fix on your RaspiBlitz v1.6.1 go to UPDATE > PATCH > PATCHMENU > PATCH and restart.
After the restart you should see the results.

Marcoblue, 2020-10-31:
It works. After the restart he has asked once, but not again since then ( 12 minutes ago).

rootzoll, 2020-10-31:
Perfect :) So this will fix will be part of v1.6.2 release ... closing issue.

closed 2020-10-31 by rootzoll

bigsk1, 2022-04-13:
Googled why this is happening also and found this, running on latest 1.7.2 same issue none stop every 10 secs. I can try and patch and restart but running latest.