raspiblitz/raspiblitz #1662: ip2tor bridge subscription stopped working - Host:91.109.21.148 (eu-fra1 - Owner:fulmo)

grnqrtr, 2020-10-13:
I am able to access BTCPay Server from local address and from the tor hidden service address, but it is no longer accessible through the ip2tor subscription address. I'm not sure when it stopped working, but checked today and it's not working.

My state of subscription is ACTIVE through Host:91.109.21.148 (eu-fra1 - Owner:fulmo)

rootzoll, 2020-10-13:
@frennkie can you see any outages on that host?

frennkie, 2020-10-13:
I would need to know the bridge id/port. I would advise not to post the full port/id here. Can you give me the first 4-5 digits/chars from `./config.scripts/blitz.subscriptions.ip2tor.py subscriptions-list | grep id`?

```
admin@raspberrypi:~ $ ./config.scripts/blitz.subscriptions.ip2tor.py subscriptions-list | grep id
    "id": "23fc2xxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxxx",
```

grnqrtr, 2020-10-13:
Here is first 5 of id: fdff5

I'm not sure if you did anything, but I just checked this morning and it actually looks like it's working again.

diskbuster, 2020-12-24:
I get stuck by paying for  IP2TOR Bridge for LND REST API.

I get the shop address: fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion
but it ends up with an error when the loop finish:
```
#### Placeing order ...
# apiPlaceOrderNew
#### Waiting until invoice is available ...
# Loop 1
# apiGetOrder
....
```
iam not sure if a can post the whole error log!?

```
###### ERROR DETAIL FOR DEBUG #######

Error Short:
timeout on getting invoice
Shop:
fulmo7x6yvgz6zs2b2ptduvzwevxmizhq23klkenslt5drxx2physlqd.onion
Bridge:
{'id': 'b85e18dc-6aa3-4a89-a1ad-d5fbf6036225', 'site': 'ip2tor.fulmo.org', 'created_at': '2020-07-27T09:36:43Z', 'modified_at': '2020-12-24T10:40:48Z', 'ip': '172.81.177.248', 'is_enabled': True, 'is_alive': True, 'name': 'nam-tor1', 'is_testnet': False, 'offers_tor_bridges': True, 'tor_bridge_duration': 86400, 'tor_bridge_price_initial': 48000, 'tor_bridge_price_extension': 40000, 'offers_rssh_tunnels': False, 'rssh_tunnel_price': 1000, 'terms_of_service': '-', 'terms_of_service_url': '-', 'ci_date': '2020-12-24T10:40:48Z', 'ci_message': '', 'ci_status': 0, 'owner': 4, 'tor_bridge_duration_hours': 24, 'tor_bridge_price_initial_sats': 48, 'tor_bridge_price_extension_sats': 40}
```

the subscription for LNbits Tor Bridge was working fine after a few try's.

did I miss a step, to get a invoice?

frennkie, 2020-12-24:
@diskbuster No, you did not miss anything. The problem is caused by issues on the backend. Unfortunately the virtual machine host frequently has performance issues. We are planning to move to another system soon. 

For now just retry it (I fixed the expired TLS (which should have no impace) and changed to a different LN Node).

diskbuster, 2020-12-24:
@frennkie thanks, thats good to know. I really need to access my node thru ssl to make my first steps with Webintegration.
Is there a alternative option to get this done in a sensible effort?

frennkie, 2020-12-24:
I don't understand your question/issue. You should absolutely not use IP2Tor for a connection that is not itself encrypted! Therefore IP2Tor e.g. checks whether you are trying to use a HTTP (instead of HTTPS) service. 

The reason for this is that the IP2Tor bridge host sees all traffic before it enters the (in itself encrypted) Tor network. This "first leg" between your browser and the IP2Tor bridge host is the problem here.

So if you service is HTTP then you need to put it behind HTTPS (e.g. use Nginx as a reverse proxy).

diskbuster, 2020-12-24:
yes, for sure I need an letsencrypt certificate. for that I was now able to activate the LND-REST API TOR2IP Bridge.
then I have the option to add a ssl therefore. 
unfortunately I can subscribe only to one certificate - I have that already for LNbits.
I think i have to figure out what's best for my next steps. going further with LNbits or deeper into Rest Api.

klnit, 2021-02-24:
Hi @frennkie 
my IP2TOR subscription says "Exception on Renew: timeout on getting invoice". My LND was locked a few days. I unlocked last week and would expect it to recover, but had no deductions in the last days (hence not working).
Service: eu-fra1
TOR: uphwrrap4zw.....

I don't know how to force a recovery.
Additionally, what is the best option to get an alert, if Raspi/LNBits not reachable?

would you have a look? Thanks

frennkie, 2021-02-24:
Wow... Embarrassing... The TLS certificate on my Raspiblitz was expired.. 🙄. Should work again.

klnit, 2021-02-25:
Thanks, its working 👍

nyxnor, 2021-04-18:
@grnqrtr Please close this.

closed 2021-04-18 by grnqrtr