raspiblitz/raspiblitz #1672: TOS on IP2Tor bridges & how to get support

rootzoll, 2020-10-14:


openoms, 2020-12-02:
Is now only one IP2Tor subscription allowed at a time on v1.6.2?

petrhejna, 2020-12-04:
HI, I just tested IP2Tor for sphinx. But it was not activated with error (something similar to) "payment not received". 

- The payment for 40 sats went out of my node.
- The subscription list is empty.

rootzoll, 2020-12-04:
@frennkie do you see any problem with the `eu-fra1`?

frennkie, 2020-12-04:
I just checked the orders.. There were successful payments before and after that approximate time. And there is no failed request...

Are there more logs? Did you also try another service besides Sphinx?

petrhejna, 2020-12-04:
If you give me some hint how to find that in logs I can share it.

No I just tested Sphimx.

closed 2023-11-14 by rootzoll