raspiblitz/raspiblitz #1675: Adding specter_warden

holgern, 2020-10-16:
https://twitter.com/alphaazeta/status/1316995228302249985

https://github.com/pxsocs/specter_warden

pxsocs, 2020-11-03:
If help is needed here, let me know. Actually going through setting up a new Raspiblitz node to test installation.

nyxnor, 2021-06-26:
Testing WARden now:
[image]

Possible to connect to specter onion service
[image]

Issue being shared here https://github.com/pxsocs/warden/issues/98

```bash
#!/bin/bash

# https://github.com/pxsocs/warden

# command info
if [ $# -eq 0 ] || [ "$1" = "-h" ] || [ "$1" = "-help" ]; then
 echo "config script to install or uninstall WARden"
 echo "bonus.warden.sh [on|off|menu]"
 exit 1
fi

source /mnt/hdd/raspiblitz.conf

# add default value to raspi config if needed
if ! grep -Eq "^warden=" /mnt/hdd/raspiblitz.conf; then
  echo "warden=off" >> /mnt/hdd/raspiblitz.conf
fi

# show info menu
if [ "$1" = "menu" ]; then
  dialog --title " Info WARden " --msgbox "
WARden is a command line tool.
Type: 'war' in the command line to switch to the dedicated user.
Usage: https://github.com/pxsocs/warden/blob/master/README.md
" 10 75
  exit 0
fi

# install
if [ "$1" = "1" ] || [ "$1" = "on" ]; then

  echo "*** INSTALL WARDEN ***"

  # create war user
  sudo adduser --disabled-password --gecos "" war

  # download source code
  sudo -u war git clone https://github.com/pxsocs/warden.git /home/war/warden

  sudo -u war pip3 install -r /home/war/warden/requirements.txt

  # setting value in raspi blitz config
  sudo sed -i "s/^warden=.*/warden=on/g" /mnt/hdd/raspiblitz.conf

  echo "# Usage: https://github.com/pxsocs/warden/blob/master/README.md"
  echo "# To start type: 'sudo su war' in the command line."
  echo "# To exit the user - type 'exit' and press ENTER"

  exit 0

fi

# switch off
if [ "$1" = "0" ] || [ "$1" = "off" ]; then

  # setting value in raspi blitz config
  sudo sed -i "s/^warden    =.*/warden=off/g" /mnt/hdd/raspiblitz.conf

  echo "*** REMOVING WARDEN ***"
  sudo userdel -rf war
  echo "# OK, war is removed."
  exit 0

fi

# switch off
if [ "$1" = "update" ]; then

  echo "*** UPDATING WARDEN ***"
  cd /home/war/warden
  git pull origin master
  cd /home/admin
  echo "# OK, war is UPDATED."
  exit 0

fi
```

Run web server:
```bash
cd /home/war/warden
sudo -u war python3 warden
```

sovereignhodler, 2022-02-15:
I tried to install this on my Raspiblitz

Created a .sh file with 
`touch bonus.warden.sh
`

Changed permissions
`sudo chmod 755 bonus.warden.sh`

Install
`./bonus.warden.sh on`

Start web server
`cd /var/war/warden
sudo -u war python3 warden`

Output:

Welcome to the WARden <> Launching Application ...

[i] Running from directory: /home/war/warden/warden

✅ Config Loaded from config.ini - edit it for customization

✅  BTC price is: $ 44,391.8

[i] Running WARden version: 0.101
[i] Newest WARden version available: 0.96

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "warden/__main__.py", line 615, in <module>
    main(debug=debug, reloader=reloader)
  File "warden/__main__.py", line 507, in main
    app = init_app(app)
  File "warden/__main__.py", line 246, in init_app
    from routes import warden
  File "warden/routes.py", line 13, in <module>
    from forms import RegistrationForm, LoginForm, TradeForm
  File "warden/forms.py", line 8, in <module>
    from wtforms.fields.html5 import DateField
ModuleNotFoundError: No module named 'wtforms.fields.html5'

nyxnor, 2022-02-16:
@sovereignhodler Did the installation fail?

> [i] Running WARden version: 0.101
[i] Newest WARden version available: 0.96

You are running a really older version.

Try running with the option `update`.

nyxnor, 2022-02-19:
Havent tried in a while, maybe try https://stackoverflow.com/questions/70068407/modulenotfounderror-no-module-named-wtforms-fields-html5

sovereignhodler, 2022-02-20:
> Havent tried in a while, maybe try https://stackoverflow.com/questions/70068407/modulenotfounderror-no-module-named-wtforms-fields-html5

Downgrading and changing the import line in forms.py worked. However, changing config to port 5050 (conflict with lnbits) runs but can't access the page.

Giving up for now.

Thanks for the help

openoms, 2024-08-10:
closing as outdated

closed 2024-08-10 by openoms