raspiblitz/raspiblitz #1676: Fix BTRFS

rootzoll, 2020-10-16:
[image]

adevoss, 2020-10-16:
Got this same when trying out btrfs. #1668 fixes this.

rootzoll, 2020-10-19:
Merged already in v1.6.1 ... but re-check for a v1.7 release.

holgern, 2020-10-21:
Parts of the problem are fixed by PR #1697 in which the tempmount parameter is changed when the filesystem is btrfs

adevoss, 2020-10-23:
#1706 is in some places the same and in other places different from #1697. #1711 is also part of the approach I took.

joernroeder, 2020-10-26:
running into this as well. Whats the best approch here? just wait for the new release or are the changes already merged and  updatable via `XXupdateScripts.sh`?

rootzoll, 2020-11-11:
Found that when BTRFS is running with a USB RAID1 thumb drive and I switch on the Backup StaticChannel to USB .. it wants to use that RAID1 devcie and format it. Will fix this as part of this issue.

rootzoll, 2020-11-11:
Ok fixed with merge of #1759

rootzoll, 2020-11-12:
TODO for final testing: Run a complete migration with a RC1 of v1.6.2

rootzoll, 2020-11-29:
Before release check that this is solved: https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-735395411

rootzoll, 2020-12-01:
OK BTRFS Migration is now looking good. Closing issue for v1.6.2 release.

closed 2020-12-01 by rootzoll