raspiblitz/raspiblitz #1681: Security Hardening Best Practices

maestro32, 2020-10-18:
hey folks,

Wondering if there is a "best practices" security guide out there (or a post or something) for hardening raspi once everything is loaded up. Ive noticed this code can get a bit finicky so I don't want to mess with it too much. I noticed that Fail2Ban is installed which is great, but wondering if there is other things that can be done to not mess up the raspiblitz code?

Things Ill generally do to a raspi

-add 2fa to SSH
-change SSH port (i know not true security but I like to do this anyway)
-enable UFW and allow only ports I need open. also customize ssh privileges, etc.
-disable root
-eliminate unnecessary services (example: bluetooth) - I think WIFI is already disabled in source code
-maybe LUKS encryption

So, if no guide (i did some quick checking), do you think its ok to do some basic hardening on this thing?

openoms, 2020-10-19:
Thank you for this list. I am planning to put together a RaspiBlitz/ routing node specific security guide as well and planning to touch on most.
Some is already done:
* wifi and bluetooth disabled: https://github.com/rootzoll/raspiblitz/blob/026e1f38beb22c04af6a7943ec7d0ee71d747bfb/build_sdcard.sh#L824
* UFW is active and only specific ports are open, closing ports when services are uninstalled
* hidden services are also removed when services are uninstalled
* disabling root is a bigger issue. By default `admin` (and the optional `joinmarket`) users have passwordless sudo access to be able to perform installations and read password without much user interaction. Asking a password each time would be a very bad UX I think and might break scripts (so needs to be thoroughly tested).

* changing the SSH port is another UX hurdle and as you write it does not add real security 
* I presume 2FA for SSH is similar to this: https://github.com/openoms/joininbox/blob/master/FAQ.md#log-in-through-ssh-using-a-hardware-wallet , could add it to the FAQ here, but I think it needs to remain a manual process
* LUKS encryption would be welcome in the future, but a big challenge to get it right. Just to note all wallets and user interfaces are password protected so this has more privacy implications (in the case of physical theft) than security

rootzoll, 2021-04-01:
Best would be to start putting this into a `SECURITY.md` into the repo ... basic chapters should be "Online Security" & "Physical Security" - a small write up should be a good start for the v1.7 final release - to be improved  than over time.

openoms, 2021-04-22:
Came accross this document: https://madaidans-insecurities.github.io/guides/linux-hardening.html

describing many good paractices which we can use, but also need testing.
Some we already apply. 
Leaving it here to dig deeper any time there are resources available.

nyxnor, 2021-06-01:
### [nix-bitcoin](https://github.com/fort-nix/nix-bitcoin#security)

>    Simplicity: Only services you select in configuration.nix and their dependencies are installed, packages and dependencies are pinned, support for doas (sudo alternative), most packages are built from the NixOS stable channel, with a few exceptions that are built from the nixpkgs unstable channel, builds happen in a sandboxed environment, code is continuously reviewed and refined.

* Only services you select in configuration.nix and their dependencies are installed -> DONE
* packages and dependencies are pinned -> DONE Mostly(most dependencies are pinned, packages are from the repository so always latest version, or if from other source, they are pinned. I dont know to each degree this should be done to pin everything)

>    Integrity: Nix package manager, NixOS and packages can be built from source to reduce reliance on binary caches, nix-bitcoin merge commits are signed, all commits are approved by multiple nix-bitcoin developers, upstream packages are cryptographically verified where possible, we use this software ourselves.

* Integrity -> DONE

>    Principle of Least Privilege: Services operate with least privileges; they each have their own user and are restricted further with systemd options, RPC whitelisting, and netns-isolation. There's a non-root user operator to interact with the various services.

* they each have their own user -> DONE
* [restricted further with systemd options](https://github.com/fort-nix/nix-bitcoin/blob/master/pkgs/lib.nix) -> MISSING
* [RPC whitelisting](https://github.com/fort-nix/nix-bitcoin/blob/master/modules/bitcoind-rpc-public-whitelist.nix) -> MISSING (But this restricts RPC usage, would need to see if there is a command missing that is used in the scripts
* [netns-isolation](https://github.com/fort-nix/nix-bitcoin/blob/master/modules/netns-isolation.nix) -> MISSING
* There's a non-root user operator to interact with the various services -> DONE (Admin and Bitcoin have root access, other users are not root)

>    Defense-in-depth: nix-bitcoin is built with a hardened kernel by default, services are confined through discretionary access control, Linux namespaces, dbus firewall and seccomp-bpf with continuous improvements.

* [hardened kernel by default ](https://github.com/NixOS/nixpkgs/blob/master/nixos/modules/profiles/hardened.nix)-> MISSING
* services are confined through discretionary access  -> MISSING
* Linux namespaces -> MISSING
* [dbus firewall](https://github.com/fort-nix/nix-bitcoin/blob/master/modules/security.nix) -> MISSING
* seccomp-bpf -> MISSING

### [cryptoanarchy-deb-repo-builder](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/blob/master/docs/security/faq.md)

* a clear policy on forks. -> MISSING (Policy missing, it is possible to choose the version)
* use btc-rpc-proxy to protect bitcoind out-of-the-box. -> MISSING
* RPC filtering for bitcoind, long before it was available in bitcoind itself. This was later implemented in nix-bitcoin, which is the only other project that does this.  -> MISSING
* special macaroon that doesn't allow BTCPayServer to steal sats from LND -> MISSING??
*  implements SSO for several applications. The application handling authentication is intentionally much, much simpler than BTCPayServer, thus the attack surface is lower.  -> MISSING
* sandboxes cookies of different apps to avoid them being passed between each-other. There's no other known project that does this.  -> MISSING
* The keys of foreign developers are verified using memory-safe GPG implementation sqck. No other known project uses sqck.  -> MISSING
* All official builds are performed in a dedicated Qubes VM. No other known project does this.  -> MISSING
* deterministic builds. (Only ThunderHub is undeterministic here.)  -> MISSING (This depends on the packages version, [discussed here](https://github.com/rootzoll/raspiblitz/issues/917)
* declarative approach -> MISSING

openoms, 2021-06-01:
Re systemd hardening I am planning to add these to all systemd services under `[Service]`:
```
# Hardening measures
PrivateTmp=true
ProtectSystem=full
NoNewPrivileges=true
PrivateDevices=true
```

Testing currently with c-lightning and the parallel services.
Will be tested in the new release branch together with the setup process refactoring.

openoms, 2021-06-04:
Another one: https://pimylifeup.com/setup-2fa-ssh/
just applied 2FA on my ssh connection in 2 mins.

works on top passwords and ssh keys also.

xanoni, 2021-07-05:
https://github.com/keks24/raspberry-pi-luks

A great & recent LUKS guide that also explains how to include Dropbear in the initrd (e.g., for remote reboots and for fully headless setups).

My main question is how it impacts SSD read/write performance, as well as CPU load. I couldn't find any benchmarks. The "adiantum" algorithm is certainly faster than AES, but I think we're talking maybe about a 20-30% difference, not 300%. (To be confirmed.)

xanoni, 2021-07-05:
Here's some info the author provided to me today: https://github.com/keks24/raspberry-pi-luks/issues/2#issuecomment-874090763

openoms, 2021-07-06:
great info @xanoni! The encryption should be first tested manually before implementing optionally and maybe as default in the future. Also looking into it for: https://github.com/openoms/joininbox/issues/24

Next question: 
Does anyone think mounting the disk with `noexec` adds any security or has some otehr benefit I am not aware of?
Discuss in: https://github.com/rootzoll/raspiblitz/issues/2405

xanoni, 2021-07-06:
> great info @xanoni! The encryption should be first tested manually before implementing optionally and maybe as default in the future. Also looking into it for: [openoms/joininbox#24](https://github.com/openoms/joininbox/issues/24)

I've run some benchmarks. See my referenced ticket (https://github.com/keks24/raspberry-pi-luks/issues/2#issuecomment-874510106).

Without encryption: ~200MB/s write and ~250MB/s read.
With encryption: ~55MBs/s write and ~85MB/s read.

Besides, all 4 CPUs are 100% utilized when encryption is turned on.

There's definitely a sacrifice involved.

xanoni, 2021-07-06:
Actually it was 85MB/sec write and 55 read, not the other way around. I added more benchmarks in the linked issue (4 different ciphers and key sizes). 

AES-XTS with 256bit (~128bit traditional AES) looks like an OK solution, speed-wise. However, I don't know yet how secure it is these days / which threat models it still satisfies.

More research to be done.

xanoni, 2021-07-06:
FYI, I'll likely go with aes-xts-plain64 and a 384 bit key.

See here: https://github.com/keks24/raspberry-pi-luks/issues/2#issuecomment-874935852

xanoni, 2021-07-08:
FYI, more and updated benchmarks added ;-)

xanoni, 2021-07-14:
I'm running LUKS in production now with bitcoind, monerod, lnd, btcrpcexplorer, mempool, joinmarket, etc. It's great. I'm sure it's slower than without encryption, but it's fast enough. CPU load is not as bad as assumed.

I reckon that these are qualitative statements, but my sense is that most humans are cool with the performance, especially given the extra peace of mind

xanoni, 2021-07-14:
Another comment on security ... in my setup I have separate users for bitcoind, lnd, joinmarket, btcrpcexplorer, etc... and then I fine granularly gave them `r` or `rw` access to the stuff that they truly need. E.g., things in the Bitcoin data dir should only be written by the bitcoin user.

xanoni, 2021-07-14:
> * disabling root is a bigger issue. By default `admin` (and the optional `joinmarket`) users have passwordless sudo access to be able to perform installations and read password without much user [...]

I was also struggling with this, but what you can do is to specify the specific commands in `/etc/sudoers` that each user is allowed to run without providing a password. I didn't find many commands that were truly needed by my scripts.

nyxnor, 2021-08-15:
https://fedoraproject.org/wiki/Changes/PrivateDevicesAndPrivateNetwork

systemd versions provide two per-service switches PrivateDevices=yes/no and PrivateNetwork=yes/no which enable services to run without access to any physical devices in /dev, or without access to kind of network sockets. So far this has seen little use in Fedora, and with this Fedora Change we'd like to change this, and enable these for all long-running services that do not require device/network access.

xanoni, 2021-08-16:
Interesting. What are examples of Raspiblitz services that could use `Privatenetwork=yes`? Don't they all use the network?

xanoni, 2021-08-16:
By the way, has anyone ever looked at `nspawn` and whether it might make sense for Raspiblitz / Raspibolt?

> systemd-nspawn may be used to run a command or OS in a light-weight namespace container. In many ways it is similar to chroot(1), but more powerful since it fully virtualizes the file system hierarchy, as well as the process tree, the various IPC subsystems and the host and domain name.

https://www.freedesktop.org/software/systemd/man/systemd-nspawn.html

It works on Pi, unfortunately this project here that leveraged it is EOL: https://github.com/sakaki-/raspbian-nspawn-64

rootzoll, 2021-08-16:
A lot of good ideas in this thread .. anything particular we should target for v1.7.1 release?

xanoni, 2021-08-17:
>  what you can do is to specify the specific commands in `/etc/sudoers` that each user is allowed to run without providing a password

Regarding my above comment, this would be helpful to make sure we're doing it properly: 

https://blog.compass-security.com/2012/10/dangerous-sudoer-entries-part-1-command-execution/
https://blog.compass-security.com/2012/10/dangerous-sudoers-entries-part-2-insecure-functionality/
https://blog.compass-security.com/2012/10/dangerous-sudoers-entries-part-3-permissions/
https://blog.compass-security.com/2012/10/dangerous-sudoers-entries-part-4-wildcards/
https://blog.compass-security.com/2012/10/dangerous-sudoers-entries-part-5-recapitulation/

It's almost a decade old, but the parts that I've already read still apply in 2021.

rootzoll, 2021-10-10:
some of the measures were implemented in v1.7.1 ... I wil push this issue for further discussing to v1.8.0

nyxnor, 2021-11-03:
If someone could help make a table to compare the node options on security measures based on https://github.com/rootzoll/raspiblitz/issues/1681#issuecomment-852154427 would be great.

nyxnor, 2021-12-03:
See [lynis](https://github.com/CISOFY/lynis) to audit the system. There is a lot of good tips. Run from git or install it from source, the deb package is old.