raspiblitz/raspiblitz #1683: BTRFS - How to change the flash drive

jodobear, 2020-10-19:
My Blitz is running smooth as a butter on BTRFS. I have a question: What if i need to change the flash drive? Looked up the documentation but, there isn't a clear answer. Objective is only to swap the flash drive for a new one in case of a failure or not. Reason, need to know the implications and how to do it.

Here's a thought:

1. Format a new flash drive to BTRFS.
2. Copy the contents of the connected flash drive to new one.
3. Plug the new drive and power on.

Will there be a mount point issue? Will the Blitz recognize it as the old drive without any issues? If not then how can we do this.

OR, say you are upgrading or just flash the SD card and go through the recovery setup, will it recognise the new flash drive as the old flash drive?

We can add this to the documentation.

rootzoll, 2020-10-19:
Changing the HDD is called "Migration" - see: https://github.com/rootzoll/raspiblitz/blob/v1.6/README.md#import-a-migration-file ... ist this what you were searching for?

jodobear, 2020-10-20:
No, my Blitz is running fine, data is safe. I only want to change the USB - everything else stays the same.

openoms, 2020-10-24:
After a full backup (MIGRATION FILE)
you can try:
* check the RAID status:  
`$ sudo config.scripts/blitz.datadrive.sh status`
```
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=1
hddRaspiData=1
isSSD=1
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=32212237312
hddGigaBytes=29
hddUsedInfo='13% & 50%'

# RAID
isRaid=1
raidHddDev='sda1'
raidUsbDev='sdb'

# SWAP
isSwapExternal=1
SwapExternalPath='/mnt/temp/swapfile'
```

*Remove the pendrive from the RAID:  
`sudo config.scripts/blitz.datadrive.sh raid off`
```
# RAID - Removing raid drive to RaspiBlitz data drive
# removing USB DEV from RAID
ERROR: error removing device 'missing': no missing devices found to remove
# OK - RaspiBlitz data is not running in RAID1 anymore
```

* (before upgrading take note of these steps: https://github.com/rootzoll/raspiblitz/issues/1702 to avoid the recovery process getting stuck)
* power off and unplug the pendrive
* proceed to flashing the SDcard and upgrade

When synced up again I successfully used these steps.
* power off and attach the pendrive
* when started run:  
`lsblk`
the pendrive should be there as `sdb`
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk 
├─sda1        8:1    0    30G  0 part /mnt/hdd
├─sda2        8:2    0 867.5G  0 part /mnt/storage
└─sda3        8:3    0    34G  0 part /mnt/temp
sdb           8:16   1 119.3G  0 disk 
mmcblk0     179:0    0  29.7G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.5G  0 part /
```

```
$ sudo config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=1
hddRaspiData=1
isSSD=1
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=32212237312
hddGigaBytes=29
hddUsedInfo='3% & 50%'

# RAID
isRaid=0
raidCandidate[0]='sdb 119 GB SanDisk'
raidCandidates=1

# SWAP
isSwapExternal=1
SwapExternalPath='/mnt/temp/swapfile'
```

* add the pendrive back to the RAID (took me ~ 15 min):  
`$ sudo config.scripts/blitz.datadrive.sh raid on sdb`
```
# RAID - Adding raid drive to RaspiBlitz data drive
# adding sdb as BTRFS raid1 for /mnt/hdd

~ 15 MIN

# OK - sdb is now part of a RAID1 for your RaspiBlitz data
```

* check the status again:
```
$ sudo config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=1
hddRaspiData=1
isSSD=1
datadisk='sda'
datapartition='sda1'
hddBlocksBitcoin=1
hddBlocksLitecoin=0
hddBytes=32212237312
hddGigaBytes=29
hddUsedInfo='13% & 50%'

# RAID
isRaid=1
raidHddDev='sda1'
raidUsbDev='sdb'

# SWAP
isSwapExternal=1
SwapExternalPath='/mnt/temp/swapfile'
```

Note this entry about the type of pendrive to use (USB SSD is recommended):
https://github.com/rootzoll/raspiblitz/issues/924

jodobear, 2020-10-25:
Wow! Thank you for the detailed response @openoms . Might have to get a new USB drive. If i upgrade to 1.6.1 now and follow the steps later to change the drive, it should work, correct?

openoms, 2020-10-25:
> Wow! Thank you for the detailed response @openoms . Might have to get a new USB drive. If i upgrade to 1.6.1 now and follow the steps later to change the drive, it should work, correct?

I was making some custom steps myself so difficult to guarantee therefore do make sure you generate a migration file beforehand! Also is best to use a new SDcard for the update, so you can fall back to the old version.

Probably best to try restarting with the old system after removing the raid to make sure it boots and don' delete the USB drive before that test.

nyxnor, 2021-05-17:
Please close this issue.
It is very informational, maybe let it marked as "documentation"

closed 2021-05-17 by rootzoll