raspiblitz/raspiblitz #1685: press enter after shell script executions

ChuckNorrison, 2020-10-19:
Hey, 

i would ask you for a more efficient way to get around common error behaviours which will occur always while automatic script installations. This is expected, cause we are using a lot of stuff together and a lot of functionality takes place. It is not possible to test everything within an Release and keep the Blitz up 2 date for users.

Currently the shell scripting is built with lack of installation verification, so it will happen that a script installation will result in success, even the installation process throwed some errors.

I would recommend to pause the installation script after it is done, to get the possibility for users to check the console on that execution and press enter to get back to menu.

In this case any failure of installation would be visible while installation and users can easily copy some interesting lines from here without search for hidden logs.

What do you think?

openoms, 2020-10-19:
Some scripts already have some installation verification like:
https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/bonus.loop.sh#L120
https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/bonus.rtl.sh#L108
improvements are welcome on this.

Need to be careful pausing the scripts because it would stall the automated SDcard recovery on updates.

Showing the console output better when installing services would be good, the info menu is displayed anyway after most services are installed form the menu.
Also should minimize the usage of `clear` commands to have the option to scroll up to check the outputs.

ChuckNorrison, 2020-10-19:
if we could tell the scripts they got called within recovery process or from menu by user, this pause/clear behaviour could be handled in a safe way. Maybe the recovery process could call the underlying scripts with a parameter to not trigger the user interaction.

Thanks for the reply, so i will make a pull request if i found something useful :+1: Maybe some more ideas to improve error handling.

openoms, 2020-10-19:
Agree that exposing the output of the scripts after installation from the menu is probably even more useful than showing the info menu when finished. 
Adding a simple option to the bonus install scripts to discern between the script being run from the menu or by the recovery script in a non-interactive mode is a good idea.

ChuckNorrison, 2020-10-19:
I tested something for LNbits here with some simple `read` calls for user input:
https://github.com/ChuckNorrison/raspiblitz/commit/6759ceefa72d6822e617bc94cb713062060bea2d

nyxnor, 2021-04-18:
@ChuckNorrison What do you think of instead of asking user confirmation on parts of the script, just at the end if he agrees with the logs above. If not, uninstall service and report on github. 
* Also, this happens when there is a new version out and blitz pinned version need to be changed mannualy by the user if he wants before Blitz releases it. 
* A problem like this happens when resetting macarrons and there is no folder for it to be saved, by the end of it, still displays `OK - DONE`. [This is the folder ](https://github.com/rootzoll/raspiblitz/blob/755b7460ca7d8f6aa614748e534ec0941fc891ea/home.admin/config.scripts/lnd.credentials.sh#L57). Never understood why this happened, just saw people solving this by flashing the SD CARD, if you solve this would be optimal your way.
https://github.com/rootzoll/raspiblitz/blob/755b7460ca7d8f6aa614748e534ec0941fc891ea/home.admin/config.scripts/lnd.credentials.sh#L138
* Now a lot of services can be updated on the menu separately, which solves a lot of problems due to outdated code.

ChuckNorrison, 2021-04-18:
Cause it's all handled in different shell scripts, a kind of teardown could take place on installation scripts to gather logs and print if some debug mode is enabled (maybe debug set in raspi config to not confuse people with.

After some time on the Blitz iam fine to execute install scripts manually for debugging purpose. ✌️