raspiblitz/raspiblitz #1691: Provisioning of "DYNAMIC DOMAIN" gets stuck

rootzoll, 2020-10-20:
On last log line "# start to create new generate new TLSCert" recovery got stuck ... check if systematic or edge case.

rootzoll, 2020-10-20:
manual fix: ssh in (with password 'raspiblitz') and deactivate dynamic DNS. turn off. reflash sd card again and recover.

openoms, 2020-10-21:
Had another user reporting this.

can deactivate the dynamicDNS after typing:
`menu`
or edit the conf manually with:
`sudo nano /mnt/hdd/raspiblitz.conf`

https://t.me/RaspiBlitzFAQ/145

openoms, 2020-10-21:
@rootzoll shall we recommend to deactivate the dynDomain before updating to avoid getting stuck?

rootzoll, 2020-10-26:
@openoms Yepp .. until we have a v1.6.2 - targeting mid november.

geco91, 2020-10-28:
Will check what is going wrong on the logs when updating with dyndomain on.

Also maybe thats a good point to optimize the `internet.dyndomain.sh` config script with:
- have a `status` parameter
- fail on `on` when no domain is given
- make sure outputs are comment safe
- check if can be set thru settings menu correctly

geco91, 2020-10-28:
These are the logs where it blocks the update/provisioning:

````
Provisioning DYNAMIC DOMAIN - run config script
number of args(2)
making sure services are not running
# switching the DynamicDNS ON
# dynDomain(home.rotzoll.de)
# dynUpdateUrl()
# lnd.tlscert.sh domain-add
# OK the domain was already added lnd.conf
# DynamicDNS is now ON
# checking if LND is running
# making sure services are not running
# keep old tls data as backup
# start to create new generate new TLSCert
```

Because LND is not running the TLS cannot get created and it blocks. But during recovering there is no need to recreate the TLS - the macaroons and certs from backup will be used anyway after setup. Will check script again to make this work again.

geco91, 2020-10-28:
Can confirm that #1721 fixed the "getting stuck on update" problem.

rootzoll, 2020-11-28:
Tested with RC2 - working fine with dynamic domain on. Closing issue for v1.6.2 release.

closed 2020-11-28 by rootzoll