raspiblitz/raspiblitz #1692: Trouble with my full node set up

shaunstar, 2020-10-20:
Hi, im not sure if this is the right place to post. Apologies if not, thanks if someone can help me :-)

I had the Raspiblitz up and running for a week or so (my first node!) but it stopped responding one time when i tried to ssh in (i think it may have been unplugged accidentally while running).

I took the sd card out, remounted the Raspiblitz v1.6 image and then returned it to the Rasperry pi(4).

I followed the instructions and it seemed to reinstall, i selected to use existing blockchain found on the HD. It seemed to get stuck, i selected somethign along the lines of delete HDD data (but not the block chain i think).

It wouldnt work again, have tried to reinstall from sd card again but get stuck with the below. Something to do with missing bitcoind(?).

Any help greatfully received on what i could try to fix it.

Thanks,
Shaun

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.0

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.0
chainnetwork: bitcoin / main
 19:08:40 up 2 min,  2 users,  load average: 2.49, 0.82, 0.30

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2020-10-20 19:06:25 BST, end at Tue 2020-10-20 19:08:40 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Tue 2020-10-20 19:06:25 BST, end at Tue 2020-10-20 19:08:41 BST. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

*** LAST NGINX LOGS ***
sudo journalctl -u nginx -b --no-pager -n20
-- Logs begin at Tue 2020-10-20 19:06:25 BST, end at Tue 2020-10-20 19:08:41 BST. --
Oct 20 19:07:38 NoDramaLama systemd[1]: Starting A high performance web server and a reverse proxy server...
Oct 20 19:07:39 NoDramaLama nginx[721]: nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)
Oct 20 19:07:39 NoDramaLama nginx[721]: nginx: configuration file /etc/nginx/nginx.conf test failed
Oct 20 19:07:39 NoDramaLama systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Oct 20 19:07:39 NoDramaLama systemd[1]: nginx.service: Failed with result 'exit-code'.
Oct 20 19:07:39 NoDramaLama systemd[1]: Failed to start A high performance web server and a reverse proxy server.
--> CHECK CONFIG: sudo nginx -t
nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)
nginx: configuration file /etc/nginx/nginx.conf test failed

- TOUCHSCREEN is OFF by config
- Loop is OFF by config
- RTL is OFF by config
- Electrum Rust Server is OFF by config
- BTCPayServer is OFF by config
- LNbits is OFF by config
- Thunderhub is OFF by config
- SPECTER is OFF by config

*** MOUNTED DRIVES ***
Filesystem     Type     1K-blocks      Used Available Use% Mounted on
/dev/root      ext4      30358348   2904924  26170116  10% /
devtmpfs       devtmpfs   1966332         0   1966332   0% /dev
tmpfs          tmpfs      2000124         0   2000124   0% /dev/shm
tmpfs          tmpfs      2000124      8676   1991448   1% /run
tmpfs          tmpfs         5120         4      5116   1% /run/lock
tmpfs          tmpfs      2000124         0   2000124   0% /sys/fs/cgroup
tmpfs          tmpfs        32768         0     32768   0% /var/cache/raspiblitz
/dev/mmcblk0p1 vfat        258095     52239    205857  21% /boot
/dev/sda1      ext4     960348184 334593076 576902372  37% /mnt/hdd
tmpfs          tmpfs       400024         0    400024   0% /run/user/1000
tmpfs          tmpfs       400024         0    400024   0% /run/user/1001

*** WIFI Info ***
activated=0
connected=0
localip=''

*** HARDWARE TEST RESULTS ***

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.93'
tempCelsius='53.5'
uptime=128
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
startcountLightning=0
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

shaunstar, 2020-10-21:
Ah, i have purchased a different hard drive https://www.amazon.co.uk/Seagate-Expansion-Portable-PlayStation-STEA1000400/dp/B00TKFEEAS

openoms, 2020-10-21:
Taht might need too much power. Try with a tested model or with some dedicated power source for the drive. A powered USB hub for example.

nyxnor, 2021-04-18:
@shaunstar As there were no more comments, did you solve it? If yes, please close this issue.

nyxnor, 2021-05-17:
@rootzoll Please close this.

closed 2021-05-17 by rootzoll