raspiblitz/raspiblitz #1694: Test seed to validate it is correctly back-uped

petrhejna, 2020-10-20:
Hi, I wonder if there is an option to test the seed. It would be nice feature to validate that seed backup is done correctly.

rootzoll, 2020-10-26:
This could be a good UX feature once we have a WebUI .. will list idea under "Future Release Collector".

openoms, 2020-10-27:
LND does not show the seed again, so cannot do this for now without triggering an onchain restore, but there could.be something with guggero's chantools:
For example could compare the BIP32 root key from the seed:  
https://github.com/guggero/chantools#showrootkey  
and the wallet.db:  
https://github.com/guggero/chantools#walletinfo

Importing to Electrum/Specter might be interesting too:
https://github.com/openoms/lightning-node-management/blob/master/RestoreLNDonchainFundsInElectrum.md

All of the above are exposing the private keys so should only be used for emergency measures and offline.
Overall validating the seed with external tools adds too big security risks, so would prefer lnd doing it (and would still need to type/show it again).

petrhejna, 2020-10-27:
Right now, I had to use another wallet (=external tool) to test my seed. I did assertion by "seeing the correct transaction".

I believe there is no need to show seeds again, just to compare keys.

Just a sidenote: I believe that some wallets do store seed backups in encrypted files (encryption key is "the wallets password").

petrhejna, 2020-10-27:
I would even claim that best practice would be:

1. Show the seed, tell user to write it.
2. Force user to type it again on another screen.
3. Allow to skip this only with "I am just testing" confirmation. 
4. Show crazy big alert every time user logs in, asking to verify backup.
5. If seed backup cannot be verified, guide user to NOT use wallet for anything serious and tell him to "reset the wallet and start again with new wallet.

openoms, 2020-10-27:
@petrhejna I agree we could store the aezeed lnd seed in an encrypted format with PasswordC as the password since the lnd wallet.db already contains the BIP32 HD root key protected by the wallet password (=PasswordC). 
Currently there is no other way to extract the seed words again from the wallet so makes sense to store it securely to be able to review and test them.

openoms, 2020-10-27:
As it is now chantools is interactive CLI and has no option for `stdin` so I don't think it can be scripted for a UI. 
For now we can add chantools as a CLI only option and give some guidance about how to do the seed test. 

The lnd seed can be stored in an encrypted file as part of https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/lnd.initwallet.py

petrhejna, 2020-11-05:
I have found here in doc describing channel backup tthis phrase: "The file itself is encrypted by your word seed" (https://github.com/rootzoll/raspiblitz/blob/v1.6/README.md#backup-for-on-chain---channel-funds)

Is it the wallet seed? How does it updates if the seed is not available? Or is it encrypted by wallet itself (private key) so id can be decrypted by key derived from the seed?

openoms, 2020-11-05:
> I have found here in [doc describing channel backup](https://github.com/rootzoll/raspiblitz/blob/v1.6/README.md#backup-for-on-chain---channel-funds) this phrase: _"The file itself is encrypted by your word seed"_

Yes, would more accurate (but more complicated) to say that the channel.backup is encrypted with the master private key of the wallet which is either stored in the the wallet file or derived from the seed (which is not stored).

nyxnor, 2021-04-18:
Seed confirmation on the merged pull request by openoms above, please close this issue.

closed 2021-04-19 by rootzoll