raspiblitz/raspiblitz #1695: Mempool: Fix order of blocks

rootzoll, 2020-10-21:
see: https://twitter.com/KevinR4v/status/1318732809050226688?s=20

rootzoll, 2020-10-21:
Check with @wiz what can be changed in the config to fix this.

wiz, 2020-10-21:
@rootzoll are you able to reproduce this issue? I checked your scripts and I don't see anything strange in the configuration that would cause this...

openoms, 2020-10-21:
Could check if the timestamp of those particular blocks are off and if the sorting is made by time not blockheight: https://twitter.com/openoms/status/1318831664303378432?s=19

wiz, 2020-10-21:
do you have reason to believe this is a bug in mempool? since nobody has ever reported to us I assume it was just a configuration problem with how raspiblitz is installing mempool. has anyone been able to reproduce the issue since this was reported?

openoms, 2020-10-21:
I did not see this happen, just trying to guess the cause.

kravens, 2020-10-21:
I reported this small bug I noticed in the mempoolspace instance running on my raspiblitz, it does not show any incorrect block anymore after running for some hours. So I think it could have been due to the RaspiBlitz beeing offline for a bit whilst upgrading and then immediately after starting mempoolspace service. Showing the blocks in the order the backend could fetch them from the node, and it takes a little while to catch up to the chain tip.

wiz, 2020-10-21:
okay, thanks for letting us know, I suppose your node was just out of sync because nobody has ever reported this to us before. since it's no longer reproducing i would suggest to close this issue for now, but let us know it happens again

nyxnor, 2021-04-18:
@kravens Please close this.

kravens, 2021-04-18:
I now see blocks in correct order again after the update, rootzoll is the owner of this issue (I reported about on twitter) and can close it.

closed 2021-04-19 by rootzoll