raspiblitz/raspiblitz #1699: "Stuck at Specter"

grnqrtr, 2020-10-21:
I noticed this in the Telegram chat before attempting the update to 1.6.1:
[image]

I then noticed that I also got "stuck at specter": 
[image]

I opened another ssh session and checked bitcoind status and also tried restarting bitcoind with `sudo systemctl restart bitcoind` as well as `start` and `stop`. A while later I realized in my original tab that the update seemed to be continuing after it said `Job for bitcoind.service cancelled.`
[image]

Things seem to be working okay now, but there was this hang up.

holgern, 2020-10-21:
I also had this problem and and to restart again. It had worked on the second try.

openoms, 2020-10-21:
Thanks for reporting!
Yes it seems it is getting stuck during recovery on restarting bitcoind at https://github.com/rootzoll/raspiblitz/blob/7f2e12472e4f3cba68db53a6879c1284860e9ad1/home.admin/config.scripts/bonus.cryptoadvance-specter.sh#L271

This was planned to be changed for the next release anyway: https://github.com/rootzoll/raspiblitz/issues/1639
so tagging for v1.6.2.

For now might be the best to deactivate Specter before starting the update and reinstall afterwards.

openoms, 2020-10-21:
To solve it while recovering run:  
`sudo systemctl stop bitcoind`

geco91, 2020-10-28:
Can confirm that on my test machine with #1699 it fixed the getting stuck when `blockfilter=0` before update.

openoms, 2020-10-29:
This should be solved by this: https://github.com/rootzoll/raspiblitz/pull/1722. Thanks @geco91!

theansweris420, 2020-11-02:
had the same issue. after the final reboot the specters  .onion adress was a new one. i didnt save the first. just the seed. is there a change of resotring it only with the seed?

openoms, 2020-11-13:
> had the same issue. after the final reboot the specters .onion adress was a new one. i didnt save the first. just the seed. is there a change of resotring it only with the seed?

Specter does not store private keys so there is no seed for it. The onion address changing does not affect the stored data.

You might be thinking of the seed given for the LND wallet. That is only enough to restore the onchain balance, not the channels. See the FAQ for details.

rootzoll, 2020-11-28:
OK tested with RC2 and looks good. Closing issue for v1.6.2 release.

closed 2020-11-28 by rootzoll