raspiblitz/raspiblitz #1700: After migrating to 1.6.1, the electrs service could not be started

holgern, 2020-10-21:
After migrating, electrs could not be started:
```
Oct 21 17:00:11 raspberrypi electrs[4739]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { message: "IO error: while open a file for lock: /mnt/hdd/app-storage/electrs/db/mainnet/LOCK: Permission d
enied" }', src/libcore/result.rs:1165:5
```
reason was the wrong user and group which could be fixed by:
```
sudo chown -R electrs:electrs /mnt/hdd/app-storage/electrs
```

openoms, 2020-10-21:
Thanks for reporting!
Wonder what is went wrong here since this line
`sudo chown -R electrs:electrs /mnt/hdd/app-storage/electrs`
is part of the script: https://github.com/rootzoll/raspiblitz/blob/7f2e12472e4f3cba68db53a6879c1284860e9ad1/home.admin/config.scripts/bonus.electrs.sh#L301
Were you updating from v1.6 where Electrs was used?
What was the user and group of the directory previously?

openoms, 2020-10-21:
If you are using a BTRFS setup this symlink might be the cause of the problem:
```
$ sudo ls -la /mnt/hdd/app-storage
lrwxrwxrwx 1 bitcoin bitcoin 24 Aug 11 09:51 /mnt/hdd/app-storage -> /mnt/storage/app-storage
```

rootzoll, 2020-10-26:
TODO: Check recovering with BTRFS+ElectRS

openoms, 2020-10-27:
Another report, but now without BTRFS:
hi, updated to V1.6.1 using the steps in the documentation. Now my Explorer, Mempool viz and Electrum connection stopped working.. Almost as if the Bitcoin node is not connected anymore
https://termbin.com/pvc14

rootzoll, 2020-11-12:
@openoms that last one seems a problem with electrs - see:
```
Oct 26 14:33:30 raspberrypi electrs[3674]: 2020-10-26T14:33:30.325+00:00 - DEBUG - opening DB at "/mnt/hdd/app-storage/electrs/db/mainnet"
Oct 26 14:33:30 raspberrypi electrs[3674]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Error { message: "IO error: while open a file for lock: /mnt/hdd/app-storage/electrs/db/mainnet/LOCK: Permission denied" }', src/libcore/result.rs:1165:5
```
There is a "permission denied" when the electrs user process tries to open the database ... hmm, but on eletcrs install this directory should be owned by the user: sudo chown -R electrs:electrs /mnt/hdd/app-storage/electrs 

Not sure how this can happen yet.

rootzoll, 2020-11-12:
I tested the this in a recover/update on a btrfs setup with the latest scripts - all look good:
[image]

OK I merged PR #1766 that will delete the LOCK file on every bootup and also gives electrs more time to shutdown if RaspiBlitz is turn off. Putting this in final testing when recover/update with electrs on.

rootzoll, 2020-11-12:
OK update/recovery worked. Closing this issue for v1.6.2 release. Please reopen issue if the problem shows up again.

closed 2020-11-12 by rootzoll