raspiblitz/raspiblitz #1702: TODO before updating to v1.6.1

openoms, 2020-10-21:
There are more reports from getting stuck during recovery when updating to v1.6.1 collected some caveats to be run before updating:
```
config.scripts/internet.dyndomain.sh off
config.scripts/bonus.cryptoadvance-specter.sh off
sudo chown -R electrs:electrs /mnt/hdd/app-storage/electrs
```
* Deactivates the Dynamic Domain: https://github.com/rootzoll/raspiblitz/issues/1691
* Switches the Specter service off (the data is preserved and will be restored when reinstalled): https://github.com/rootzoll/raspiblitz/issues/1699
* fixes the electrs database owner just in case: https://github.com/rootzoll/raspiblitz/issues/1700

If running with BTRFS consider updating without the USB drive attached or apply this PR: https://github.com/rootzoll/raspiblitz/pull/1697/files

geco91, 2020-11-01:
Also check if nginx is running still fine when RTL & LNbits are installed ... had a problem on one test machine, but not sure if that is also a problem on a fresh machine.

rootzoll, 2020-11-12:
Ok all the mentioned problems have a seperate issue and the update/recovery worked with RTL & LNbits. So closing this issue for v1.6.2 release.

closed 2020-11-12 by rootzoll