raspiblitz/raspiblitz #1703: ssh admin@<IP> "Connection refused"

petrhejna, 2020-10-21:
After upgrade I got into this situation.

```
ssh admin@<IP>
ssh: connect to host <IP> port 22: Connection refused
```

How I got there:
1. I have backed lnd-rescue
2. I have turn raspiblitz off
3. I have flashed SD card with new raspiblitz `raspiblitz-v1.6.1-2020-10-20.img.gz`
4. When I tried to login to SSH it failed multiple times.
5. Before I realized, I should try default password, it started to produce this error
6. I am stucked.

What I checked:
* I have check the IP is correct in my router.
* The raspiblitz can be pinged.
* I have tried https://github.com/rootzoll/raspiblitz/blob/v1.5/FAQ.md#how-can-i-repair-my-ssh-login but with no luck.

Please help.

petrhejna, 2020-10-21:
UPDATE: After some time (= time needed to write this Issue) I have desperately tried again and **it worked**! 

Now I am in the phase when I am looking at `tail -n1000 -f raspiblitz.log` bootstraping. 

I wont close the issue right now, as it may be helpful to others. Not sure if the `ssh.reset` helped and it needed time, OR it just needed time.

openoms, 2020-10-21:
it is a security feature called fail2ban. Bans you for 10 minutes after 3 failed password entries. You can close this issue, there is nothing actionable.

petrhejna, 2020-10-21:
Ah, ok thank you.

closed 2020-10-21 by petrhejna