raspiblitz/raspiblitz #1704: The LND service is running with error.

petrhejna, 2020-10-21:
After upgrade to 1.6.1 I have ended with this error.

```
****************************************
* The LND service is running with error.
****************************************
If you just started some config/setup, this might be OK.

7 RESTARTS DETECTED - LND might be in a error loop
Oct 21 22:57:23 raspberrypi lnd[32526]: unable to initialize tor controller: unable to connect to Tor server: dial tcp 127.0.0.1:9051: connect: connection refused
```

I used Raspiblitz with Tor so it may be connected.

petrhejna, 2020-10-21:
Here is the debug log:
```

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.1
chainnetwork: bitcoin / main
 23:03:02 up 22 min,  2 users,  load average: 1.39, 1.42, 1.26

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-10-21 22:43:01 BST; 20min ago
  Process: 5439 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 5445 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 5498 (bitcoind)
    Tasks: 19 (limit: 4915)
   Memory: 3.1G
   CGroup: /system.slice/bitcoind.service
           └─5498 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Oct 21 22:43:01 raspberrypi systemd[1]: Starting Bitcoin daemon...
Oct 21 22:43:01 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Wed 2020-10-21 22:35:48 BST, end at Wed 2020-10-21 23:03:02 BST. --
Oct 21 22:43:01 raspberrypi systemd[1]: Starting Bitcoin daemon...
Oct 21 22:43:01 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-10-21T22:01:35Z Syncing basic block filter index with block chain from height 242483
2020-10-21T22:01:41Z Pre-allocating up to position 0x1000000 in fltr00007.dat
2020-10-21T22:01:46Z Pre-allocating up to position 0x100000 in fltr00008.dat
2020-10-21T22:01:51Z Pre-allocating up to position 0x200000 in fltr00008.dat
2020-10-21T22:01:56Z Pre-allocating up to position 0x300000 in fltr00008.dat
2020-10-21T22:02:00Z Pre-allocating up to position 0x400000 in fltr00008.dat
2020-10-21T22:02:05Z Pre-allocating up to position 0x500000 in fltr00008.dat
2020-10-21T22:02:06Z Syncing basic block filter index with block chain from height 246213
2020-10-21T22:02:10Z Pre-allocating up to position 0x600000 in fltr00008.dat
2020-10-21T22:02:15Z Pre-allocating up to position 0x700000 in fltr00008.dat
2020-10-21T22:02:20Z Pre-allocating up to position 0x800000 in fltr00008.dat
2020-10-21T22:02:24Z Pre-allocating up to position 0x900000 in fltr00008.dat
2020-10-21T22:02:29Z Pre-allocating up to position 0xa00000 in fltr00008.dat
2020-10-21T22:02:34Z Pre-allocating up to position 0xb00000 in fltr00008.dat
2020-10-21T22:02:37Z Syncing basic block filter index with block chain from height 250048
2020-10-21T22:02:39Z Pre-allocating up to position 0xc00000 in fltr00008.dat
2020-10-21T22:02:43Z Pre-allocating up to position 0xd00000 in fltr00008.dat
2020-10-21T22:02:48Z Pre-allocating up to position 0xe00000 in fltr00008.dat
2020-10-21T22:02:54Z Pre-allocating up to position 0xf00000 in fltr00008.dat
2020-10-21T22:02:59Z Pre-allocating up to position 0x1000000 in fltr00008.dat

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-10-21 22:58:23 BST; 4min 39s ago
  Process: 4972 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 4974 (lnd)
    Tasks: 10 (limit: 4915)
   Memory: 23.7M
   CGroup: /system.slice/lnd.service
           └─4974 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735

Oct 21 22:58:23 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Oct 21 22:58:23 raspberrypi systemd[1]: Started LND Lightning Daemon.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
[... 174 lines cut ...]
localIP='192.168.0.38'
tempCelsius='54.0'
uptime=1369
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=653747
initialSync=0
syncProgress=100.00 
startcountLightning=7
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
```

rootzoll, 2020-10-21:
Please try update/recovery again with a fresh sd card ... maybe internet had a hickup during Tor install.

petrhejna, 2020-10-21:
Uff, it works, Thank you very much.

closed 2020-10-21 by petrhejna

ChuckNorrison, 2020-10-24:
My blitz is announcing his lnd  at 127.0.0.1 as his public ip and port 9735... i dont know why yet.

rootzoll, 2020-10-26:
@ChuckNorrison this sounds like a different problem... please open a seperate issue if you still have that errpor.