raspiblitz/raspiblitz #1712: Timezone

woeisme, 2020-10-23:
Just noticed my node was in the showing the wrong timezone and fixed it with raspi-config.
An option should probably be added to the 'SETTINGS' Menu that launches raspi-config menu for localization / timezone setting.

woeisme, 2021-05-13:
This still seems to be present in 1.7 Release. 
Except now certain options seem to crash raspi-config (such as Atlantic ocean)

rootzoll, 2021-05-18:
I think it would be easy to call just raspi-confi from settings menu - but the question is how to "store" these options so that after an update with a fresh sd card all those options are back as set before?

NilByte, 2021-06-03:
Maybe put it in the RaspiBlitz menu for user to set each time a fresh SD card is loaded?

`sudo timedatectl set-timezone Europe/Berlin`

https://www.thedigitalpictureframe.com/cheat-sheet-raspberry-pi-configuration-settings-terminal-commands/#Where_the_configuration_settings_are_stored

philihp, 2022-09-08:
The default should really be UTC, if only to avoid DST issues.

I just updated from 1.6.3 to 1.8.0, and my block database isn't loading because of this change. I guess I could wait until the last Sunday in October for BST to end, but I'd expect this to be a perennial problem for others.

```
The block database contains a block which appears to be from the future. This may be due to your computer's date and time being set incorrectly. Only rebuild the block database if you are sure that your computer's date and time are correct.
```

rootzoll, 2024-05-30:
PR #4604 is ready and will be part of RaspiBlitz v1.11.1rc1 

Under SSHMENU > SYSTEM > TIME you can now select your time zone and on next recover/update it will be kept.

**For Background:**

Finally settled on using `timedatectl` in the background (will work on every debian based system) and introced a new script called `./config.scripts/blitz.time.sh` that offers 2 parameters:

```
choose-timezone  --> user can choose timezone from list and it gets stored to raspiblitz config
set-by-config    --> resets the time on the RaspiBlitz based on the config
```

PR #4604 (2024-06-09): #1712 Add setting Timezone
closes: #1712
closed 2024-06-09 by rootzoll