raspiblitz/raspiblitz #1715: bad btrfs superblock on update to 1.6.1?

philbertw4, 2020-10-25:
I went to upgrade my raspiblitz to 1.6.1 by burning a new SD card. Unfortunately I was unable to make a migration file backup or anything prior as, oddly, I was unable to SSH into the blitz at all. So, I figured that burning a new SD card would be the best path forward anyway. My blitz has been running with a 32gb usb thumb drive for BTRFS, and the whole thing is behind surge-protected backup/power supply.

Anyway, I followed openom's suggestion of unplugging the USB thumb drive for the upgrade (per the comment in #1702), yet, unfortunately was greeted with this:

```
HDD was already initialized/prepared
Now needs to be mounted

# *** 40addHDD.sh ***

# --> Linking drives/directories
# hddCandidate='sda'
# hddPartitionCandidate='sda2'
# - /home/bitcoin/.bitcoin -> is not a link, cleaning
# - /home/bitcoin/.litecoin -> is not a link, cleaning
# Creating BTRFS setup links
# - linking blockchains into /mnt/hdd
# linking lnd for user bitcoin
# - linking blockchain for user bitcoin
# - linking storage into /mnt/hdd
# - linking temp into /mnt/hdd
# - creating snapshots folder
# OK - all symbolic links build

# --> Adding the data drive to OS ...
# hddCandidate='sda'
# hddPartitionCandidate='sda2'
# hddFormat='btrfs'
# BTRFS: Updating /etc/fstab & mount
mount: /tmp/btrfs: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.
FAIL blitz.datadrive.sh fstab --> mount sda1 failed
Please report issue to the raspiblitz github.
```

So I promptly opened this issue. Due to the fact that I could not SSH in, I suppose this may be less of an upgrade issue and more likely that something else went wrong with my node?

As always, thanks for such a great community and product!

openoms, 2020-10-26:
Did to boot with the old USB plugged in?

openoms, 2020-10-26:
It is probably related to this: https://github.com/rootzoll/raspiblitz/issues/1683#issuecomment-716034494
The above message is because the filesystem is still set up as a RAID with the other disk (the pendrive) missing.

There are two options:

1) Apply this PR: https://github.com/rootzoll/raspiblitz/pull/1697
on the RaspiBlitz type:
`menu`  
go to the UPDATE menu -> PATCH -> PR and put the number 1697 and reboot.
Alternatively boot a Linux OS on your desktop and edit these files on the SDcard:
https://github.com/rootzoll/raspiblitz/pull/1697/files

2) Try switch the RAID off for the update:
Best to try in this order (depending what you have or what works):
* boot with the old pendrive attached and your old 1.6 SDcard
* boot with the old pendrive attached with the 1.6.1 SDcard
* boot without the old pendriv and 1.6.1 SDcard
In any case when you can get to the command line first run:
`sudo config.scripts/blitz.datadrive.sh raid off`

philbertw4, 2020-10-26:
> Did to boot with the old USB plugged in?

I had booted first with the USB pendrive plugged in, but then saw the other issue where you commented to consider keeping it unplugged. With the USB pendrive plugged the raspiblitz seemed to want to use it as the external drive for the blockchain. So that's actually the first time I realized something was odd. Shutdown, pulled the usb pendrive out, reflashed the SD card with 1.6.1 (so, sadly, I do not have my original 1.6 SD card anymore), and tried again. That's when I ended up with the message that prompted this issue.

I will try the #1697 PR patch method now to see how that goes. Thanks for the help!

philbertw4, 2020-10-26:
Ok, after applying that PR, shutting down, plugging back in the usb pendrive and booting up, it seemed to recover everything (the update and contents of `raspiblitz.log` seemed to show it all worked). However then it went down for one final reboot and now I seem unable to SSH in. Oddly, being unable to ssh in originally is what prompted me to have to start this whole process (I had already upgraded LND a couple weeks ago before losing ssh access).

So, as far as I can tell now, my node has been repaired/updated, but I cannot confirm further until being able to ssh in.

openoms, 2020-10-26:
That is good news, thank you for confirming that the PR works.
For the `ssh` problem you can try this: https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-can-i-repair-my-ssh-login

(and don't forget that the intial password after the update is still `raspiblitz` and fail2ban bans you for 10 mins after 3 bad attempts)

philbertw4, 2020-10-27:
Ok, thanks again! What is most strange is that the blitz is not even showing up in my router anymore as an attached device. It has always been connected via wifi and, generally, has worked fine that way. This is also what was happening prior to the upgrade (and the btrfs issue here).

On the new SD card (with 1.6.1) I made sure to put the `wpa_supplicant` per usual. It must have worked at least somewhat since I was able to ssh in and start the repair/install, but then only after those reboots did it stop working. I'll keep looking into it.

openoms, 2020-10-27:
Place the wpa_supplicant again or use a wire and set up the wifi with:
`sudo raspi-config`

philbertw4, 2020-10-27:
Thanks again for the help. Still having this problem, so I tried the whole process again on a freshly re-burned SD card. This time it never seemed to show up on my network at all (despite doing the wpa_supplicant step like normal). I will move the device over to where it can be wired directly in and will try again. It is headless, so hopefully I haven't further corrupted anything with all these power-ups/downs!

philbertw4, 2020-10-27:
Ok, just following up. I seem to have been able to get everything working again by switching over to a wired connection. So, at this point, I think the problems I am having with my node are now in the realm of wifi and/or my local network setup as opposed to the original btrfs superblock error that prompted the opening of this issue. So, I think we can consider this issue closed and I will try to solve my wifi issues separately. Thanks much for all your help @openoms!!

closed 2020-10-27 by philbertw4