raspiblitz/raspiblitz #1718: Lnd gets announced as 127.0.0.1:9735

ChuckNorrison, 2020-10-27:
### Bug
After LAN got disconnected and reconnected, 1ml.com shows up my node with a new address: 127.0.0.1:9735. This Address keeps active till reboot and is shown in my menu screen.

### Environment
- Clearnet (no IPSec or TOR)
- OpenVPN Service which connects as Client to VPS with static IP (port forwardings at the VPS ok)
- DynDNS with duckdns.org (blitz is recognizing the public static ip fine)
- `curl -s http://v4.ipv6-test.com/api/myip.php` from internet.sh works if my I-Net is up, hangs if it is down

### Logs
- raspiblitz.conf publicIP=127.0.0.1

### Expected Behaviour
Never write private or loopback IPs as a publicIP

ChuckNorrison, 2020-10-27:
https://github.com/rootzoll/raspiblitz/pull/1719

rootzoll, 2020-10-29:
Which IP is announced is set by parameter when LND starts - and is is getting started by sytstemd service:
https://github.com/rootzoll/raspiblitz/blob/eea158949899f816d59f9adca19b1d8972814036/home.admin/assets/lnd.service#L14

And the value of `${publicIP}` is set by the `raspiblitz.conf` with:
https://github.com/rootzoll/raspiblitz/blob/eea158949899f816d59f9adca19b1d8972814036/home.admin/assets/lnd.service#L12

So the question would be: Why is their a `127.0.0.1` as publicIP in `raspiblitz.conf`?

The value get written there by config-script `internet.sh update-publicip`:
https://github.com/rootzoll/raspiblitz/blob/eea158949899f816d59f9adca19b1d8972814036/home.admin/config.scripts/internet.sh#L210

And that is called during bootup (before LND is started by systemd) by `_bootstrap.sh`:
https://github.com/rootzoll/raspiblitz/blob/eea158949899f816d59f9adca19b1d8972814036/home.admin/_bootstrap.sh#L481

So how to fix .. I would recommend to fix this in `inernet.sh` by not setting `127.0.01` as public IP ... but what to set as an replacement if that case is detected? If we keep it blank it could be that LND not starting because the parameter would be invalid. Should boostrap 'block' at that point until internet is available? I am not sure yet what the best solution would be. Open for input.

ChuckNorrison, 2020-10-29:
Oh i missed that part where the service gets started with that parameter. So at this point (publicIP=127.0.0.1) i would ignore the sed call in internet.sh. Maybe internet.sh should just quit if connectivity fails and wait for it. 

After that, i think i found the reason for 127.0.0.1 here:
https://github.com/rootzoll/raspiblitz/issues/312#issuecomment-462675101

So no ip is not a solution. 127.0.0.1 is still a bad idea.

rootzoll, 2020-10-29:
@ChuckNorrison can you try if the PR #1729 solves this? You can use a v1.6.1 release and then get the PR with `UPDATE > PATCH > PATCHMENU > PR > 1729` and restart.

ChuckNorrison, 2020-10-29:
Already done successful :) awesome thanks

rootzoll, 2020-10-29:
Perfect ... I merged the PR and will be part of v1.6.2 release. Closing issue.

closed 2020-10-29 by rootzoll