raspiblitz/raspiblitz #1723: Increase default Mempool Size

geco91, 2020-10-28:
see: https://twitter.com/BTCSchellingPt/status/1321548973337833473?s=20

By default (based on old raspberrypi3 raspibolt optimizations) it was set to `maxmempool=64` in https://github.com/rootzoll/raspiblitz/blob/3432b62d4fa86d795d18b75a7ff9ec7e471a7811/home.admin/assets/bitcoin.conf#L25

It should be bigger by default now and on setup it should check how much RAM the pi has and give more space here (up to 500GB on a 4GB RP4).

openoms, 2020-10-29:
conceptACK here.

Shared the manual settings and rebroadcast options in this thread: https://twitter.com/openoms/status/1321564988146917377?s=20

openoms, 2020-10-29:
Checked and this is already happening:
https://github.com/rootzoll/raspiblitz/blob/eea158949899f816d59f9adca19b1d8972814036/home.admin/_bootstrap.provision.sh#L148

over 1.5GB RAM `maxmempool=256` is set.

we could increase it further comfortably for the nodes with 4GB RAM and above.

rootzoll, 2020-10-30:
Hehe .. when you optimize and forget about it :)
Yes an increase to 512MB when over 3.5GB RAM should be much future save.

rootzoll, 2020-10-30:
PR #1730 merged. Will be part of v1.6.2 release. Closing issue.

closed 2020-10-30 by rootzoll