raspiblitz/raspiblitz #1724: BTCpay: use with testnet

openoms, 2020-10-29:
Currently BTCPayServer does not work with testnet: https://github.com/rootzoll/raspiblitz/issues/1198

Will need to identify the relevant config options and apply automatically.

rootzoll, 2020-10-30:
The settings menu does not allow to switch to testnet while BTCPayServer is installed:
https://github.com/rootzoll/raspiblitz/blob/ae4be0596bc68feef42adb684e3c9d4b80f8f3e8/home.admin/00settingsMenuBasics.sh#L119

So I think we can push this to a later release.

openoms, 2020-10-30:
Agree that it is not urgent. 
To play around on testnet there is also: https://testnet.demo.btcpayserver.org

we-do-it-lu, 2021-01-20:
Although not urgent newbies will fall in this issue as they use TESTNET often by default. 
Updating the README with one line to mention this issue #1724.

openoms, 2021-03-17:
Got all the config options:

```
admin@rpi4GB64:~ $ sudo -u btcpay /home/btcpay/dotnet/dotnet run --no-launch-profile --no-build -c Release -p "/home/btcpay/btcpayserver/BTCPayServer/BTCPayServer.csproj" -- -h
BTCPay
Open source, self-hosted payment processor.

Usage: BTCPay [options]

Options:
  -? | -h | --help            Show help information
  -n | --network              Set the network among (mainnet,testnet,regtest) (default: mainnet)
  --testnet | -testnet        Use testnet (deprecated, use --network instead)
  --regtest | -regtest        Use regtest (deprecated, use --network instead)
  --signet | -signet          Use signet (deprecated, use --network instead)
  --allow-admin-registration  For debug only, will show a checkbox when a new user register to add himself as admin. (default: false)
  --chains | -c               Chains to support as a comma separated (default: btc; available: btc)
  --postgres                  Connection string to a PostgreSQL database
  --mysql                     Connection string to a MySQL database
  --sqlitefile                File name to an SQLite database file inside the data directory
  --externalservices          Links added to external services inside Server Settings / Services under the format service1:path2;service2:path2.(default: empty)
  --bundlejscss               Bundle JavaScript and CSS files for better performance (default: true)
  --rootpath                  The root path in the URL to access BTCPay (default: /)
  --sshconnection             SSH server to manage BTCPay under the form user@server:port (default: root@externalhost or empty)
  --sshpassword               SSH password to manage BTCPay (default: empty)
  --sshkeyfile                SSH private key file to manage BTCPay (default: empty)
  --sshkeyfilepassword        Password of the SSH keyfile (default: empty)
  --sshauthorizedkeys         Path to a authorized_keys file that BTCPayServer can modify from the website (default: empty)
  --sshtrustedfingerprints    SSH Host public key fingerprint or sha256 (default: empty, it will allow untrusted connections)
  --torrcfile                 Path to torrc file containing hidden services directories (default: empty)
  --socksendpoint             Socks endpoint to connect to onion urls (default: empty)
  --updateurl                 Url used for once a day new release version check. Check performed only if value is not empty (default: empty)
  --debuglog                  A rolling log file for debug messages.
  --debugloglevel             The severity you log (default:information)
  --disable-registration      Disables new user registrations (default:true)
  --plugin-remote             Which github repository to fetch the available plugins list (default:btcpayserver/btcpayserver-plugins)
  --recommended-plugins       Plugins which would be marked as recommended to be installed. Separated by newline or space
  --xforwardedproto           If specified, set X-Forwarded-Proto to the specified value, this may be useful if your reverse proxy handle https but is not configured to add X-Forwarded-Proto (example: --xforwardedproto https)
  --btcexplorerurl            URL of the NBXplorer for BTC (default: http://127.0.0.1:24444/)
  --btcexplorercookiefile     Path to the cookie file (default: /home/btcpay/.nbxplorer/Main/.cookie)
  --btclightning              Easy configuration of lightning for the server administrator: Must be a UNIX socket of c-lightning (lightning-rpc) or URL to a charge server (default: empty)
  --btcexternallndgrpc        The LND gRPC configuration BTCPay will expose to easily connect to the internal lnd wallet from an external wallet (default: empty)
  --btcexternallndrest        The LND REST configuration BTCPay will expose to easily connect to the internal lnd wallet from an external wallet (default: empty)
  --btcexternalrtl            The Ride the Lightning configuration so BTCPay will expose to easily open it in server settings (default: empty)
  --btcexternalspark          Show spark information in Server settings / Server. The connection string to spark server (default: empty)
  --btcexternalcharge         Show lightning charge information in Server settings/Server. The connection string to charge server (default: empty)
  -c | --conf                 The configuration file
  -p | --port                 The port on which to listen
  -b | --bind                 The address on which to bind
  -d | --datadir              The data directory
```

openoms, 2021-03-17:
For NBXplorer:
```
sudo -u btcpay /home/btcpay/dotnet/dotnet "/home/btcpay/NBXplorer/NBXplorer/bin/Release/netcoreapp3.1/NBXplorer.dll" -c /home/btcpay/.nbxplorer/Main/settings.config -h
NBXplorer
Lightweight block explorer for tracking HD wallets

Usage: NBXplorer [options]

Options:
  -? | -h | --help         Show help information
  -n | --network           Set the network among (mainnet,testnet,regtest) (default: mainnet)
  --testnet | -testnet     Use testnet
  --regtest | -regtest     Use regtest
  --signet | -signet       Use signet
  --chains                 Chains to support comma separated (default: btc, available: agm,btc,btx,ltc,doge,bch,grs,btg,dash,trc,polis,mona,ftc,ufo,via,xmcc,gbx,colx,chc,lbtc,qtum,mue)
  --dbcache                If more than 0, the size of the cache for the database, in MB. Else, no limit on the size of the cache. (default: 50)
  --agmrescan              Rescan from startheight
  --agmrpcuser             RPC authentication method 1: The RPC user (default: using cookie auth from default network folder)
  --agmrpcpassword         RPC authentication method 1: The RPC password (default: using cookie auth from default network folder)
  --agmrpccookiefile       RPC authentication method 2: The RPC cookiefile (default: using cookie auth from default network folder)
  --agmrpcauth             RPC authentication method 3: user:password or cookiefile=path (default: using cookie auth from default network folder)
  --agmrpcurl              The RPC server url (default: default rpc server depended on the network)
  --agmstartheight         The height where starting the scan (default: where your rpc server was synched when you first started this program)
  --agmminutxovalue        The minimum value of tracked UTXOs, any UTXO with value less than this is ignored. (default: 1 (satoshi))
  --agmnodeendpoint        The p2p connection to a Bitcoin node, make sure you are whitelisted (default: default p2p node on localhost, depends on network)
  --agmhastxindex          If true, NBXplorer will try to fetch missing transactions from the local node (default: false)
  --btcrescan              Rescan from startheight
  --btcrpcuser             RPC authentication method 1: The RPC user (default: using cookie auth from default network folder)
  --btcrpcpassword         RPC authentication method 1: The RPC password (default: using cookie auth from default network folder)
  --btcrpccookiefile       RPC authentication method 2: The RPC cookiefile (default: using cookie auth from default network folder)
  --btcrpcauth             RPC authentication method 3: user:password or cookiefile=path (default: using cookie auth from default network folder)
  --btcrpcurl              The RPC server url (default: default rpc server depended on the network)
  --btcstartheight         The height where starting the scan (default: where your rpc server was synched when you first started this program)
  --btcminutxovalue        The minimum value of tracked UTXOs, any UTXO with value less than this is ignored. (default: 1 (satoshi))
  --btcnodeendpoint        The p2p connection to a Bitcoin node, make sure you are whitelisted (default: default p2p node on localhost, depends on network)
  --btchastxindex          If true, NBXplorer will try to fetch missing transactions from the local node (default: false)
  --btxrescan              Rescan from startheight
  --btxrpcuser             RPC authentication method 1: The RPC user (default: using cookie auth from default network folder)
  --btxrpcpassword         RPC authentication method 1: The RPC password (default: using cookie auth from default network folder)
  --btxrpccookiefile       RPC authentication method 2: The RPC cookiefile (default: using cookie auth from default network folder)
  --btxrpcauth             RPC authentication method 3: user:password or cookiefile=path (default: using cookie auth from default network folder)
  --btxrpcurl              The RPC server url (default: default rpc server depended on the network)
  --btxstartheight         The height where starting the scan (default: where your rpc server was synched when you first started this program)
  --btxminutxovalue        The minimum value of tracked UTXOs, any UTXO with value less than this is ignored. (default: 1 (satoshi))
  --btxnodeendpoint        The p2p connection to a Bitcoin node, make sure you are whitelisted (default: default p2p node on localhost, depends on network)
  --btxhastxindex          If true, NBXplorer will try to fetch missing transactions from the local node (default: false)
  --ltcrescan              Rescan from startheight
  --ltcrpcuser             RPC authentication method 1: The RPC user (default: using cookie auth from default network folder)
  --ltcrpcpassword         RPC authentication method 1: The RPC password (default: using cookie auth from default network folder)
  --ltcrpccookiefile       RPC authentication method 2: The RPC cookiefile (default: using cookie auth from default network folder)
  --ltcrpcauth             RPC authentication method 3: user:password or cookiefile=path (default: using cookie auth from default network folder)
  --ltcrpcurl              The RPC server url (default: default rpc server depended on the network)
  --ltcstartheight         The height where starting the scan (default: where your rpc server was synched when you first started this program)
[... cut ...]
  --cachechain             Whether the chain of header is locally cached for faster startup (default: true)
  --rpcnotest              Faster start because RPC connection testing skipped (default: false)
  --exposerpc              Expose the node RPC through the REST API (default: false)
  -v | --verbose           Verbose logs (default: true)
  -c | --conf              The configuration file
  -p | --port              The port on which to listen
  -b | --bind              The address on which to bind
  -d | --datadir           The data directory
```

rootzoll, 2021-03-17:
merged #2067 - test with "testnet"

rootzoll, 2021-04-18:
Should be OK for v1.7 ... but pushing final testing to v1.7.1

danielcharrua, 2022-05-08:
Hey @rootzoll @openoms do you think that in a near future we could run an installation of BTCPay Server on testnet? I'm looking to make some test payments and validate them with my node but I can't manage how to do it.