raspiblitz/raspiblitz #1725: BTCpay: use postgres database by default

openoms, 2020-10-29:
Postgres is better supported database format for for BTCPayServer as described in:  https://github.com/btcpayserver/btcpayserver/issues/1996

Will need to install postgres in the install script and have it running on BTCpay install to be used by default.

openoms, 2020-11-10:
Will try to make this work asap as restoring my sqllite database did not work no the second time. BTCpay cannot generate invoices any more despite all data was preserved.

openoms, 2020-11-11:
Not exactly sure how to go on configuring postgres.
so far found the compatible package:
`$ sudo apt install postgresql-common`
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  postgresql-client-common ssl-cert
Suggested packages:
  libjson-perl openssl-blacklist
The following NEW packages will be installed:
  postgresql-client-common postgresql-common ssl-cert
0 upgraded, 3 newly installed, 0 to remove and 123 not upgraded.
Need to get 331 kB of archives.
After this operation, 911 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive-bm.raspbian.org/raspbian buster/main armhf postgresql-client-common all 200+deb10u4 [85.1 kB]
Get:2 http://archive-bm.raspbian.org/raspbian buster/main armhf ssl-cert all 1.0.39 [20.8 kB]
Get:3 http://archive-bm.raspbian.org/raspbian buster/main armhf postgresql-common all 200+deb10u4 [225 kB]
Fetched 331 kB in 1s (450 kB/s)            
Preconfiguring packages ...
Selecting previously unselected package postgresql-client-common.
(Reading database ... 100123 files and directories currently installed.)
Preparing to unpack .../postgresql-client-common_200+deb10u4_all.deb ...
Unpacking postgresql-client-common (200+deb10u4) ...
Selecting previously unselected package ssl-cert.
Preparing to unpack .../ssl-cert_1.0.39_all.deb ...
Unpacking ssl-cert (1.0.39) ...
Selecting previously unselected package postgresql-common.
Preparing to unpack .../postgresql-common_200+deb10u4_all.deb ...
Adding 'diversion of /usr/bin/pg_config to /usr/bin/pg_config.libpq-dev by postgresql-common'
Unpacking postgresql-common (200+deb10u4) ...
Setting up postgresql-client-common (200+deb10u4) ...
Setting up ssl-cert (1.0.39) ...
Setting up postgresql-common (200+deb10u4) ...
Adding user postgres to group ssl-cert

Creating config file /etc/postgresql-common/createcluster.conf with new version
Building PostgreSQL dictionaries from installed myspell/hunspell packages...
  en_gb
Removing obsolete dictionary files:
Created symlink /etc/systemd/system/multi-user.target.wants/postgresql.service → /lib/systemd/system/postgresql.service.
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
```

BTCpayserver should pick it up when it is running as per https://github.com/btcpayserver/btcpayserver/issues/1996#issuecomment-711172039:
>We cannot really create the postgres like we did with sqlite as you need postgres running as a service, configured and ready to be connected. We do create the database once we connect to the postgres server though.

frennkie, 2020-11-11:
> warn: Configuration:  SQLite is not widely tested and should be considered experimental, we advise you to use postgres instead.

I assume they mean "not widely tested in BTCPay" - right?! (not sure that is actually true..)

> Not exactly sure how to go on configuring postgres.

The plan is to run the PostgreSQL server, right?

```
sudo apt install -y postgresql-11
```

Check version:

```
sudo -u postgres psql --version
psql (PostgreSQL) 11.9 (Raspbian 11.9-0+deb10u1)
```

Open DB shell

```
sudo -u postgres psql
\l   <- this is list the databases
\q  <- quit
``` 

Where are you stuck?

frennkie, 2020-11-11:
I would assume that (as usually with this stuff) a database needs to be created for BTCPay and a user (which also needs to be assigned access to the DB).

frennkie, 2020-11-11:
> Nov 10 21:04:59 raspberrypi dotnet[5703]:  ---> Microsoft.Data.Sqlite.SqliteException (0x80004005): SQLite Error 19: 'UNIQUE constraint failed: AddressInvoices.Address'.

You can manually go into your sqlite file.. maybe you are able to fix the problem. The `UNIQUE constraint failed` on `AddressInvoices.Address` error indicates that it appears to create a new address entry for an address that already exists. It's very likely that this issue would be exactly the same after switching from SQLite to PostgreSQL.

frennkie, 2020-11-11:
Currently the sqlite file for BTCPay is stored on the external disk. If we move the PostgreSQL we would either need to put the database on the sd card (not a good idea) or on to the external disk (also not such a good idea).

SD-Card:

* reads and writes
* gone on update (requires robust back&restore)
* ...?

External Disk:

* might cause issues when disk is missing and service doesn't come up
* ...?

openoms, 2020-11-11:
Thank you for all the input @frennkie!

>I assume they mean "not widely tested in BTCPay" - right?! (not sure whether that is actually true..)
yes, it iso nly about btcpay. they say, that there is a lot of conversion needed to make some features work.

First of all will try to fix my database and would probably make sense to put off the issue to v1.7.

I think we should continue keeping the app-data on the disk including the postgres database.
The main reason to do this is the recommendation from the BTCpay devs.

openoms, 2020-11-12:
> > Nov 10 21:04:59 raspberrypi dotnet[5703]:  ---> Microsoft.Data.Sqlite.SqliteException (0x80004005): SQLite Error 19: 'UNIQUE constraint failed: AddressInvoices.Address'.
> 
> You can manually go into your sqlite file.. maybe you are able to fix the problem.
> 
> ```
> btcpay@raspberrypi:~ $ sqlite3 ~/.btcpayserver/Main/sqllite.db
> .tables  <- list tables
> select * from AddressInvoices limit 3;
> ```
> 
> The `UNIQUE constraint failed` on `AddressInvoices.Address` error indicates that it tries to create a new address entry for an address that already exists. It's very likely that this issue would be exactly the same after switching from SQLite to PostgreSQL.

@frennkie thanks so much you pointed me to the right direction, was able to fix the issue pretty quickly by deleting all the previous invoices from the database. Good thing is my shop settings, API keys etc did all remain intact.

what I did:
```
$ sudo apt install sqlite3
$ sudo su - btcpay
$ sqlite3 ~/.btcpayserver/Main/sqllite.db
sqlite> select * from AddressInvoices;
 # shows all the invoice
sqlite> delete from AddressInvoices;
sqlite> vacuum;
sqlite> select * from AddressInvoices;
 # all invoices gone
```
now have no errors on generating invoices.

openoms, 2020-12-13:
Installation of postgres covered here: https://docs.btcpayserver.org/ManualDeploymentExtended/#-install
and https://docs.btcpayserver.org/ManualDeploymentExtended/#2-create-postgresql-database

rootzoll, 2020-12-14:
so is this also an issue vor v1.6.3 or do we keep it on 1.7?

openoms, 2020-12-15:
Can  be kept for a future release. BTCPay is fully functional with sqllite.

tisvargas, 2020-12-22:
---
__Advertisement :)__

PrinzMonty, 2022-10-02:
I think BTCPay is not fully functional with sqllite #3339

In the "bonus.btcpayserver.sh" there are already some commented out lines about the topic, what is it about?

rootzoll, 2022-12-11:
OK PostgreSQL is already getting installed/used by the btcpay install script -  see:
https://github.com/rootzoll/raspiblitz/blob/1f0416bbbb3a297ee2bd4c824c1575b3c9141963/home.admin/config.scripts/bonus.btcpayserver.sh#L727

But its just used for the NBXplorer - the BTCPay database is still SQLlite:
https://github.com/rootzoll/raspiblitz/blob/1f0416bbbb3a297ee2bd4c824c1575b3c9141963/home.admin/config.scripts/bonus.btcpayserver.sh#L86

I will push this up to future milestone .. but if anybody will submit & test a PR with postgesSQL including providing either a migration for old SQLlite users when updating or just makes sure new users start with PostgreSQL while old users keep using SQL lite ... I can inlude in the upcoming release.

PrinzMonty, 2022-12-12:
> 

If I find time over Christmas, I will work on it, but I have a lot of work here and I have to prioritize. If I don't manage, we'll have to wait for 1.9... 

Thank you for adding it to the 1.9 milestone! :)

rootzoll, 2022-12-13:
I am trying to have an first Release Candidate out before Christmas .. but if possible I will see if we can add it for a second RC then for v1.9.0.

openoms, 2023-01-29:
> Is a migration service from SQLite to PostgreSQL planned? Similar to DBTrieToPostgresMigratorHostedService in nbxplorer.

I have been informed by the btcpay team that there is no recommended way currently to migrate to postgres so from v1.9.0 we will have the new installs start with a postgres databse. Sqlite is planned to be deprecated so eventually users will need to start a new database. For now the existing sqlite databases are kept and being used.

rootzoll, 2023-02-03:
waiting for PR #3641 to merge & fix this

openoms, 2023-04-14:
the simplest way to update to a new BTCpay version using the postgres database:
* uninstall btcpay and delete store data when asked
* sync to the v1.9 (or dev) branch from the UPDATE - PATCH menu - this will get you the updated install script
* install btcpay - it will default to using postgres
* from there the BTCPAY UPDATE should work inclusing updating to v1.9.0 of BTCPay

if you want to keep the store data would need to migrate - this is to be done once manually: https://github.com/btcpayserver/btcpayserver/blob/master/docs/db-migration.md

(also posted at https://t.me/raspiblitz/126873)

rootzoll, 2023-05-10:
merged PR #3756 - this will be part of v1.9.0

PR #3641 (2023-02-04): BTCpay and NBXplorer update, use postgres for new installs 
fixes #1725 and #3622