raspiblitz/raspiblitz #1727: SSH connection refused during BC sync and after reboot

nderambure, 2020-10-29:
Just to report that during Blockchain sync during fresh install, if I disconnect SSH and try to reconnect, the password is refused (yes it's the good one ;) )
Same after reboot.
I'm restarting again from scratch to see if anything changed.
I've seen couple of messages with same problem.
I'm using Raspiblitz 1.6.0 on a Pi4 and the hard drive from shopping list.

ChuckNorrison, 2020-10-29:
you could try `sudo systemctl restart sshd` to solve it temporary, i observed a failed sshd at boot time in unknown circumstances

nderambure, 2020-10-30:
Thx, I'll try that in my next attempt.

ChuckNorrison, 2020-11-07:
this check_network attempt could be helpful in here:
https://github.com/raspberrypi/linux/issues/3034#issuecomment-723420437

nyxnor, 2021-05-17:
@rootzoll Please close this

closed 2021-05-18 by rootzoll