raspiblitz/raspiblitz #1740: Service: Add Lightning Pool / Terminal

joernroeder, 2020-11-03:
Read More on lightning pool https://lightning.engineering/posts/2020-11-02-pool-deep-dive/ and https://github.com/lightninglabs/lightning-terminal

petrhejna, 2020-11-03:
There is already this PR https://github.com/rootzoll/raspiblitz/pull/1739

:tada:

openoms, 2020-11-03:
Started to collect some Pool usage notes I found useful: https://openoms.gitbook.io/lightning-node-management/pool

Let us know your experiences!

rootzoll, 2020-11-06:
Merged the PRs ... now needs testing with install, deinstall & recovery.

rootzoll, 2020-11-06:
OK tested recovery on update process - ran thru OK. Also install & deinstall over SSH menus is working.
So this will be part of v1.6.2 release - closing issue.

closed 2020-11-06 by rootzoll

petrhejna, 2020-11-07:
I am happy about the pool, but will there be Terminal as well? I went through code of referenced PR and it seems that it is for pool only.

I have read that Terminal has some UI for the pool. If it is the case, I believe it would be super handy to have this tool available in Raspiblitz.

Thank you ;)

glowacki-dev, 2020-11-07:
@petrhejna It seems like LiT only added pool daemon for now and doesn't have any UI available yet.

I've actually tried adding LiT to raspiblitz some time ago, but abandoned the idea after finding out that it doesn't allow using existing daemons for loop and faraday (and now for pool as well) but will always start them on it's own. Since raspiblitz allows you to run those daemons separately there could be some issues when LiT runs them again (probably some conflicting ports or separate databases and configs leading to different behaviors depending on which daemon is used). However, their documentation [mentions](https://github.com/lightninglabs/lightning-terminal/blob/master/doc/config-lnd-remote.md#additional-configuration) that ability to use existing daemons will be available in the future, and it should make integration much easier.

rootzoll, 2020-11-11:
> I am happy about the pool, but will there be Terminal as well? I went through code of referenced PR and it seems that it is for pool only.

Yes sorry. This issue just ads pool as a CLI. Mostly because of the findings written by @glowacki-dev - will edit issue topic

openoms, 2020-11-13:
here are some notes about pool:
https://openoms.gitbook.io/lightning-node-management/pool

The script linked contains the latest Pool version and now has an update option.