raspiblitz/raspiblitz #1741: Reserach: Local HTTP to Torv3 forward

rootzoll, 2020-11-03:
For RaspiBlitz WebUI we are aiming to let the user use the Tor browser to have a secured communication line from the beginning. But there was always the problem, that the user has to input the TorV3 address into the Tor browser first. The LCD could display it - but its too long for any acceptable UX.

So here is one idea to try:
- The LCD will display a HTTP address with the local IP like: http://192.168.8.34 ... we cannot use https here because we would not get a letsencrypt for a local IP and having the user to accept a browser security warning is bad too ... so we will display that unsecure just http site.
- If the user calls it already with a Tor browser, it will forward directly to the TorV3 address and he user can confirm that there was no Man-In-The-Middle by checking a 4 digit code from the LCD being the same as the displayed in the browser (or maybe the first 6 chars of tor address or something like that)
- If the user call the http-address normal web browser it will explain how to download, install and use the Tor browser or to use SSH login as fallback.

There is still the edge case that the users laptop/mobile are not in the same local network ... for that we could offer in the LCD a button that would create a thrid-party URL forwarding like bit.ly/67383 or raspiblitz.org/76d743 that would forward to the Torv3 address.

nyxnor, 2021-07-22:
display web ui onion address on QR code to be scanned for mobile

rootzoll, 2022-05-09:
To reduce complexity "Tor & WebUI" will not be a focus of the v1.8.0 release. Pushing this to v1.8.2 for now.

rootzoll, 2024-09-26:
Closing this reasearch - see: https://github.com/raspiblitz/raspiblitz/issues/3105#issuecomment-2377474533