raspiblitz/raspiblitz #1745: LCD Screen Blinking after boot

Aki113s, 2020-11-04:
I can get in via SSH, so the system is working, it seems after every update I battle with the LCD screen. I had saved the commands to install the driver and used it the last 3 updates and it worked fine once rebooted, this one not so much. Ideas??

openoms, 2020-11-04:
Sounds like you are not using the LCD screen recommended in the readme and supported by the project with drivers installed in the SDcard image.
Can only  suggest to follow the manufacturer's instructions to set it up.
Can post the exact type and maybe pictures including the display's back, someone might have used it.

Aki113s, 2020-11-04:
[https://ibb.co/LrggmXm](url)
 I have a 3.5 inch display so ill try that, but it is a XPT2046. I also tried the following commands to install drivers which always worked before.

sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MPI4008-show

Ben10Brasileiro, 2021-01-10:
`sudo raspi-config`
7 Advanced options
A6 Screen Blanking
Select "No"

1Vincent2, 2021-01-10:
Unsubscribe

nyxnor, 2021-04-18:
Please close this issue if it was solved.

nyxnor, 2021-05-17:
@rootzoll Please close this

closed 2021-05-17 by rootzoll